dresden-elektronik/deconz-rest-plugin #4381: RaspBee II / ConBee II - can't add devices

schneidermarc, 2021-02-09:
## RaspBee II / ConBee II - can't add devices

I am completely lost ...

A few days ago I installed my RaspBee II and added the first devices.
At some point I lost the connection to my Zigbee devices.

Since I couldn't find the error and I had only added 10 devices - I decided to:
1. delete the Deconz container
2. delete the linked config folder
3. reboot
4. update the firmware again
5. re-deploy the container

... and try to start with a fresh installation to add the devices as the installation before.

I reset the devices and put them into bind mode but I can't find anything with the Phoscon PWA.

I tried to re-bind:
- Phillips Hue Light Bulbs
- Osram SMART+ Plugs

After a few tries I added the devices via Hue Bridge to make sure that they are in bind mode and that worked fine. In the next step I removed the devices again and tried to add it via the Phoscon GW - nothing to find.

At this point I thought my RaspBee 2 is defective and I bought a Conbee 2 to check.
Unfortunately, I have the same result with ConBee 2 :-(

I repeated _my installation steps_ (2-4) with both devices and tried different Zigbee channels - same result.

## Environment
Raspberry Pi 4 8GB
_- USB Boot from SSD (no other devices connected)_
Docker version 20.10.2, build 2291f61
Deconz Version: 2.09.01
Phoscon Version: 23.12.2020
ConBee II Firmware: 0x26680700
RaspBee II Firmware: 0x26680700
USB extension cable: no
Zigbee Channel: 11

## My Installation Steps

1. /boot/config.txt:
```text
dtparam=audio=on
max_usb_current=1
dtoverlay=disable-bt
dtoverlay=disable-wifi
enable_uart=1
dtoverlay=pi3-miniuart-bt
dtoverlay=uart0-full
core_freq=250
dtoverlay=pi3-miniuart-bt-overlay
```

2. Update Firmware
```console
root@raspberrypi:/home/pi# sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f  deCONZ_ConBeeII_0x26680700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2401245-if00 (ConBee II)
deCONZ firmware version 26680700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164373 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
deCONZ firmware version 26680700
```
[... 178 lines cut ...]
13:43:35:574 CTRL got nwk update id 8
13:43:35:581 Device TTL 2840 s flags: 0x7
13:44:35:572 Current channel 11
13:44:35:578 CTRL got nwk update id 8
13:44:35:585 Device TTL 2780 s flags: 0x7
13:44:41:628 dev /dev/ttyAMA0
13:44:41:629 GW firmware version: 0x26680700
13:44:41:629 GW firmware version is up to date: 0x26680700
13:45:35:573 Current channel 11
13:45:35:580 CTRL got nwk update id 8
13:45:35:587 Device TTL 2720 s flags: 0x7
13:46:35:571 Current channel 11
13:46:35:578 CTRL got nwk update id 8
13:46:35:585 Device TTL 2660 s flags: 0x7
13:47:35:570 Current channel 11
13:47:35:576 CTRL got nwk update id 8
13:47:35:583 Device TTL 2600 s flags: 0x7
13:48:35:563 Current channel 11
13:48:35:567 CTRL got nwk update id 8
13:48:35:571 Device TTL 2540 s flags: 0x7
13:48:41:639 dev /dev/ttyAMA0
13:48:41:641 GW firmware version: 0x26680700
13:48:41:641 GW firmware version is up to date: 0x26680700
13:49:35:567 Current channel 11
```

Smanar, 2021-02-09:
So first, thx for your issue, lol, I think I have never see an issue with so much information ^^.
Now if you have a Raspbee why are you using conbee firmware ?

schneidermarc, 2021-02-09:
> So first, thx for your issue, lol, I think I have never see an issue with so much information ^^.
> Now if you have a Raspbee why are you using conbee firmware ?

I wrote:
> At this point I thought my RaspBee 2 is defective and I bought a Conbee 2 to check.
> Unfortunately, I have the same result with ConBee 2 :-(

I flashed both devices with the correct firmware :-)

bsheepdiy, 2021-02-09:
A new stable version of deCONZ is out today 2.09.02, check if you still have the issue.
You could try using an USB extension cable.

If you could not add new device, maybe your zigbee channel interfere with some existing Wi-Fi network. Use Wifi scan on your smartphone to scan WiFi network. (see https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html)

Another solution, sniff zigbee using CC2531 with snif firmware to watch in deep Zigbee trames.

Smanar, 2021-02-09:
> I flashed both devices with the correct firmware :-)

Lol, my bad, too much informations to read this time ^^.

But if you have a conbee, will be better, after have made the bsheepdiy suggestion (USB extension and avoid usb 3.0), if you still have the issue, can you make a try on a PC machine ?

Mimiix, 2021-02-10:
AS other mentioned: Try a extension cable. Both for the SSD and Conbee. 

What also might be worth your while, to confirm if theres any interference:  Put your OS on a SD card and remove the SSD.

schneidermarc, 2021-02-11:
@Smanar @Mimiix thank you for your helpful comments! :-)

The last few days I had so many test scenarios and nothing seemed to solve the problem.
With a new Raspberry PI OS installation on a fresh SD-Card I was able to find devices again for the first time.

To keep it short, it was not the SSD, the power supply or something with interferences - **after resetting the GPIO settings in the Raspi-Config everything worked fine** with my old setup.

But to keep in mind - nothing has changed since the last time everything worked fine. I think it's a Raspbian OS Bug (maybe ;-) )

closed 2021-02-11 by schneidermarc

schneidermarc, 2021-02-11:
Btw why it didn't work with the Conbee either, I didn't find out.

Smanar, 2021-02-11:
There is a reset procedure in raspi-config for GPIO ?

hixus, 2021-06-08:
> @Smanar @Mimiix thank you for your helpful comments! :-)
>
> The last few days I had so many test scenarios and nothing seemed to solve the problem.
> With a new Raspberry PI OS installation on a fresh [...]

I'm facing similar situation but did only find remote GPIO setting from raspi-config. Could you elaborate what did "resetting the GPIO settings in the Raspi-Config" require?

Smanar, 2021-06-09:
You have same results when trying to update the firmware ?

hixus, 2021-06-10:
> You have same results when trying to update the firmware ?

This was with latest stable firmware but after trying latest beta it seems to work fine.