dresden-elektronik/deconz-rest-plugin #4382: ROBB smarrt ROB_200-010-0 version 2.5.3_r48 (version 2.5.3_r46 works)

etmrugl, 2021-02-09:
## Device

- Product name: ROB_200-010-0
- Manufacturer: ROBB smarrt
- Model identifier: ROB_200-010-0
- Device type : 
  - Other: Window Covering Controller (shutter)

## Screenshots
[image]
[image]
[image]

I attached a log. The not-working module is: 15:52:55:626 LightNode 35: Rolluik Keuken added

[deCONZ.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5952160/deCONZ.log)

I can see the node in the deCONZ-Conbee II windows on the Raspberry and I see the node in the Phoscon App. The node does NOT respond to any command.

Mimiix, 2021-02-10:
Can you please add all cluster screenshots?

etmrugl, 2021-02-10:
Is this what you ar elooking for? I am not sure if I understand your question.
[image]

etmrugl, 2021-02-10:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

etmrugl, 2021-02-10:
It gets weirder...
[image]

If I use the 'exec' button at Down/close the shutter closes. Also works with the 'exec' button Up/Open. The shtter does not work in the Phoscon App

SwoopX, 2021-02-10:
What's weird operating a device as it is supposed to be?

You said you can't control it via Phoscon. Well, I also can't control mine, so nothing extraordinary from my perspective. How would one do that at all?

etmrugl, 2021-02-10:
version 2.5.3_r46 can be controlled in Phoscon as a light on/off. In Domoticz I defined this device as a inverted blind. and voila, it works.

The weird thing is that version 2.5.3_r48 can be controlled through the exec function in the Window covering cluster. In the Phoscon app I see the module as a light, but commands do not work.

etmrugl, 2021-02-13:
Do I need to add more information?

closed 2021-02-25 by etmrugl