dresden-elektronik/deconz-rest-plugin #4389: OTA update Headless

MeandtheDevilknockingonyourdoor, 2021-02-10:
## Describe the question or issue you are having

Is it possible to do a OTA update without Gui on a headless system?

## Screenshots

## Environment

 - Host system: Raspberry
 - Running method: Rasbian Headless
 - Firmware version: 26680700
 - deCONZ version: 2.09.02 / 23.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

Mimiix, 2021-02-10:
Nope. You need the gui :)

other option would be to follow this:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

Smanar, 2021-02-10:
It s not possible for sensor, but it s for light nope ?
If files are at good place, and if you enable the option in the old webapp, it don't work ?

MeandtheDevilknockingonyourdoor, 2021-02-13:
Ok, no with the vnc I try but it didn’t work. 
where is a good place for the updates?
Thx

Smanar, 2021-02-13:
You can find some informations here 
https://github.com/dresden-elektronik/deconz-ota-plugin
https://phoscon.de/en/support#ota-update-osram-devices

But from my memory it can work without the GUi for light, just need to enable the feature, what is your brand and device ?

MeandtheDevilknockingonyourdoor, 2021-02-13:
Ok I will check it,

I have only ikea and hue lights and in the future a ikea motion sensor.

closed 2021-03-14 by github-actions[bot]