dresden-elektronik/deconz-rest-plugin #4391: Device name Amiko Home Smart Plug (Manufacturer Feibit)

Satmedia, 2021-02-10:
## Device

- Product name: Amiko Smart Plug
- Manufacturer: Feibit
- Model identifier: FB56+SKT12KJ1.6
- Device type : Other: Smart Plug

Device is connected as Light in Phoscon APP and it is working as an On/Off Plugin Switch.
BUT it is not working as a router in deCONZ!

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

Mimiix, 2021-02-10:
Hi,

I told you on discord: Devices pick their own routes. We can't do anything on that. There's some information about it here:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4376

As far as i am concerned, this device seems to be suppoted already. I'd like to have @SwoopX or @smanar confirming this.

SwoopX, 2021-02-10:
Yep, nothing to be done here.

Satmedia, 2021-02-10:
Thank you for your fast answers. Yes the Smartplug is working well, only no routing :-(

Mimiix, 2021-02-10:
@Satmedia As i said, devices pick themselves. As long as they work, don't worry.

closed 2021-02-10 by Mimiix

Satmedia, 2021-02-10:
Sorry it was a translation error.The devices are working if they are connected to conbee usb stick directly.Means it are in range for direct connection.They do not make a connection to a router.Mfg Ing. Roland Kroiss Satmedia Austriawww.satmedia.atVon meinem Samsung Galaxy Smartphone gesendet.