dresden-elektronik/deconz-rest-plugin #4392: No communication after Update (nothing works anymore)

Maschler, 2021-02-10:
## Describe the bug
Hello,

after I made an update from the Raspi, unfortunately nothing works anymore. In the Deconz representation no more connections between the individual components arise. In the Phoscon APP all components are there, but they are grayed out. Restart etc. I have already tried everything. I have also pulled the Conbee 2 in and out....

## Steps to reproduce the behavior
I do not know.

## Expected behavior

## Screenshots

## Environment
 - Host system: Raspberry Pi 4 4gb
 - Running method: I am using Raspbian Buster Desktop form https://phoscon.de/en/raspbee/sdcard
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 8.12.2020
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? --> HDD

## deCONZ Logs

## Additional context

Mimiix, 2021-02-10:
16:16:13:339 APS-DATA.confirm id: 72, status: 0xE1

That's interference :) I recommend using a Extension cable.

Big4SMK, 2021-02-10:
I have the exact same issue after an update today. Same firmware and deconz verzion. Only difference is I'm running on Ubuntu in a VM instead of a rpi4. About 1 in 6 devices are connected, but I can't control them. Very annoying as all my lights are on while I want to go to sleep....

Mimiix, 2021-02-11:
@Big4SMK please open a seperate issue.

Maschler, 2021-02-11:
Hello,

this is strange because i didn't change anything and it worked great before.

I installed a backup back to the SD card yesterday. After starting the Raspberry everything worked right away. At the same time the stick was in the same position as always.

As long as this does not occur again, you can close this.

Thanks :)

Mimiix, 2021-02-11:
@Maschler you had a usb hdd instead of the SD before?

Big4SMK, 2021-02-11:
Today, I've restored from backup too. Leading me back to software 2.5.80 but still with new firmware 26660700. This got me back control of about 1/3rd of my devices, with still a lot of 0xE1 but less than before.
Then I installed a USB extension cable which got me back control of all my devices.
Finally I upgraded to 2.09.03 (somehow apt gave me a newer than the 2.07.01 I got yesterday), and things are still working fine.

Looks like the firmware might have changed something for the worse that can be mitigated by a USB extension cable. I'm all good now.

@mimiix, do you want me to log that in a separate issue?

Mimiix, 2021-02-11:
@Big4SMK I'd like to have your full issue in a seperate issue ;) 

0xE1 can pop up anytime if there's no extension cable. Could be changes in your surroundings (new wifi, changing wifi channels etc.). A restart of the network (with a firmware update or a reboot for example, where the Conbee is out of the air) causes 0xE1 easiliy as there's not connections yet. So it works, but it's crippled. 

It should be seen as a broken water-pipe. Yes, it works and you lose water. But eventually it cracks.

jonnycastaway, 2021-02-12:
Exactly this issue hits me last 3 times i updates the Conbee2 Firmware.
And i ended up 3x to pair all devices again. And also i had to correct my automations in iobroker etc.
So in my eyes it's a but because recovering an gateway backup isn't repair the situation for me.
So i'm, searching for a way to completely backup the entire system including Conbee2 Stick config so in a disaster fail i can recover everithing.

closed 2021-03-13 by github-actions[bot]