dresden-elektronik/deconz-rest-plugin #4394: No Updates from Friends of Hue Switch

BendaCoding, 2021-02-10:
## Describe the bug
Hi Everyone, im having issues pairing a Friends of Hue Switch, im following the recommendation chose: Add Switch -> Other, held down btn 1 for 10 sec. It shows "Compound was prepared" in green and the 4x Switch is available in Phoscon, however there is nothing happening when i press the switches
also in iobroker the switch shows up but, again no updates in the list
it is this device: https://www.led-trading.de/zigbee-unterputz-modul-fuer-friends-of-hue

## Steps to reproduce the behavior
  1. Go to Phoscon App - Switches - Add new Switch -> Other
  2. Hold Switch button for 10 seconds
  3. Success message pops up (Compound was prepared)
  4. Bind the switch to a lightgroup
  5. Nothing happens when pressing the switch

## Expected behavior
Switch should trigger, also state in iobroker should update

## Screenshots

## Environment
 - Host system: Raspberry Pi
 - Running method: deCONZ Add-on iobroker
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context
The switch works fine with the hue bridge

BendaCoding, 2021-02-14:
forgot to mention i followed the FAQ, so when pairing i held the button 1 as im on zigbee channel 15.
On the manual of the switch the mapping is different though:
button 1 is top right, button 2 bottom right, button 3 top left and button 4 bottom left

i tried pairing with each button, but there is never any response after the pairing :/

closed 2021-02-14 by BendaCoding

SwoopX, 2021-02-14:
Great it works now. However, is that a RTFM thing or would such a hint make sense on the webpage?