dresden-elektronik/deconz-rest-plugin #4395: Rediscover Devices - Wrong name when connection is lost

scobby, 2021-02-10:
## Describe the bug

When a sensor (mostly)  or a lamp loses connection to the gateway, and are displayed in grey i can reconnect them via "search for new sensors" and press the discovery button on the device and the connection is reestablished. But the Name is changed to the default name. Thats pretty annoying.
Can you please reassign the last name to the device.

## Steps to reproduce the behavior

1. Loose the connection to a zigbee device, must be grey in sensors. Mostly applies for sensors.
2. click search for new sensors
3. put the device into pairing mode
4. the frontends acccept the new/old sensor
5. default name is displayed, i think assigment to rules / scenes / ids / buttons are the same.

## Expected behavior

When rediscover already know sensors, use the old name. Just reconnect the device to the old already existing configuration. Just reaktivate.
## Screenshots

## Environment

 - Host system: Raspberry Pi /
 - Running method: docker 
 - Firmware version: (262F0500)
 - deCONZ version: (2.09.02 / 23.12.2020)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? 
 Hyperion via usb, usb3 ssd, video capture card

## deCONZ Logs
cant provide

## Additional context

Smanar, 2021-02-11:
Ha yes agree.
For me too, if the sensor is re-included without deleting, will be better if it keep the same name.

bsheepdiy, 2021-02-11:
I confirm, I have this bug too

majorsl, 2021-02-12:
Same happened to me after upgrading today. The only thing I noted about the lost device names was that they both had apostrophes in them eg "Steve's Closet".  Maybe that is a clue given how the ' is used in programming quotes too?

closed 2021-03-16 by github-actions[bot]

manup, 2021-03-16:
This should work now with version v2.10.3-beta.