dresden-elektronik/deconz-rest-plugin #4398: Add AMIKO SMART TEMPERATURE & HUMIDITY SENSOR to DeCONZ

Satmedia, 2021-02-11:
## Device

- Product name: AMIKO SMART TEMPERATURE & HUMIDITY SENSOR
- Manufacturer: Feibit,co.ltd
- Model identifier: FB56-THM12HM1.4
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots
[image]

[image]

[image]

[image]

[image]

https://www.amikohome.com/product/85-smart-home-temperature-humidity-sensor

Hi, I was asking to add amiko home (feibit) plugins/switches/sensors to phoscon app and I got this answer:

YKO-de commented
It looks like your devices are not yet correctly supported in deCONZ either. At least they are not listed. Therefore, please open the issue once at https://github.com/dresden-elektronik/deconz-rest-plugin.

Mimiix, 2021-02-12:
Missing the Power Configuration cluster screenshot. 

Before proceeding (and cluttering the deCONZ/phoscon repo's) please make sure to fully understand what goes where.

Satmedia, 2021-02-12:
here please

[image]

[image]

SwoopX, 2021-02-15:
@Satmedia The attributes have not been written for power configuration and basic cluster.

Satmedia, 2021-02-16:
yes, It do not read the Attributes in deCONZ when press the read button.

closed 2021-03-17 by github-actions[bot]