dresden-elektronik/deconz-rest-plugin #4400: Device name

Satmedia, 2021-02-11:
## Device

Product name: AMIKO SMART DOOR SENSOR
Manufacturer: Feibit,co.ltd
Model identifier: FB56-DOS06HM1.3
Device type : Please remove all unrelated device types.
-Sensor

## Screenshots

[image]

[image]

[image]

Mimiix, 2021-02-11:
Missing screens of other clusters.

Mimiix, 2021-02-11:
Can you add the Powerconfiguration cluster?

Satmedia, 2021-02-11:
[image]

[image]

Mimiix, 2021-02-12:
Please add them to the above post.

closed 2021-03-25 by manup