dresden-elektronik/deconz-rest-plugin #4401: Tuya Thermostat (HY368 Zigbee Radiator Actuator)

poiuztr123, 2021-02-11:
- Product name: MOES tuya Zigbee Radiator Actuator (Model: HY368)
- Manufacturer: _TZE200_ywdxldoj
- Model identifier: TS0601
- Device type : Thermostat

## Screenshots
[image]

[image]

[image]

I saw in other issues that people have been able to pair the HY368 TS0601 with deconz and phoscon. However, I can't pair it in phoscon and there is no Thermostat cluster in Deconz. I am wondering if this is because the Manufacturer identifier is slightly different (different letters behind TZE200_...) compared to the other posts.

Am I doing something wrong? Can anybody help?

Mimiix, 2021-02-11:
@Smanar Seems to be _another_ version.

Smanar, 2021-02-11:
I can say if it s evolution or mutation but tuya are like pokemons.

I have added your device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4377
In blind mode, have you a linux machine to test the code ? For the moment I just mimic the HY369, but I don't know the difference between the 2 devices.

poiuztr123, 2021-02-11:
It runs on a raspi with Raspbian. So, i guess that means that it is Linux, right? What do i have to do to test it with my HY368? Wait until it is merged in the latest beta?

easybeat, 2021-02-12:
@poiuztr123 Did you run the inclusion 2x for each device?

I had to run it always 2x just after the first one. Otherwise they would not join correctly. But I got 5 devices working. 

Issues I have with them is that they lose connection after a certain time. I'm running tests now on 2.9.2 and since 2 days they seem to be more stable.

poiuztr123, 2021-02-12:
> I have added your device here #4377
In blind mode, have you a linux machine to test the code ? For the moment I just mimic the HY369, but I don't know the difference between the 2 devices.

@Smanar:
I tried to figure out how to compile the product_trv branch as I am a github newbie.  I followed the explanation in the readme of the product_trv branch. However, the Tuya thermostat was still not recognized by phoscon. So, I checked if your changes in the source code actually made onto my raspberry by checking the source files in the deconz-rest-plugin folder onto my raspberry. E.g. in the de_web_plugin.cpp your changes with the new product code weres not included. As a work-aorund I copied the files with wget from rawgithub and was able to compile. However, before I had to resolve one error in the source code (de_web_plugin.cpp and rest_sensors.cpp) where a " ||" was missing. The other error (...epoch.addSecs) was apparently due to some other files that still had to be copied, which I did. I tried to identify other files in (https://github.com/dresden-elektronik/deconz-rest-plugin/commit/6447e42d675fc69f34288064abb765a3e3097c86) that included the epoch "thing". Not sure if you can make sense of it (Sorry, I didn't really know exactly what I was doing and I didn't have the time to dive into it. Hope it helps anyway.)

`de_web_plugin.cpp:6484:83: error: expected ‘)’ before ‘R_GetProductId’                                                                
                 R_GetProductId(&sensorNode) == QLatin1String("Tuya_THD HY368 TRV")                                                   
                                                                                   ^
                                                                                   )
                 R_GetProductId(&sensorNode) == QLatin1String("Tuya_THD HY369 TRV"))
                 ~~~~~~~~~~~~~~
de_web_plugin.cpp:6477:16: note: to match this ‘(’                                                                                    
             if (sensorNode.modelId().startsWith(QLatin1String("SLR2")) ||   // Hive
                ^
de_web_plugin.cpp: In member function ‘void DeRestPluginPrivate::updateSensorNode(const deCONZ::NodeEvent&)’:                         
de_web_plugin.cpp:9399:50: error: ‘epoch’ was not declared in this scope                                                              
                                 QDateTime time = epoch.addSecs(ia->numericValue().u32);
                                                  ^~~~~
de_web_plugin.cpp:9428:50: error: ‘epoch’ was not declared in this scope                                                              
                                 QDateTime time = epoch.addSecs(ia->numericValue().u32 - QDateTime::currentDateTime().offsetFromUtc())
;                                                                                                                                     
                                                  ^~~~~
de_web_plugin.cpp:9443:48: error: ‘epoch’ was not declared in this scope                                                              
                               QDateTime time = epoch.addSecs(ia->numericValue().u32);
                                                ^~~~~
make[1]: *** [Makefile.Release:737: release/de_web_plugin.o] Fehler 1                                                                 
make[1]: *** Es wird auf noch nicht beendete Prozesse gewartet....                                                                    
make[1]: Verzeichnis „/home/pi/deconz-rest-plugin“ wird verlassen                                                                     
make: *** [Makefile:40: release] Fehler 2                                                                                             `

At the end I was able to compile it without errors and tried it. However, when opening deconz-gui the REST-api plugin was missing. Is there a log somewhere to check why the plugin is not loading?

Smanar, 2021-02-12:
Yeah thx, have corrected the PR, and update my branch for epoch issue.

```
sudo apt install deconz-dev
git clone --branch product_trv https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But if you already have done that, you just need
```
cd deconz-rest-plugin
git pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

poiuztr123, 2021-02-12:
thanks for the quick reply and thank you for your help. With your instructions it works a lot easier.

There are still two errors. In two locations there is bracket too much or "too early".

```
de_web_plugin.cpp:6484:85: error: expected primary-expression before ‘||’ token                                                       
                 R_GetProductId(&sensorNode) == QLatin1String("Tuya_THD HY368 TRV")) ||
                                                                                     ^~

rest_sensors.cpp:1503:92: error: expected primary-expression before ‘||’ token                                                        
                             R_GetProductId(sensor) == QLatin1String("Tuya_THD HY368 TRV")) ||
                                                                                            ^~
```

After manual correction compiling went through with no errors and I was able to test the plugin.

The phoscon app found a new sensor when pairing, but it didn't show up in the sensor list afterwards. In deconz-gui there is a "Lichterkette" (I guess, light strip) now, with no Thermostat cluster. I guess something is not yet working as expected.

[image]

Smanar, 2021-02-12:
Ok, thx have corrected code too, sorry haven't the machine to test it ATM.

For the integration, it s a long sory, but you need to make 2 inclusions, without deleting the device.
- the first time deconz ask for data, and make a bad entry
- the second time deconz use the previously data to reconize the device.

But I think the device will be invisible in phoscon (I don't think phoscon support TRV)
So you need to use directly the API (I can explain how to do) or use a third application.

poiuztr123, 2021-02-12:
Thanks. I didn't recompile as I had done the changes manually.

I tried to pair the Thermostat via phoscon twice without deleting the node in deconz. But no Thermostat cluster info appeared. Do I have to go into the pair mode again on the Thermostat (option 5)?

I can use fhem (homeautomation software) to connect via deconz to devices and a ZHAThermostat "Licherkette" is showing up, but I don't know how to control it

Here are some lines from the deconz debug-view window. I took some random lines that looked interesting. It was from different pairing attempts. Hope this helps

```
23:12:50:627 [4.1] Get date code
23:12:50:628 skip check bindings for client clusters (no group)
23:12:50:628 don't create binding for attribute reporting of sensor Lichterkette
23:12:50:629 CTRL skip polling while permit join is set
23:12:50:785 APS-DATA.indication from child 0x7036
23:12:50:787 Tuya debug 1 : Time sync request
23:12:50:788 Send Tuya command 0x24, data: 00726026fd6260270b72
23:12:50:788 add task 46082 type 41 to 0x847127FFFE110A9A cluster 0xEF00 req.id 82
23:12:50:789 FP indication 0x0104 / 0xEF00 (0x847127FFFE110A9A / 0x7036)
23:12:50:790                       ...     (0x847127FFFE110A9A / 0x0DE0)
23:12:50:850 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:12:50:851 [4.1] Get date code
23:12:50:852 skip check bindings for client clusters (no group)
23:12:50:853 don't create binding for attribute reporting of sensor Lichterkette
23:12:50:853 CTRL skip polling while permit join is set
23:12:50:964 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:12:50:966 [4.1] Get date code
23:12:50:968 skip check bindings for client clusters (no group)
23:12:50:969 don't create binding for attribute reporting of sensor Lichterkette
23:12:50:970 CTRL skip polling while permit join is set
23:12:50:997 Erase task req-id: 82, type: 41 zcl seqno: 247 send time 0, profileId: 0x0104, clusterId: 0xEF00
23:12:51:075 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A

23:13:45:377 Force read attributes for ZHAThermostat SensorNode Lichterkette
23:13:45:378 don't create binding for attribute reporting of sensor Lichterkette
23:13:45:379 Force binding of attribute reporting for node Lichterkette

23:16:15:955 [4.1] Get date code
23:16:15:956 skip check bindings for client clusters (no group)
23:16:15:956 don't create binding for attribute reporting of sensor Lichterkette
23:16:15:957 CTRL skip polling while permit join is set
23:16:16:017 APS-DATA.indication from child 0x7036
23:16:16:019 Tuya debug 4 : Address 0x847127FFFE110A9A Payload 033c6602000400000005
23:16:16:020 Tuya debug 5 : Status: 3 Transid: 60 Dp: 614 (0x02,0x66) Fn: 0 Data 5
23:16:16:021 FP indication 0x0104 / 0xEF00 (0x847127FFFE110A9A / 0x7036)
23:16:16:021                       ...     (0x847127FFFE110A9A / 0x7036)
23:16:16:051 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:16:16:051 [4.1] Get date code
23:16:16:052 skip check bindings for client clusters (no group)
23:16:16:053 don't create binding for attribute reporting of sensor Lichterkette
23:16:16:053 CTRL skip polling while permit join is set
23:16:16:163 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:16:16:163 [4.1] Get date code
23:16:16:164 skip check bindings for client clusters (no group)
23:16:16:164 don't create binding for attribute reporting of sensor Lichterkette
23:16:16:165 CTRL skip polling while permit join is set
23:16:16:225 APS-DATA.indication from child 0x7036
23:16:16:227 Tuya debug 4 : Address 0x847127FFFE110A9A Payload 033d6702000400000023
23:16:16:228 Tuya debug 5 : Status: 3 Transid: 61 Dp: 615 (0x02,0x67) Fn: 0 Data 35
23:16:16:229 FP indication 0x0104 / 0xEF00 (0x847127FFFE110A9A / 0x7036)
23:16:16:229                       ...     (0x847127FFFE110A9A / 0x7036)
23:16:16:274 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:16:16:275 [4.1] Get date code
23:16:16:276 skip check bindings for client clusters (no group)
23:16:16:276 don't create binding for attribute reporting of sensor Lichterkette
23:16:16:277 CTRL skip polling while permit join is set
23:16:16:337 APS-DATA.indication from child 0x7036
23:16:16:340 Tuya debug 4 : Address 0x847127FFFE110A9A Payload 033e0202000400000096
23:16:16:340 Tuya debug 5 : Status: 3 Transid: 62 Dp: 514 (0x02,0x02) Fn: 0 Data 150
23:16:16:341 FP indication 0x0104 / 0xEF00 (0x847127FFFE110A9A / 0x7036)
23:16:16:341                       ...     (0x847127FFFE110A9A / 0x7036)
[... 137 lines cut ...]
23:20:03:873 skip check bindings for client clusters (no group)
23:20:03:874 don't create binding for attribute reporting of sensor Lichterkette
23:20:03:875 CTRL skip polling while permit join is set
23:20:04:127 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:20:04:128 [4.1] Get date code
23:20:04:129 skip check bindings for client clusters (no group)
23:20:04:129 don't create binding for attribute reporting of sensor Lichterkette
23:20:04:130 CTRL skip polling while permit join is set
23:20:04:367 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:20:04:368 [4.1] Get date code
23:20:04:368 skip check bindings for client clusters (no group)
23:20:04:369 don't create binding for attribute reporting of sensor Lichterkette
23:20:04:370 CTRL skip polling while permit join is set
23:20:04:371 enqueue event config/localtime for /config/
23:20:04:970 poll node 00:15:8d:00:03:27:7e:aa-01-0702
23:20:04:972 Poll ZHAConsumption sensor node Adapter plug SP 120 (2)
23:20:04:975 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:20:04:977 [4.1] Get date code
23:20:04:978 skip check bindings for client clusters (no group)
23:20:04:979 don't create binding for attribute reporting of sensor Lichterkette
23:20:04:980 CTRL skip polling while permit join is set
23:20:04:983 MAC poll fastEnddeviceProbe() 0x847127FFFE110A9A
23:20:04:984 [4.1] Get date code

```

Smanar, 2021-02-13:
>I tried to pair the Thermostat via phoscon twice without deleting the node in deconz. But no Thermostat cluster info appeared. Do I have to go into the pair mode again on the Thermostat (option 5)?

Tuya is a little "exotic" no thermostat cluster ^^, all is managed by the tuya cluster.

>I can use fhem (homeautomation software) to connect via deconz to devices and a ZHAThermostat "Licherkette" is showing up, but I don't know how to control it. Would a REST-API client liek RESTED help here? I have set it up and can see teh Thermostat there under /sensors but don't know what to do then.

It s perfect, but I don't know FHEM, if you can see the "ZHAThermostat" it mean the inclusion is good, else you will see a plug.
FHEM don't support this kind of device ? at least temperature and heatsetpoint ?

>Would a REST-API client liek RESTED help here?

Sure but will be difficult for you, if you have FHEM can you take a look on deconz integration ? I can explain wich one API command to use. Will be better for you if you can use them using FHEM.

poiuztr123, 2021-02-13:
With the deconz integration in fhem I was able to set the heatsetpoint on the thermostat. I have used a configuration in fhem from a Eurotronic Spirit Zigbee thermostat:

`/heatsetpoint (.*)/:perl:{'{"heatsetpoint":' . $VALUE1 * 100 . '}'}`

 I don’t get any readings like temperature, valve, battery though. Also heatsetpoint is not updated in fhem.

I need to dive a little bit deeper into fhem. Is there a list of config commands and readings that the tuya thermostat supports? If you point me into the right direction on how it would work with Rest-API I can then apply it to fhem as well.

Smanar, 2021-02-13:
Sure here you have a JSON exemple

```
"2":{
   "config":{
      "heatsetpoint":2200,
      "locked":true,
      "mode":"auto",
      "offset":50,
      "on":true,
      "preset":null,
      "reachable":false,
      "schedule_on":null,
      "setvalve":null,
      "windowopen_set":null
   },
   "ep":1,
   "etag":"73f0c1567b0dd3f1ce08a39f0e9995cb",
   "lastseen":"2020-10-31T16:56Z",
   "manufacturername":"_TZE200_aoclfnxz",
   "modelid":"TS0601",
   "name":"TS0601",
   "state":{
      "lastupdated":"none",
      "on":null,
      "temperature":1800
   },
   "type":"ZHAThermostat",
   "uniqueid":"60:a4:23:ff:fe:9f:4f:a7-01-0201"
}
```

Your device probably don't support all of them, idk if you can see the device JSON in FHEM ? Else this one is visible in the API
You have the temperature in state/temperature.
And your device need to use "mode" or "preset" but not both, to be sure check the device JSON.

If you haven't it in FHEM you can find it at
`http://IP:PORT/api/KEY/sensors`

IP and PORT are the same used for phoscon
KEY is an api key, you can use the same used by FHEM.

poiuztr123, 2021-02-13:
Using a REST client I got this for two thermostats. I assume that I have to fid out from the fhem documentation or ask in the fhem forum how to get the readings into fhem automatically. Thanks again.

```
 "81": {
    "config": {
      "heatsetpoint": 1800,
      "locked": false,
      "mode": null,
      "offset": -100,
      "on": true,
      "preset": "manual",
      "reachable": true,
      "schedule": {},
      "schedule_on": false,
      "setvalve": false,
      "windowopen_set": false
    },
    "ep": 1,
    "etag": "9469bdada24c5ec31f65ea89902524ce",
    "lastseen": "2021-02-13T20:25Z",
    "manufacturername": "_TZE200_ywdxldoj",
    "modelid": "TS0601",
    "name": "Buero-T-klein",
    "state": {
      "lastupdated": "none",
      "lowbattery": false,
      "on": true,
      "temperature": 1850,
      "valve": 25
    },
    "type": "ZHAThermostat",
    "uniqueid": "84:71:27:ff:fe:11:0a:9a-01-0201"
  },

"83": {
    "config": {
      "heatsetpoint": 2100,
      "locked": false,
      "mode": "auto",
      "offset": -100,
      "on": true,
      "preset": "manual",
      "reachable": true,
      "schedule": {},
      "schedule_on": false,
      "setvalve": false,
      "windowopen_set": false
    },
    "ep": 1,
    "etag": "68221fa160b01f2f84d1a19ef0b1c356",
    "lastseen": "2021-02-13T16:46Z",
    "manufacturername": "_TZE200_ywdxldoj",
    "modelid": "TS0601",
    "name": "Schlafzimmer-T",
    "state": {
      "lastupdated": "none",
      "lowbattery": false,
      "on": true,
      "temperature": 2000,
      "valve": 100
    },
    "type": "ZHAThermostat",
    "uniqueid": "84:71:27:ff:fe:10:03:d4-01-0201"
  },
```

Smanar, 2021-02-13:
I m seing your device have both "mode" and "preset", I don't think it will support both, generally tuya TRV support only "preset"
Just tell me if I need to disable one of them.

poiuztr123, 2021-02-14:
Hi Smanar,

I don't know why one of the two devices has got a value under "mode" and the other doesn't. I also don't know the implications of both "mode" and "preset" being shown. Is there a more detailed description fo all of the functions, readings and configuration settings?
How do I disable one of the two?

Smanar, 2021-02-15:
>How do I disable one of the two?

I will do, I just need to know wich one is working or not on your device

"mode" can be "auto" (normal working mode) "heat" (force heat) "off"
"preset" can be holiday/auto/manual/confort/eco/boost/complex, but not all feature can be supported too, at least auto/manual I think
It s indicated too on your device manual.

poiuztr123, 2021-02-15:
I was able to set the "preset" field and saw a result on the Thermostat for:

- auto
- manual
- holiday
- boost

- eco, comfort, complex: not sure, if it did something

I didn't try the "mode" field as "preset" seemed to be more comprehensive.

Is there a documentation somewhere about all the different fields and configuration options?

Smanar, 2021-02-15:
You have that https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#supported-config-attributes_1

But it s realy device dependent. But as "preset" seem work, I don't think "mode" will work.
I have added those command from the documentation, and they are "global" not all can work  for all devices.
The better way is looking the device manual, if you have one, and not in chinese.

easybeat, 2021-02-16:
@Smanar Not sure if it is ok to put this here but I've to another TRV which looks like Tuya:

[image]

It is also not updating lastupdated and only shows 2 Clusters.

Should I open a new ticket for this?

Kind regards

poiuztr123, 2021-02-16:
> You have that https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#supported-config-attributes_1
> 
> But it s realy device dependent. As "preset" seem work, I don't think "mode" will work ?
> I have added those command from the documentation, and they are "global" it s realy generic, not all can work for all devices.
> The better way is looking the device manual, if you have one, and not in chinese.
> 
> "Preset" is generaly for thermostat.
> "Mode" is generally for TRV.

Thanks for the link. I wasn't aware of this list.

I checked if "mode" is working on the devices as well and it does.

- "off" puts the device in off mode (OF shown on the display) and closes the valve
- "auto" turns the device on again and in normal thermostat mode that allows to set a temperature or use the automatic schedule etc
- "on" turns the device apparently in on mode (ON shown on the display) and probably opens the valve 100% (I couldn't really check)

It seems that when the device is in "auto" mode one can use preset to determine what exactly it is supposed to do. I would assume that "auto" in preset (in addition to "auto" in mode) uses the schedule  nd changes the heatsetpoint according to a schedule (6 different auto mode periods) that can be set in the device. I have attached the manual. The different presets that can be set manually on the device with the buttons and they are mentioned throughout the manual. But it seems that nor all general presets are implemented, i.e. eco, comfort, complex seem to be missing.

[image]

[image]

[image]

[image]

Smanar, 2021-02-16:
> @Smanar Not sure if it is ok to put this here but I've to another TRV which looks like Tuya:
> 
> [image]
> 
> It is also not updating lastupdated and only shows 2 Clusters.

For the lastupdated issue, the PR is not yt validated, it s this one for the tuya siren
Only 2 clusters can be normal, tuya hide sometime the tuya cluster.

But the device is working or not ?

If not and if you are using the last version (2.10.0), there is already an issue for that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4417

>"on" turns the device apparently in on mode (ON shown on the display) and probably opens the valve 100% (I couldn't really check)

Huuu ? there is no "on" in code, idk what happened ^^

But yes, I can't explain why there is 2 "auto" (and both are working to help), perhaps the "auto" for "mode" is just a "on". 

But an easy test you can do, read the manuel for the feature you want, use the device according to it, wait some minuts (this kind of device are lazy) and check the modified field in the JSON, the next time you can do the reverse.

poiuztr123, 2021-02-16:
I meant “heat”, not “on”. Have corrected it above.

Yes, I think “auto” for “mode” is device on and the “preset”determines in which operating mode the thermostat should be.

poiuztr123, 2021-02-17:
After having used the thermostat for a few days I have a different question related to the zigbee connection, pairing and the use of routers in the zigbee network.

I have used the thermostat on the same floor as the raspbee/conbee over a few day and after everything seemed to work fine without lost connections I have tried to move one of the thermostats down to the basement to check if they work over long distances as well. If this would have been successful I would have ordered more to retrofit the entire house.

However, I couldn't access the thermostat anymore and in deconz-gui I saw that it had lost connection (red dot) and it also didn't try to use a router. I tried to repair it in its final location in the basement, but that didn't work.

Does anybody have experience with the HY368 Tuya thermostats when they are used via routers, so over long distances? What is the correct procedure? Removing from the zigbee network and then repairing in the final location? Or pairing close to the conbee and then moving to the final location? It doesn't seem to find a new route or select a new router. Is there a way to facilitate that?

easybeat, 2021-02-17:
@poiuztr123 Hi
I have 5 Moes TRV in action and I did face a lot of connection problems. Since I've updated deconz to version 2.9.3 these connection problems are gone and all my devices are online since more than a week.

Which deconz version are you using?

poiuztr123, 2021-02-17:
I am using  2.9.3. I have compiled the Rest-plugin with the changes made by Smanar and the instructions above regarding the actual commands to compile it on the raspbee using his branch 
[image]

Do you use it across different floors? And does deconz show you that some TRV's go through different routers? I have innr SP120 and hoped that they would relay the messages to the TRV in the basement as they do with a bunch of Xiaomi devices.

easybeat, 2021-02-17:
Yes I use them on different floors. The ones on the same floor as the deconz box work fine. Upstairs I had to install Ikea range extender. Before the update they did only connect to them but now I have 2 which also connected to a light and using this as router...

Smanar, 2021-02-17:
>What is the correct procedure? Removing from the zigbee network and then repairing in the final location? Or pairing close to the conbee and then moving to the final location? It doesn't seem to find a new route or select a new router. Is there a way to facilitate that?

Good question, correct device rebuild the mesh themself automatically, bad one like Xiaomi never.
So it s better to make the inclusion at definitive place.

poiuztr123, 2021-02-18:
When using the device a new issue came up.

I didn't get readings like temperature and valve in fhem and I tracked down the issue. In Json the "lastupdated" field is always empty and in fhem it is apparently required to trigger the evaluation of the state which includes the temperature and valve to give me the reading in fhem.

Is this related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4147 ? Is this resolved? Or is there anything that I can do to test it?

Smanar, 2021-02-18:
> in fhem the lastupdated filed is required to trigger the evaluation of the state which includes the temperature and valve to give me the reading for thosw two values in fhem

You have this information from someone from FHEM ? Because deconz send websocket event itself when the temperature change, and all third app use that.

poiuztr123, 2021-02-18:
yes, one of the developer pointed me to the part of the code where fhem evaluates lastupdated. If it is empty or not changed then the code doesn't evaluate the variables (e.g. temperature, valve) in state. Is there a way to make lastupdated work?

Is this related to lastseen and reachable discussion? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2590

Smanar, 2021-02-18:
More here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4147.
Was corrected in this PR, if I m right > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4379

I can explain how to compile it if you want to test it  ?

poiuztr123, 2021-02-18:
yes, I would like to compile and test it. How do I do it?

Smanar, 2021-02-18:
You just need a linux machine, with deconz installed.

The procedure is explained here 
 https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch tuya_siren_clean https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you are using your production machine you can make a backup of the file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder (renaming is not enought)

poiuztr123, 2021-02-18:
Thank you. I have tried it and it seems to work with the two HY368 Tuya thermostats! Lastupdated has a value now and can be used in fhem. I will monitor over the next few days to see if everything else is fine.

closed 2021-03-03 by manup

h3llh0und, 2021-03-13:
Hi, i have the same Thermostat and just want to know if the code is already in the newest beta Version or is it still needed to build it by myself? I cant see the Device in Iobroker. :(

Smanar, 2021-03-13:
I think he is inside now, what is your deconz version and the device manufacture name ?
Remember you need 2 inclusions without deleting the previous one.

h3llh0und, 2021-03-13:
If just installed latest Beta. With 2 inclusions you mean starting 2 times Pairing mode on deconz and Thermostat?
[image]

[image]

Smanar, 2021-03-13:
Ok, you don't need search more ^^
You have a new device, not in code yet. Do you have a product name, or a similar tuya device you know to mimic the code ?
Can you share too a node picture ?

h3llh0und, 2021-03-13:
Ok, i thought it was the same as the Model Identifier is the same and Manufacturer Name starts with the same.
[image]

Smanar, 2021-03-13:
Can you show a node picture, to see cluster and if the device is a router or and end device pls ?

h3llh0und, 2021-03-13:
you mean this?
[image]

Smanar, 2021-03-13:
Ok so have made a PR for it, but this device is relay new, so need to test it.
For the moment I have mimic the one on this issue, but thoses device are realy not the same.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch trv_again https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

h3llh0und, 2021-03-13:
Thx, you'r awsome :)
[image]
Now it has a name i can see it in iobroker. Now i just have to find out how i can change temprature ;)
[image]

Smanar, 2021-03-13:
There is state/heatsetpoint.
But the more unknown if the "mode" and "preset", some device support both, other only "mode" and other again only "preset"
if you can say me the one I need to disable ?

mode can be "on" "off" "auto" "heat" (not all will work)
preset can be
```
                        if (data == 0) { preset = QLatin1String("holiday"); }
                        else if (data == 1) { preset = QLatin1String("auto"); }
                        else if (data == 2) { preset = QLatin1String("manual"); }
                        else if (data == 3) { preset = QLatin1String("comfort"); }
                        else if (data == 4) { preset = QLatin1String("eco"); }
                        else if (data == 5) { preset = QLatin1String("boost"); }
                        else if (data == 6) { preset = QLatin1String("complex"); }
```

h3llh0und, 2021-03-13:
I will Test tomorrow. Do i need to set present to the string auto and so on or the data value in the if statement?
Do you know if I should see anything on the thermostat when I change values?

Smanar, 2021-03-13:
By defaut I think the device is "auto" mode, so just need to change setheatpoint.

And yes, if you change the value on the TRV it will be visible on deconz, and vice versa, but can take time.
And better all tuya request are visible on log (deconz/help/log) for in and out, so if a command is not working, you can just do it manualy on the device, and give me the log to mimic the request.

h3llh0und, 2021-03-14:
So i tried it this morning, i placed the Thermostat at the heater and started it.
I tried to change the Temprature at the Termostat and i see this in Log:
11:42:35:517 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00696902000400000023
11:42:35:518 Tuya debug 5 : Status: 0 Transid: 105 Dp: 617 (0x02,0x69) Fn: 0 Data 35
But i dont see any changes in iobroker about the new temperature. The lastupdated time didnt get updated also.

When i change temperature in iobroker i see this:
11:45:57:697 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x02, data: 000000b4
11:45:58:463 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00020204000100
11:45:58:464 Tuya debug 5 : Status: 0 Transid: 2 Dp: 1026 (0x04,0x02) Fn: 0 Data 0
But i dont see anything on the Thermostat. And lastupdated is also not updated.

When i set mode to off i hear the Thermostat doing something(sounds like closing) and i see this in log:
11:49:59:668 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x6A, data: 02
11:50:02:805 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 006a6a01000101
11:50:02:806 Tuya debug 5 : Status: 0 Transid: 106 Dp: 362 (0x01,0x6A) Fn: 0 Data 1
After that i have a number around 6850 visible on Thermostat that is counting down and i see around every 10 seconds this in log:
1:50:25:905 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00767602000400000377
11:50:25:906 Tuya debug 5 : Status: 0 Transid: 118 Dp: 630 (0x02,0x76) Fn: 0 Data 887
Where the value in Data and Payload is decreased by 1 with every log message.

When i send auto i hear the Thermostat doing something(sounds like opening) the countdown stops but i still receive that messages around every 10 seconds.
11:53:49:908 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x6A, data: 00
11:53:56:463 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00767602000400000363
11:53:56:464 Tuya debug 5 : Status: 0 Transid: 118 Dp: 630 (0x02,0x76) Fn: 0 Data 867

I hope you have any idea what happens :)

h3llh0und, 2021-03-14:
And when i change Present, i dont see anythin in log.

Smanar, 2021-03-14:
Have modified the code
- removed unused field (like preset), you will need at least to restart deconz (perhaps remake the inclusion)
- adding the mode, off/o/auto
- adding the heatsetpoint
- - adding childlock

Now, some explanations, it seem your device have a strange working mode, the normal mode is "auto" but you can have a mode "heat", in this mode you can have the time counter and new heatsetpoint values.

The code is not tested on my side, so you can have error during compilation.

For the moment avoid chaging value in "heat" mode, idk how the code will react

h3llh0und, 2021-03-14:
There are some Compile Errors and a warning that appeared several times:
tuya.h:46: warning: "DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_2" redefined
 #define DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_2 0x1E
 
tuya.h:45: note: this is the location of the previous definition
 #define DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_2 0x28
 
rest_sensors.cpp: In member function ‘int DeRestPluginPrivate::changeSensorConfig(const ApiRequest&, ApiResponse&)’:
rest_sensors.cpp:1398:101: error: expected ‘;’ before ‘)’ token
               sendTuyaRequest(task, TaskThermostat , DP_TYPE_ENUM, 0x02, data))
                                                                               ^
                                                                               ;
rest_sensors.cpp:1403:97: error: expected ‘;’ before ‘)’ token
               sendTuyaRequest(task, TaskThermostat , DP_TYPE_ENUM, 0x6a, data))
                                                                               ^
                                                                               ;
rest_sensors.cpp:1718:38: error: ‘DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_3’ was not declared in this scope
                                 dp = DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_3;
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rest_sensors.cpp:1718:38: note: suggested alternative: ‘DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_1’
                                 dp = DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_3;
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                      DP_IDENTIFIER_THERMOSTAT_CHILDLOCK_1
make[1]: *** [Makefile.Release:1219: release/rest_sensors.o] Fehler 1

Smanar, 2021-03-14:
There was so much problem ^^, but it s ok now compiling on my side

h3llh0und, 2021-03-14:
Ok, now some Attributes are gone.
Now i see a lot of these in Log for the first Minutes after i reconnected: 
15:35:57:486 Tuya debug 1 : Time sync request
15:35:57:488 Send Tuya command 0x24, data: 0000604e2d5d604e2d5d

I tried to change temperature but nothin happens on Thermostat:
15:36:06:282 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x10, data: 00000190
15:36:07:981 Tuya debug 1 : Time sync request
15:36:07:982 Send Tuya command 0x24, data: 0000604e2d67604e2d67
15:36:07:984 delay sending request 204 dt 1 ms to 0x847127FFFE9642E3, ep: 0x01 cluster: 0xEF00 onAir: 1
15:36:08:002 delay sending request 204 dt 1 ms to 0x847127FFFE9642E3, ep: 0x01 cluster: 0xEF00 onAir: 1
15:36:08:098 delay sending request 204 dt 2 ms to 0x847127FFFE9642E3, ep: 0x01 cluster: 0xEF00 onAir: 1
15:36:08:198 delay sending request 204 dt 2 ms to 0x847127FFFE9642E3, ep: 0x01 cluster: 0xEF00 onAir: 1
15:36:13:263 Tuya debug 1 : Time sync request
15:36:13:264 Send Tuya command 0x24, data: 0015604e2d6d604e2d6d
15:36:18:519 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00222202000400000092
15:36:18:520 Tuya debug 5 : Status: 0 Transid: 34 Dp: 546 (0x02,0x22) Fn: 0 Data 146

And when i change temperature on Thermostat nothing happens to level.temperature value.
I changed on Thermostat from 20° to 21° and i saw some entries in Log:
15:49:41:986 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00222202000400000095
15:49:41:986 Tuya debug 5 : Status: 0 Transid: 34 Dp: 546 (0x02,0x22) Fn: 0 Data 149
15:49:52:470 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 0069690200040000002a
15:49:52:471 Tuya debug 5 : Status: 0 Transid: 105 Dp: 617 (0x02,0x69) Fn: 0 Data 42
15:50:02:990 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00222202000400000096
15:50:02:991 Tuya debug 5 : Status: 0 Transid: 34 Dp: 546 (0x02,0x22) Fn: 0 Data 150
But i am not 100% sure if these are all Messages for that change.

I changed mode to manual on thermostat than i had a heat in the status field.
When i changed it back to auto in iobroker thermostat showed auto. So this is working :)

I changed to heat in iobroker this happens in log:
15:52:01:783 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x02, data: 01
15:52:05:001 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00020204000101
15:52:05:002 Tuya debug 5 : Status: 0 Transid: 2 Dp: 1026 (0x04,0x02) Fn: 0 Data 1
And Thermostat switched to manual Mode.

And this is the log for switching back to auto
15:53:25:293 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x02, data: 00
15:53:26:438 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00020204000100
15:53:26:439 Tuya debug 5 : Status: 0 Transid: 2 Dp: 1026 (0x04,0x02) Fn: 0 Data 0

Smanar, 2021-03-14:
>15:35:57:486 Tuya debug 1 : Time sync request
15:35:57:488 Send Tuya command 0x24, data: 0000604e2d5d604e2d5d

This is used if your device have time displayed

>15:49:41:986 Tuya debug 5 : Status: 0 Transid: 34 Dp: 546 (0x02,0x22) Fn: 0 Data 149

No information about this command, but not used on Z2M

>15:49:52:471 Tuya debug 5 : Status: 0 Transid: 105 Dp: 617 (0x02,0x69) Fn: 0 Data 42
> changed on Thermostat from 20° to 21° 

I think I have reversed the 2 mode, from that I m reading the device use
- (0x02,0x69) for auto mode
- (0x02,0x10) for heat mode

So have reversed both , and make the code compatible with both command, if I m still wrong you can reverse it in file rest_sensor.cpp around line 1280

New code online

h3llh0und, 2021-03-14:
So i just played a bit around and had a look in the code. (What a loooong function :D)

You had a little error in line 1280, you used sensorNode instead of sensor.

There is something wrong with the calculation of the temperature i think. When i change it to 20° the display shows me 72°, 21° is 82°.
Log for 20:
18:20:34:980 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x69, data: 00000190
18:20:36:044 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 00181802000400000104
18:20:36:045 Tuya debug 5 : Status: 0 Transid: 24 Dp: 536 (0x02,0x18) Fn: 0 Data 260
21:
18:21:20:300 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x69, data: 000001a4
18:21:23:281 Tuya debug 4 : Address 0x847127FFFE9642E3 Payload 006969020004000000a4
18:21:23:282 Tuya debug 5 : Status: 0 Transid: 105 Dp: 617 (0x02,0x69) Fn: 0 Data 164

I saw that you divide by 10 in other Thermostats so i tried this out but then i get for 20° ~5°.
It is normal that heatsetpoint for 20°  is a value of 200?
My last programming in c++ was a long time ago, so i hope you have an idea.

Smanar, 2021-03-14:
Yep, probably a convertion error between all tuya mode.

Have corrected the code (I hope)

h3llh0und, 2021-03-14:
Nice :) Now it looks good :)
Okay * 2 / 10 :D 

Now its in sync. And if i change temperature at thermostat its synced back and i see the right Value. Then i can start playing around tomorrow with my window and temperature sensor :)

Smanar, 2021-03-20:
@h3llh0und nothing special ? can make the PR ?

h3llh0und, 2021-03-20:
@Smanar No, everything working fine :) So you can do a PR. 👍

KMDobrev, 2021-07-29:
Hello, I have HY 368 Radiator Actuators from Moes brand.  I use them on Home Assistant with Conbee 2.  But there are no "Valve Settings" and "Keylock" options.  Is there any way to add them?

Smanar, 2021-07-30:
Hello, not all device are same, even they are same model.
Do you have the manufacture name of your device ? Or better direclty the JSON.

[8 more replies]