dresden-elektronik/deconz-rest-plugin #4402: Is my new Conbee 2 brocken?

eic1cr, 2021-02-11:
## Describe the question or issue you are having
I have a new conbee 2 stick. 
I tried to get started with a raspberry 4B, but it can not find the stick. 
So i tried it on Windows 10 but it failed also. 
I use an 2m USB2.0 extention cable which is plugged in the USB2.0 Port of my raspberry.
the raspberry sd card was flashed with the phoscon sd card image to get shure deCONZ is latest version.
I have tried to upgrade the firmware with this manual:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually
But it failed.
I have also tried to find the problem with this manual with no positiv result:
https://phoscon.de/de/support#conbee2-connection-issues
I´ve checket my user pi to be in the dialout group.
I also checked with a normal USB stick that there is no issue with the extention cable, the dialout group, the usb port of the raspberry an so on... this worked fine.

In my opinion the conbee 2 is broken, but how can i be shure?
Can i do some tests for this?

I´m affraid of bying an new one and have the same issues.
To rule that out, i want to know if i did something wrong by getting started conbee 2?

## Screenshots

[image]
[image]
CM0 -f deCONZ_ConBeeII_0x26680700.bin.GCF -x 3

17:41:10:365 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:10:416 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:10:416 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
17:41:10:416 device /dev/ttyACM0 not found
17:41:10:416 retry, elapsed 40 seconds
17:41:10:522 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26680700.bin.GCF -x 3

17:41:10:522 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:10:603 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:10:603 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
17:41:10:603 device /dev/ttyACM0 not found
17:41:10:604 retry, elapsed 40 seconds
17:41:10:699 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26680700.bin.GCF -x 3

17:41:10:699 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:10:757 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:10:758 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
17:41:10:759 device /dev/ttyACM0 not found
17:41:10:759 retry, elapsed 40 seconds
17:41:10:858 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26680700.bin.GCF -x 3

17:41:10:859 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:10:913 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:10:913 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
17:41:10:914 device /dev/ttyACM0 not found
17:41:10:918 retry, elapsed 40 seconds
17:41:11:023 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26680700.bin.GCF -x 3

17:41:11:023 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:11:075 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:11:075 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
17:41:11:075 device /dev/ttyACM0 not found
17:41:11:075 retry, elapsed 40 seconds
17:41:11:170 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26680700.bin.GCF -x 3

17:41:11:171 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:11:226 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:11:226 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
17:41:11:226 device /dev/ttyACM0 not found
17:41:11:226 retry, elapsed 41 seconds
17:41:11:331 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26680700.bin.GCF -x 3

17:41:11:332 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:11:415 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
[... 823 lines cut ...]

17:41:30:147 using firmware file: deCONZ_ConBeeII_0x26680700.bin.GCF 
17:41:30:199 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:30:199 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
17:41:30:199 device /dev/ttyACM0 not found
retry, failed
pi@phoscon:~ $ 

[image]

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian (later i whant to run it in Home Assistent deCONZ Add-on, but to get it run i tried to use the phoscon sd card image, to be sure everything is correct)
 - Firmware version: Not connected
 - deCONZ version: 2.09.02 / 23.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
The deCONZ GUI doesn´t start if i try this in Raspbian / programming / denconz. How can i start deCONZ GUI?

## Additional context
 A single house. My router is a telekom speedport.

Mimiix, 2021-02-11:
What does the stick do in a Windows machine?

eic1cr, 2021-02-11:
Nothing... When I plug it into my Laptop with windows 10 nothing happens.
The stick does not appear in the device manager. 
I tried to install firmware also on windows, but i didn´t make a screenshot.

eic1cr, 2021-02-11:
I just tried it on windows again: 
this is the result:
[image]

Smanar, 2021-02-11:
I don't understand why you have not the same result on raspian ...
You can try 4/5 time, the classic issue is usb disconnection so you nee to run the command at good moment ^^.

Even the result of lusb is not encouraging for me.

eic1cr, 2021-02-11:
I tried it many times. It fails every time 👎 
So do you also think the conbee 2 is broken?
Should I bought an new conbee 2 and send this one back or do you think there is another chance to get it started?
Is there anything i can try?
Here again on raspbian:
[image]

Smanar, 2021-02-12:
I don't work for their support, there is an email for that, it s depend of re-sellers
But no, they will probably purpose an echange, so don't buy yet a new one.

eic1cr, 2021-02-12:
Hi, thanks for your answer. 
support@dresden-elektronik.de told me to update the firmware and to use an extension cable. After this failed, I asked what else I can do, but I didn´t get an answer yet.
Today I ordered an new conbee 2. After testing the new one, which hopefully runs, i will know if the first one is broken. Then it goes back to the seller.
I will share the result next week and hopefully close this issue then.
Have a nice weekend.

Mimiix, 2021-02-15:
@eic1cr It could very much be a DOA.

eic1cr, 2021-02-19:
Hi all,
the new conbee 2 works fine. The broken one was already send back to the seller.
I´ve already implemented the new conbee 2 in home assistant and updated it to the latest firmware.
Works also fine :-)
So this issue is closed.
Thanks all for your help.

closed 2021-02-19 by eic1cr

Mimiix, 2021-02-24:
Great to hear that!