dresden-elektronik/deconz-rest-plugin #4403: Support for Orvibo Contact Sensor -  Model no SM11

ashar38, 2021-02-11:
Hello team,

Going through the old issues I noticed there was a thread open to add support for Orvibo door/window sensors, but looks like that thread went cold. Hence, I am starting a new issue hoping that support could be added for this device in one of the future firmwares.
I have tried to add as much info as possible from what I see through Deconz. But please let me know if I can add some more.
## Device

- Product name: LivingWise / Orvibo Contact Sensor
- Manufacturer: ORVIBO
- Model identifier: e70f96b3773a4c9283c6862dbafb6a99
- Device type : Sensor

## Screenshots

  Required screenshots:
Node Info.
[image]

Clusters:
 Basic:
[image]

Identity:
[image]

IAS Zone:
[image]

IAZ Zone: cont'd
[image]

Power Config:
[image]

SwoopX, 2021-02-22:
When support has been added, you need to enter sensor search in Phoscon and pair the device. However, you also need to keep an eye on deconz GUI for the device to show up, as the manufacturer name and model ID are not read automatically. When it appears in deconz GUI, you need to read the basic cluster's attributes while sensor search is still running (which is the case even when Phoscon already issued the greenish success message) for the sensor resource to be created.

ashar38, 2021-03-01:
> When support has been added, you need to enter sensor search in Phoscon and pair the device. However, you also need to keep an eye on deconz GUI for the device to show up, as the manufacturer name [...]

Thank you. Much appreciated. I will wait for the next release and confirm if everything works well.

closed 2021-03-27 by manup