dresden-elektronik/deconz-rest-plugin #4404: Philips Hue Dimmer switch (new one)

rkeet, 2021-02-11:
## Device

- Product name:  Philips Hue Dimmer switch ([new version](https://www.philips-hue.com/nl-nl/p/hue-dimmer-switch--nieuwste-model-/8719514274617#specifications))
- Manufacturer: Signify (I think)
- Model identifier: Don't know
- Device type : Please remove all unrelated device types. 
  - Switch
  - Other: Dimmer ?  

I don't mind helping out more. I've literally started with Home Assistant last week, so I'm new to all this home automation stuff. 

What I can say is that this device is not showing up in the Phoscon Web App when trying to add a new switch. Following the steps, selecting Philips and then factory resetting the remote blinks a green light on the remote, but it does not stay on. 

On the page asking for a reset of the remote, the "Ready" button also never enables. 

[image]

Figured I'd open a request for support. Let me know if I can help out providing data.

closed 2021-02-11 by close-issue-app[bot]

rkeet, 2021-02-11:
Bruh

Mimiix, 2021-02-11:
@rkeet if you use search, you notice there's already a request. So even if you did stick to the template, I'd closed it.