dresden-elektronik/deconz-rest-plugin #4406: Conbee2 can't join after changing USB port.

NathanSweet, 2021-02-12:
## Describe the question or issue you are having

I am running DeCONZ on Windows. Everything works fine, except I need to change the USB port the Conbee2 is plugged into. I do that (tried many other ports) and DeCONZ still sees the Conbee2, but can't join the network. If I put it back on the old USB port, it works again.

## Screenshots

## Environment

 - Host system: PC
 - Running method: Windows
 - Firmware version: 264A0700
 - deCONZ version: 2.05.71
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? Yes. If so: keyboard, mouse, webcam.

## deCONZ Logs

I would love to see logs. How do I enable logging on Windows? `--dbg-info=2` etc does nothing, there is no console output.

No information is provided about what is going on, it's frustrating.

## Additional context

I'd rather not update unless sure it will fix this problem, as in the past I've lost my entire setup. Since that I'm scared to touch it if it's working.

Thanks for your help.

Mimiix, 2021-02-12:
It would be nice to see some logs :) However, you need to be on a newer version for that:
 
To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

As you are not using a extension cable, that might be the case here. Is your mouse/keyboard wireless?

NathanSweet, 2021-02-12:
Alright, if it's gotta be updated, here we go... updated to `2.09.03`. Sweet sweet logs:
```
11:06:41:032 start reconnect to network
11:06:46:030 try to reconnect to network try=1
11:06:46:031 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
11:06:46:037 Device protocol version: 0x010B
11:06:46:093 Current channel 15
11:06:46:105 CTRL got nwk update id 2
11:06:46:109 CTRL ANT_CTRL 0x03
11:06:46:133 Device protocol version: 0x010B
11:06:46:188 Current channel 15
11:06:46:200 CTRL got nwk update id 2
11:06:46:204 CTRL ANT_CTRL 0x03
11:06:46:227 NET ZDO start network status 0xC4
11:06:51:030 try to reconnect to network try=2
11:06:51:059 Device protocol version: 0x010B
11:06:51:113 Current channel 15
11:06:51:126 CTRL got nwk update id 2
11:06:51:130 CTRL ANT_CTRL 0x03
11:06:51:153 Device protocol version: 0x010B
11:06:51:207 Current channel 15
11:06:51:219 CTRL got nwk update id 2
11:06:51:223 CTRL ANT_CTRL 0x03
11:06:51:247 NET ZDO start network status 0xC4
11:06:52:423 New websocket 169.254.219.142:4117 (state: 3) 
11:06:53:030 Skip idle timer callback, too early: elapsed 605 msec
11:06:56:031 try to reconnect to network try=3
11:06:56:032 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
11:06:56:036 Device protocol version: 0x010B
11:06:56:091 Current channel 15
11:06:56:103 CTRL got nwk update id 2
11:06:56:107 CTRL ANT_CTRL 0x03
11:06:56:131 Device protocol version: 0x010B
11:06:56:185 Current channel 15
11:06:56:197 CTRL got nwk update id 2
11:06:56:201 CTRL ANT_CTRL 0x03
11:06:56:225 NET ZDO start network status 0xC4
11:07:01:029 try to reconnect to network try=4
...
```
Not sure why it looks for a firmware update file, or if that is a problem.

Thanks for the help!

NathanSweet, 2021-02-12:
I flashed `deCONZ_ConBeeII_0x264a0700.bin.GCF` manually, but I get the same behavior and same log saying `GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF`.

Mimiix, 2021-02-12:
That Firmware files are shiped with new versions. 
You need to use a extension cable here. That probably is your issue.

NathanSweet, 2021-02-14:
I used a USB hub to move it farther away and you're right, it works like that. Crazy it worked on just one USB port and not the others, but OK! Thanks for your help.

closed 2021-02-14 by NathanSweet