dresden-elektronik/deconz-rest-plugin #4407: Philips hue dimmer switch events lost or delayed

peterjuras, 2021-02-12:
## Describe the bug

Sometimes when pressing buttons on Hue dimmer switches (RWL021), the single events like 1000, 1002 etc. appear to be delayed or sometimes completely lost or missing.

As an example that can be seen in the log messages below, a short press (and release) of the hue dimmer at `09:12:04:920` with the hue button event 1000 was not followed by a short release event 1002. This leads to my automations not firing, since I have light turn on animations that in home assistant that are triggered on release, not already on press.

Similarly, another short press and release at `09:12:11:514` was only followed by the corresponding 1002 event 855 milliseconds later, which leads to a humanly noticable delay. Typically the delay is within 100 or 200 milliseconds and not noticable.

I already tried switching Zigbee channels and moving the ConBee stick up/down, further away from the raspberry, closer to the raspberry but I'm starting to run out of ideas. I also don't appear to be the only one that is having issue, since I've seen others reporting similar issues (e.g. #3707, #3911, #3998).

I have 18 hue light bulbs (various types, always powered on) and 6 hue dimmer switches. All nodes and connections are "green" in the deconz UI and none of the lights or switches are usually unavailable. However, the connections from the switch appear not always to be the physically closest connection from a switch to a light bulb as a router in the UI, but that connection is still green.

Is there a known issue that I can workaround somehow? Is there anything I can do to fix this or troubleshoot further to identify and fix the issue? I had the official hue bridge before and never had any issues with unresponsive lights or lost events).

## Steps to reproduce the behavior

Sadly I can't reliable reproduce the issue. But it does appear to happen at least once a day on a random switch.

## Screenshots

Screenshot of the logs, can't easily copy them out of VNC in home assistant

[image]

[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistent deCONZ Add-on (Host: Debian 64bit)
 - Firmware version: 26660700
 - deCONZ version: 2.09.03 / 12/23/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes, extension cable plugged in USB 2 port
 - Is there any other USB or serial devices connected to the host system? If so: Yes, SSD plugged in USB 3 port

Mimiix, 2021-02-12:
Are you able to provide some logs with INFO, APS and APS_L2?

Please in a log output instead of a screen :)

peterjuras, 2021-02-12:
I enabled the log levels you mentioned now and will post the logs once the issue occurs again. Do you know of an easy way to copy logs through VNC?

ebaauw, 2021-02-12:
I added support for the 0xFC00 cluster in the GUI in v2.9.3, so you can monitor the button events in real-time in the _Cluster Info_ panel.  If the button events don't show there, the issue is with the Zigbee communication between the dimmer and the coordinator.  If they do show there, but not in the web socket notifications, the issue is with the API plugin.
[image]

peterjuras, 2021-02-12:
Thanks @ebaauw , I see all events in real time in your newly added panel.

The issue however is that most of the time (>95%) everything is very fast and working as expected. If a switch is misbehaving, it will likely be healed again after 5 seconds and therefore looking at this panel would show everything as working again.

Therefore I hope that the debug messages that I showed above are also a realistic representation of the button events that the coordinator receives.

ebaauw, 2021-02-12:
> Therefore I hope that the debug messages that I showed above are also a realistic representation of the button events that the coordinator receives.

Not quite; (most of) the debug messages are generated by the REST API plugin; the _Cluster Info_ panel is updated by the deCONZ core.  The most realistic representation would be sniffing the Zigbee network.

> If a switch is misbehaving, it will likely be healed again after 5 seconds 

I think you're experiencing Zigbee network issues:
- The dimmer sends the notification for the x000 event;
- The coordinator receives this event, and the API plugin issues the x000;
- The API plugin sends a _Default Response_ to the notification, but this doesn't reach the dimmer;
- The dimmer finds a new parent (that could actually be the _Device Annoucement_ message from your log) and doesn't send the 0x002 event, because it thinks the x000 event hasn't been received.
- On the next press, the dimmer sends another x000 notification, which is met by a _Default Response_, so now the dimmer sends the x002 on release.

Mimiix, 2021-02-12:
> I enabled the log levels you mentioned now and will post the logs once the issue occurs again. Do you know of an easy way to copy logs through VNC?

I'd like to have them already.  As there might be related error codes.

No. That's a thing with the Addon. I suggest to report a bug report to them.  You need to use a native VNC client.

peterjuras, 2021-02-12:
Thanks for the fast responses and help.

I know gathered some logs with the `INFO`, `APS` and `APS_L2` levels. I made a lot of clicks with different switches and one of them showed the red light for a brief period, although the switches were mostly working as expected:

<details>
<summary>Logs</summary>

```

13:42:05:358 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
13:42:05:366 CTRL. 3.27.2
13:42:05:420 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2409577-if00
13:42:05:459 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2409577-if00 / serialno: DE2409577, ConBee II
13:42:05:460 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
13:42:05:568 parent process s6-supervise
13:42:05:569 gw run mode: docker/hassio
13:42:05:569 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
13:42:05:571 DB sqlite version 3.27.2
13:42:05:573 DB PRAGMA page_count: 43
13:42:05:573 DB PRAGMA page_size: 4096
13:42:05:574 DB PRAGMA freelist_count: 2
13:42:05:574 DB file size 176128 bytes, free pages 2
13:42:05:574 DB PRAGMA user_version: 7
13:42:05:575 DB cleanup
13:42:05:576 DB create temporary views
13:42:05:587 started websocket server at port 8081
13:42:05:590 [INFO] - Found file containing button maps. Parsing data...
13:42:05:593 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
13:42:05:600 [INFO] - Button maps loaded.
13:42:05:602 dlg action: Read binding table
13:42:05:603 found node plugin: libde_rest_plugin.so - REST API Plugin
13:42:05:606 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
13:42:09:830 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
13:42:09:952 LightNode 1: Configuration tool 1 added
13:42:09:962 LightNode 2: Living room color light added
13:42:09:969 LightNode 3: Hallway light 1 added
13:42:09:972 LightNode 4: Hallway light 2 added
13:42:09:975 LightNode 5: Hallway light 3 added
13:42:09:978 LightNode 6: Hallway light 4 added
13:42:09:980 LightNode 7: Hallway light 5 added
13:42:09:984 LightNode 8: Hallway color light added
13:42:09:987 LightNode 9: Kitchen light 1 added
13:42:09:991 LightNode 10: Kitchen light 2 added
13:42:09:994 LightNode 11: Bedroom light 1 added
13:42:09:996 LightNode 12: Bedroom light 2 added
13:42:09:999 LightNode 13: Bedroom light 3 added
13:42:10:003 LightNode 14: Bathroom light 1 added
13:42:10:006 LightNode 15: Bathroom light 2 added
13:42:10:010 LightNode 16: Bathroom light 3 added
13:42:10:011 SensorNode 2 set node 0x00178801080cd358
13:42:10:014 SensorNode 3 set node 0x0017880106e94ec0
13:42:10:016 SensorNode 4 set node 0x0017880106a6a05c
13:42:10:019 SensorNode 5 set node 0x00178801086b2eba
13:42:10:021 SensorNode 6 set node 0x0017880104ee35c4
13:42:10:024 SensorNode 7 set node 0x0017880106e94e7e
13:42:10:028 LightNode 17: Bedroom strip 1 added
13:42:10:031 LightNode 18: Bedroom strip 2 added
13:42:10:034 LightNode 19: Living room floor lighter added
13:42:10:080 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2409577-if00 / serialno: DE2409577, ConBee II
13:42:10:385 New websocket 172.30.32.1:38264 (state: 3) 
13:42:11:122 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2409577-if00 / serialno: DE2409577, ConBee II
13:42:11:139 COM check bootloader
13:42:11:180 Announced to internet https://phoscon.de/discover
13:42:11:181 discovery server date: Fri, 12 Feb 2021 12:41:58 GMT
13:42:11:181 	 local time seems to be ok
13:42:13:146 COM timout query bootloader, assume application
13:42:13:170 Device firmware version 0x26660700 ConBee II
13:42:13:199 unlocked max nodes: 200
13:42:13:271 Device protocol version: 0x010C
13:42:13:357 Current channel 15
[... 1594 lines cut ...]
13:44:51:753 APS-DATA.request id: 154 erase from queue
13:44:52:052 APS-DATA.request id: 159, addrmode: 0x03, addr: 0x001788010841b9c6, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
13:44:52:074 APS-DATA.confirm id: 159, status: 0x00 SUCCESS
13:44:52:074 APS-DATA.confirm request id: 159 -> confirmed, timeout 1613133892
13:44:52:103 APS-DATA.indication srcAddr: 0x592b, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -55
13:44:52:104 	asdu: 9a001203036ab406ffff2e210000242e06018817000e2525020f746ab406ffff2e210093c2cf0401881700a82525020f4d6ab406ffff2e21008e2a010901881700582625020f9f
13:44:52:104 APS-DATA.indication request id: 159 -> finished
13:44:52:105 APS-DATA.request id: 159 erase from queue
13:44:52:400 APS-DATA.request id: 177, addrmode: 0x03, addr: 0x00212effff06b46a, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
13:44:52:418 APS-DATA.confirm id: 177, status: 0x00 SUCCESS
13:44:52:420 APS-DATA.confirm request id: 177 -> confirmed, timeout 1613133892
13:44:52:450 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8033, lqi: 255, rssi: -55
13:44:52:452 	asdu: 3400000000
13:44:52:452 APS-DATA.indication request id: 177 -> finished
13:44:52:453 APS-DATA.request id: 177 erase from queue
13:44:52:901 APS-DATA.request id: 168, addrmode: 0x03, addr: 0x001788010946d9bc, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
13:44:52:926 APS-DATA.confirm id: 168, status: 0x00 SUCCESS
13:44:52:927 APS-DATA.confirm request id: 168 -> erase from queue
13:44:52:958 APS-DATA.indication srcAddr: 0x113b, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -52
13:44:52:958 	asdu: 18c4010000001000
13:44:52:959 APS-DATA.request id: 168 erase from queue

```

</details>

ebaauw, 2021-02-12:
> one of them showed the red light for a brief period

That's when the switch is finding a new parent, because it didn't receive a _Default Response_.

SwoopX, 2021-02-12:
@ebaauw Do those really require a _default response_ on anything? If so, than this is destined to fail since deconz barely gives them (apart from any attribute reports).

The enhanced debug output for the buttons also shows that the commands are unicast to the coordinator, so you lose the benefit of the switch's group.

Apart from that, the logs make me wonder:
1. Why the commands from the switch are evaluated twice via different functions
2. Why and how those events can occur (seen them a couple of times lately):
`13:44:32:611 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task`

ebaauw, 2021-02-13:
> Do those really require a default response on anything?

Hm, checking some old sniffer logs, it does set _Disable Default Response_.  Maybe it's a regular ACK, but the Hue dimmer switch does expect some sort of response.  

> The enhanced debug output for the buttons also shows that the commands are unicast to the coordinator

Yes.  Remember, the Hue dimmer switch acts as sensor, sending a 0x00 command from 0xFC00 to the coordinator (or the Hue bridge) with the button event value.  ACK or default response only apply to unicast.

> Why the commands from the switch are evaluated twice via different functions

Somehow all traffic from a device with a ZHASwitch sensor is evaluated for button events; you also see the "no button map (entry)" for e.g. battery reports.

peterjuras, 2021-02-19:
Is there anything I can do to improve the situation? I have also switched between multiple zigbee channels for a few days but the problems appear to remain.

0rsa, 2021-03-08:
I have exactly the same problem with the hue dimmers as the author of the post (sometimes there is a delay, sometimes the order is completely lost).
For me this happens when the switch has not been used for several hours / days, sometimes I see the switch greyed out in the Conbee web interface.
After a first lost or late order, everything is back to normal and the switch works as planned.

ahocquet, 2021-03-14:
It seems related to my other issue that I have with Aqara will switches #4584 

Nothing happens on first press, then it works.

peterjuras, 2021-04-05:
The issue still persists, if anything it appears to get worse with increased uptime without powering off the deconz stick and powering it on again after ~ a week.

peterjuras, 2021-04-29:
The issue still persists.

closed 2021-06-28 by github-actions[bot]

peterjuras, 2021-07-01:
It would be great if this could be reopened. The issue still exists, hue dimmer switches are not always responding.

0rsa, 2021-07-01:
It also happens using Aqara Opple switchs (I had also Philips hue switchs which are impacted in a same way).