dresden-elektronik/deconz-rest-plugin #4409: MOES 2 gang switch (tuya)

erikje14, 2021-02-12:
## Device

- Product name: MOES 2 gang wireless scene switch 22/44
- Manufacturer: TZ3000_arfwfgoa
- Model identifier: TS0042
- Device type : Please remove all unrelated device types. 

  - Switch

## Screenshots

erikje14, 2021-02-12:
[image]

erikje14, 2021-02-12:
[image]

erikje14, 2021-02-12:
[image]

erikje14, 2021-02-12:
[image]
[image]

Mimiix, 2021-02-12:
@Smanar

erikje14, 2021-02-12:
it's this one fyi:
https://nl.aliexpress.com/item/1005001504737652.html?spm=a2g0s.9042311.0.0.6ff84c4dhC00ci

Smanar, 2021-02-12:
Added in the list https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4157

closed 2021-02-15 by manup