dresden-elektronik/deconz-rest-plugin #4411: How to specify serial port/device for deCONZ (in 2.09.03)?

mptei, 2021-02-12:
## Describe the question or issue you are having
I would like to use another serial device than /dev/ttyAMA0 to communicate with the RaspBee.

I know it was possible before (in 2.05.39) with command line option "--dev". But now in version 2.09.03 this is no longer possible.
It still uses only /dev/ttyAMA0.

How can I specify the device?

## Screenshots

## Environment
 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: 2.09.03
 - Device: RaspBee I
 - Do you use an USB extension cable: no -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? raspberries bluetooth uart

Command Line: /usr/bin/deCONZ -platform minimal --http-port=88 --dev=/dev/ttyAMA2

## deCONZ Logs

## Additional context
I rewired the Raspbee module to use another serial port (/dev/ttyAMA2) this worked nicely in 2.05.39.

manup, 2021-02-12:
Currently the code has a hard coded filter which includes ttyAMA0 and ttyS0 for RaspBee, even with `--dev` specified. I'll extend the filter to include whatever is given via `--dev`.

mingan666, 2021-03-06:
This is still an issue preventing me from upgrading my deconz.
Could we get a status update from the deconz team?

Mimiix, 2021-03-08:
Let me check with Manuel :)

manup, 2021-03-20:
Can you please check if this works now with the latest version?

mingan666, 2021-03-20:
I'm not sure I'm using the correct deconz version, however it does not seem to be working in the version I tried:

```
[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.10.03
....
12:36:05:619 dev can't create file: /dev/ttyACM0 via mknod (missing permissions)

As you can see, I use the marthoc/deconz docker container.

mptei, 2021-03-21:
I can confirm that it works again with deconz-2.10.03-beta.

That means I've added "--dev=/dev/ttyAMA2" to the command line of deCONZ. A RaspbeeII is connected to /dev/ttyAMA2 and is recognized and used,

Thanks for the fix.

closed 2021-03-31 by mptei