dresden-elektronik/deconz-rest-plugin #4413: Sengled E1E-G7F switch button mapping is incorrect

kyletpugh, 2021-02-13:
## Describe the bug

Pressing any button on the Sengled E1E-G7F switch results in event 1002, which is the on button single press.

## Steps to reproduce the behavior

Press any button on the switch, happens every time.

## Expected behavior

The buttons should send the correctly mapped events.

## Environment

 - Host system: PC
 - Running method: Ubuntu 20.04 LTS (up to date)
 - Firmware version: 26660700
 - deCONZ version: 2.09.03 / 12/23/2020 (beta-channel packages)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes, a CyberPower CP1500PFCLCDa UPS.

## deCONZ Logs

I will get these for you as soon as I can, I need to access the server which I cannot do currently but will ASAP.

## Additional context

See #3308 and #4275 for more information.

kyletpugh, 2021-02-24:
Sorry I took so long to grab these logs, I have been pretty sick. As you can see from the log I attached every button is given the 1002 code for each event. I did single press all four buttons in order from top to bottom, double press top and bottom buttons, long press top and bottom buttons. They all result in 1002 event code.
[deconz-log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6035714/deconz-log.txt)

mdolnik, 2021-03-01:
This seems like it may be related to issues I was having with a `Eria Adurosmart Wireless Dimming Switch`.

Refer to [my comment](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4495#issuecomment-787565658) in #4495. I don't know yet if it will be an accepted solution and I don't consider it very ideal, but it currently works.

closed 2021-03-03 by manup

kyletpugh, 2021-03-15:
This is corrected for the right mappings now. Though I'm not sure if it's the switch or deconz but double pressing the on and off buttons receives continuous events while the button is depressed the second time.