dresden-elektronik/deconz-rest-plugin #4414: (Tuya) Rotary Dimmer Switch TS0601 _TZE200_dfxkcots

choss38, 2021-02-13:
## Device

- Product name: Rotary Smart Dimmer Switch 

There is no brand on the package but seller write Tuya 
https://fr.aliexpress.com/item/1005001704726670.html

Very similar to https://zigbee.blakadder.com/Earda_EDM-1ZAB-EU.html

- Manufacturer: _TZE200_dfxkcots
- Model identifier: TS0601
- Device type : Switch

Phoscon recongnize this product as a light, Vendor: _TZE200_dfxkcots, Model: Fairy lights and names it "Smart switch"

It is integrated in HA as a simple on/off switch, there is no brightness support.

## Screenshots
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Smanar, 2021-02-13:
Ha nice, something new.

Do you have a linux machine to test the code change ?

choss38, 2021-02-13:
For sure I have one: I've got plenty !  :smiley: 
After 25 years with MS, I'm free for 5 years and I don't regret ! One of the best decisions of my life !! :smiley: 
I'm Ok, I'm ready to listen your instructions.

Smanar, 2021-02-13:
Nice, because we will need them ^^.

So you need to instal deconz then take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch tuya_dimmer https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then you need to re-include your device, so it s realy something new in deconz, all will not work immediatly. So, if I m right with the new code
- the device will have the field state/bri
- a new type : "dimmable light"
- the on/off will work (It is already working I think ?
- Starting to implement the "bri" command for tuya stuff

On deconz/help/log with "info" you will have some logs everytime you will try to change the level, if you can share them ?
For information all tuya request in and out are logged and easy to reconize.

If you are using your production machine, you can make a backup of the file "/usr/share/deCONZ/plugins/libde_rest_plugin.so" but you need to put it in another folder.

choss38, 2021-02-13:
Ok, i'll see that in a few days.
I have a doubt. Actually I run Deconz with a Conbee on a production HA. I think it's not very carefull to play with this machine.ill
But if I put Deconz on a other machine, and plug the Conbee on this machine, is there any risk to break all the associations with it ?
The Conbee will not be reset with a new ID that would'nt be recognize by all my devices ?
Because I'm not very ready to have to pair again all my devices (and I really need a working heating  these days!!)

Smanar, 2021-02-14:
No it s safe, and it s better to do that.

- Using deconz in an other machine with a full OS with be easier than use it in HA (mostly if you are suing hassio or other docker install)
- The network setting are stored on the conbee
- The devices config are stored on the host
- You can configure HA to use the distant deconz install for test too.
- You can too make a zigbee backup (setting + device config) using phoscon
- You can just switch the conbee from a machine to another, make test during 15 mn and re-switch it.

Just a problem, Xiaomi/Aquara sensor devices don't like to be without parent, so if they don't find their network after some hours, they can leave.

choss38, 2021-02-18:
Hi,
i tried this morning. 
I'm silly blocked by Phoscon which doesn't accept default password and I don't know how to pair a device without it.
Do you know what I have to do ?

Smanar, 2021-02-18:
You have a link "lost password" on the password page.

choss38, 2021-02-18:
Oh yes, I saw it but I was afraid to reset the Conbee. I'm sorry to misunderstand how it works.
Anyiway I tried to reset but I've got a new problem: Failed to reset password. Make sure the gateway was powercycled recently.
I unplugged than plugged again the Conbee: same message. 
I tried deconz/edit/reboot the device: same.
I tried to start deconz via terminal, same.
Now I don't know where I am, but Phoscon app doesn't find the gateway (look up 
192.168.0.2:8080 100 %10.0.0.1 100 %10.0.1.1 100 %10.0.10.1 100 %10.1.1.1 100 %10.10.1.1 100 %10.90.90.90 99 %172.16.0.1 99 %172.16.1.1 83 %...), therefore Webapp open without problem(http://192.168.0.2:8080/login.html), but doesn't want to reset password too. I tried different browsers.

I reinstall all but nothing better. No more phoscon !

There is some warning, maybe there is a answer:

` ~/deconz-rest-plugin $ /usr/bin/deCONZ-autostart.sh
/sbin/setcap
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
22:22:46:115 void zmController::loadNodesFromDb(),406: assertion 'stmt' failed
22:22:46:115 void zmController::loadNodesFromDb(),407: assertion 'rc == SQLITE_OK' failed
22:22:46:115 NodeInfo zmController::nodeFromQVariant(const QVariantMap&),9454: assertion 'stmt' failed
22:22:46:115 NodeInfo zmController::nodeFromQVariant(const QVariantMap&),9455: assertion 'rc == SQLITE_OK' failed
22:22:46:115 NodeInfo zmController::nodeFromQVariant(const QVariantMap&),9454: assertion 'stmt' failed
22:22:46:115 NodeInfo zmController::nodeFromQVariant(const QVariantMap&),9455: assertion 'rc == SQLITE_OK' failed
22:22:54:419 unhandled APS-DATA.confirm id: 12 status 0xE1
22:23:01:716 void zmController::saveNodesState(),9073: assertion 'stmt' failed
22:23:01:717 void zmController::saveNodesState(),9074: assertion 'rc == SQLITE_OK' failed
22:23:08:016 unhandled APS-DATA.confirm id: 21 status 0xE1
22:23:14:716 void zmController::saveNodesState(),9073: assertion 'stmt' failed
22:23:14:716 void zmController::saveNodesState(),9074: assertion 'rc == SQLITE_OK' failed
22:24:08:913 unhandled APS-DATA.confirm id: 58 status 0xE1
22:24:11:328 unhandled APS-DATA.confirm id: 59 status 0xE1

`

I will all try again tomorrow on an other machine, a raspberry pi 4.

choss38, 2021-02-19:
Unable to change password on the pi 4 too. I don't understand. I'm going to retry with phoscon sdcard image https://phoscon.de/en/conbee/sdcard.

choss38, 2021-02-19:
The deconz-dev is not configured like deconz. When I started Phoscon from Deconz, It asked for a gateway name and a login password. This didn't occured with deconz-dev and I wasn't able to reset the password: Failed to reset password. Make sure the gateway was powercycled recently.

choss38, 2021-02-19:
Other difference: Deconz-dev gateway has 172.30.33.6 for IP when deconz has 192.168.0.46, the IP of the raspberry in the LAN.

choss38, 2021-02-19:
I install deconz-dev over deconz (without uninstalling it) on the sdcard supplied by Phoscon. I don't know if I start deconz-dev or deconz, but now I'm not able to use phoscon: password don't work and I can't reset it.

Smanar, 2021-02-19:
deconz-dev is just some lib for compilation, it s not an application (so it can't use other ip), there is no deconz + deconz_dev.
Deconz don't support 2 instances, so if you can see 2 instances, deconz + the one you call deconz-dev, it can be the problem.

You can kill the application you are calling deconz_dev.

choss38, 2021-02-19:
There is just one instance of deconz, I checked.
From the Phoscon sd-card, with deconz already installed, I had run ` sudo apt install deconz-dev` and all the commands you wrote.

I tried each time delight/delight. No way.

Smanar, 2021-02-19:
So all seem fine ...
You have no  error message when trying to reset the password ?
Nothing interesting in log in deconz/help ?

choss38, 2021-02-20:
Maybe that:

> 10:02:10:796 HTTP client GET /usr/share/deCONZ/webapp//api/config
> 10:02:10:796 	/api/config --> HTTP/1.1 404 Not Found
> 10:02:10:798 http client (127.0.0.1) handled 1 requests
> 10:02:10:799 ~http client 0x0x168e0d8
> 10:02:14:555 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
> 10:02:16:745 HTTP client hdr detected:
> GET /api/config HTTP/1.1
> Host: 127.0.0.1
> User-Agent: curl/7.64.0
> Accept: */*

Strange double / in url GET request!

choss38, 2021-02-20:
Restarting from scratch...
Flashing Pi4 sdcard with Phoscon supplied image. Not installing dev for the beginning
Starting Deconz, no error in deconz help debug http

> 09:58:24:065 HTTP API GET /api/config - 127.0.0.1
> 09:58:24:065 {"apiversion":"1.16.0","bridgeid":"00212EFFFF03B2FF","datastoreversion":"93","devicename":"ConBee","factorynew":false,"mac":"dc:a6:32:04:fb:16","modelid":"deCONZ","name":"Phoscon-GW","replacesbridgeid":null,"starterkitid":"","swversion":"2.5.88"}
> 09:58:24:086 http client (127.0.0.1) handled 1 requests
> 09:58:24:087 ~http client 0x0x197e4e8
> 09:58:24:573 HTTP client hdr detected:
> GET /api/383/config/wifi/ HTTP/1.1
> Host: 127.0.0.1
> User-Agent: curl/7.64.0
> Accept: */*
> 
> 
> 09:58:24:574 HTTP API GET /api/383/config/wifi/ - 127.0.0.1
> 09:58:24:576 {"active":"active","eth0":"192.168.0.46","ip":"192.168.8.1","lastupdated":1606399819,"name":"Phoscon-Gateway-B2FF","pw":null,"state":"ap-configured","type":"accesspoint","wifi":"configured","wifiavailable":
> ...

Open Phoscon, url in address bar: http://192.168.0.46/pwa/login.html#host/192.168.0.46:80
Be drived to the page http://192.168.0.46/pwa/login.html#passwordchange/00212EFFFF03B2FF 
which ask to set password.
Installing password, quit the webpage, quit deconz and restarting...
Lauching deconz, Phoscon wep page, logging... no problem.

Smanar, 2021-02-20:
Ha, I m french too if you prefer discord ?

>Strange double / in url GET request!

Yeap, idk what happen, but there is so much user that do this command, first time I see this issue, I realy don't understand.
You have tried to re-install

`sudo apt install deconz-dev`

Just that, for me the problem is coming from my code, and the command that have broken your OS is 
`sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins`

This command replace the API plugin file.

choss38, 2021-02-20:
Oh, I'm not using Discord. I go on with english for the forum users.
Ok, next step:
`sudo apt install deconz-dev`
No apt install error. Phoscon still works.

```
git clone --branch tuya_dimmer https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```
Next to be continued...

choss38, 2021-02-20:
and yes, this is libde_rest_plugin.so which make the problem.
when I paste the original backup file, all is working fine again.
I stop my investigations for today.
If you want to reproduce it: I use a Raspeberry Pi 4 b 1Go with the Phoscon Raspbian Buster Desktop Stable  (https://phoscon.de/en/conbee/sdcard) and a Conbee gateway.

Smanar, 2021-02-20:
No problem, I will make test on my side.

Smanar, 2021-02-20:
Ok, I have updated the code with the lasts modifications from official git, 1 hour the code is running, all is still working, and phoscon too.

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

choss38, 2021-02-21:
Hi,
that's ok now. Phoscon works.

I include the dimmer.
It's recognized as Smart plug.
[image]

[image]

[image]

Here is the log info of the pairing:

> 
> 09:48:53:316 send permit join, duration: 59
> 09:48:53:318 send GP proxy commissioning mode
> 09:48:56:091 Beacon src: 0xBBAB ch: 11 updateId: 1
> 09:48:56:219 Beacon src: 0xBBAB ch: 11 updateId: 1
> 09:48:56:587 unknown node 0x60A423FFFEEC6B6F (0xAA2B), lqi: 255
> 09:48:56:588 APS-DATA.indication from unknown node 0x60A423FFFEEC6B6F
> 09:48:56:590 new node - ext: 0x60a423fffeec6b6f, nwk: 0xAA2B
> 09:48:56:590 device announce 0x60A423FFFEEC6B6F (0xAA2B) mac capabilities 0x8E
> 09:48:56:590 device announce 0x60A423FFFEEC6B6F (0xAA2B) mac capabilities 0x8E
> 09:49:02:320 LightNode 1: Configuration tool 1 added
> 09:49:02:327 LightNode 2: On/Off light 2 added
> 09:49:02:329 LightNode 3: Dimmable light 3 added
> 09:49:02:331 drop request to zombie (rx = 0)
> 09:49:02:617 drop request to zombie (rx = 0)
> 09:49:02:816 drop request to zombie (rx = 0)
> 09:49:05:889 Current channel 11
> 09:49:05:905 CTRL got nwk update id 1
> 09:49:05:921 Device TTL 2076 s flags: 0x7
> 09:49:06:854 saved node state in 0 ms
> 09:49:07:012 sync() in 157 ms
> 09:49:07:128 DB UPDATE device_descriptors SET data = x'01408e0210525200002c520000', timestamp = 1613900947 WHERE device_id = (SELECT id FROM devices WHERE mac = '60:a4:23:ff:fe:ec:6b:6f') AND endpoint = 0 AND type = 2
> 09:49:07:129 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'01408e0210525200002c520000', 1613900947 FROM devices WHERE mac = '60:a4:23:ff:fe:ec:6b:6f'
> 09:49:07:495 LightNode 4: Smart plug 4 added
> 09:49:07:496 DB UPDATE device_descriptors SET data = x'0104015100010400000400050000ef0219000a00', timestamp = 1613900947 WHERE device_id = (SELECT id FROM devices WHERE mac = '60:a4:23:ff:fe:ec:6b:6f') AND endpoint = 1 AND type = 4
> 09:49:07:498 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104015100010400000400050000ef0219000a00', 1613900947 FROM devices WHERE mac = '60:a4:23:ff:fe:ec:6b:6f'
> 09:49:07:573 ZDP discovery done in 10983 ms
> 09:49:07:587 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010101000101
> 09:49:07:587 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1
> 09:49:07:588 Tuya debug 10 : Updating model ID
> 09:49:07:616 delay sending request 198 dt 0 ms to 0x60A423FFFEEC6B6F, ep: 0x01 cluster: 0x0004 onAir: 1
> 09:49:07:827 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010101000101
> 09:49:07:827 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1
> 09:49:07:827 Tuya debug 10 : Updating model ID
> 09:49:07:859 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 000202020004000001f4
> 09:49:07:859 Tuya debug 5 : Status: 0 Transid: 2 Dp: 514 (0x02,0x02) Fn: 0 Data 500
> 09:49:07:891 Add to group response for light 4. Status:0x00, capacity: 0
> 09:49:08:019 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 000202020004000001f4
> 09:49:08:019 Tuya debug 5 : Status: 0 Transid: 2 Dp: 514 (0x02,0x02) Fn: 0 Data 500
> 09:49:08:099 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:08:099 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:08:259 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:08:260 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:08:467 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:08:468 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:08:659 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:08:660 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:08:867 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:08:868 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:09:058 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:09:059 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:09:267 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:09:267 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:09:459 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:09:461 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 09:49:09:667 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0004030200040000000a
> 09:49:09:668 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10
[... 68 lines cut ...]
> 10:15:44:698 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 000102020004000000f0
> 10:15:44:699 Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 240
> 10:15:44:827 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 000202020004000000a0
> 10:15:44:828 Tuya debug 5 : Status: 0 Transid: 2 Dp: 514 (0x02,0x02) Fn: 0 Data 160
> 10:15:44:973 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010202000400000078
> 10:15:44:974 Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 120
> 10:15:45:307 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010202000400000050
> 10:15:45:308 Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 80
> 10:15:45:403 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00020202000400000028
[... cut ...]
> 10:15:45:501 Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 10

and when I push on it to switch off

> 10:16:34:952 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010101000100
> 10:16:34:953 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 0
> 

and push again  to switch on

> 10:16:42:874 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010101000101
> 10:16:42:875 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1

I hope it could help.

Smanar, 2021-02-21:
When you have used the device the first time

> 09:49:07:588 Tuya debug 10 : Updating model ID
> 09:49:07:827 Tuya debug 10 : Updating model ID

This happen, when deconz don't know the model id, or the manufacture name.

Try to include the device 2 time, without deleting it, like that deconz can use data previously retreived.

If the device is boring to include, just
- set phoscon in permit join (add new light)
- read the basic attribute again in deconz, cluster "basic" 0x0000, press button "read".

The other logs are realy fine, no problems visibles.

choss38, 2021-02-22:
Hi,
I don't know if I did the good manipulations.
So I reseted the device without delete it in Deconz once and I included it again:

> 19:05:50:390 verified group capacity: 15 and group count: 1 of LightNode 0x01124b001bae791a
> 19:05:50:397 0x01124b001bae791a found group 0xFFF0
> 19:05:51:117 verified group capacity: 255 and group count: 1 of LightNode 0xbc33acfffe8b972e
> 19:05:51:117 0xbc33acfffe8b972e found group 0xFFF0
> 19:05:51:745 remove binding from 0xBC33ACFFFE8B972E cluster 0x2000 due non existing cluster
> 19:05:51:933 verified group capacity: 255 and group count: 1 of LightNode 0x60a423fffeec6b6f
> 19:05:51:933 0x60a423fffeec6b6f found group 0xFFF0
> 19:05:52:831 Unbind response failed with status 0x88 for 0xbc33acfffe8b972e ep: 0x01 cluster: 0x2000
> 19:06:45:568 Current channel 11
> 19:06:45:584 CTRL got nwk update id 1
> 19:06:45:600 Device TTL 3373 s flags: 0x7
> 19:06:48:887 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010101000101
> 19:06:48:887 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1
> 19:06:55:335 Beacon src: 0xBBAB ch: 11 updateId: 1
> 19:06:56:344 Beacon src: 0x553C ch: 11 updateId: 0
> 19:06:56:345 * node has lower updateId should be 1
> 19:06:56:360 Beacon src: 0xBBAB ch: 11 updateId: 1
> 19:06:58:185 Beacon src: 0x553C ch: 11 updateId: 0
> 19:06:58:186 * node has lower updateId should be 1
> ....
> 19:08:17:727 DeviceAnnce of LightNode: 0x60a423fffeec6b6f Permit Join: 42
> 19:08:17:728 device announce 0x60A423FFFEEC6B6F (0xAA2B) mac capabilities 0x8E
> 19:08:17:728 device announce 0x60A423FFFEEC6B6F (0xAA2B) mac capabilities 0x8E
> 19:08:17:854 ZDP discovery done in 129 ms
> 19:08:18:157 verified group capacity: 255 and group count: 1 of LightNode 0x60a423fffeec6b6f
> 19:08:18:158 0x60a423fffeec6b6f found group 0xFFF0
> 19:08:25:490 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010101000101
> 19:08:25:490 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1
> 19:08:25:634 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 000202020004000001f4
> 19:08:25:634 Tuya debug 5 : Status: 0 Transid: 2 Dp: 514 (0x02,0x02) Fn: 0 Data 500
> 19:08:25:778 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:25:779 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:25:971 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:25:972 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:26:178 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:26:179 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:26:370 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:26:371 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:26:578 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:26:579 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:26:771 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:26:772 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:26:979 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:26:980 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:27:171 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:27:172 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:27:363 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:27:364 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:27:571 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:27:572 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:27:763 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:27:764 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:27:972 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:27:972 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:28:179 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:28:180 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:28:372 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:28:373 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:29:013 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:29:014 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:29:172 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:29:173 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:29:828 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:29:829 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
> 19:08:29:973 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 0001030200040000000a
> 19:08:29:973 Tuya debug 5 : Status: 0 Transid: 1 Dp: 515 (0x02,0x03) Fn: 0 Data 10
[... 75 lines cut ...]
> 19:14:36:201 Tuya debug 1 : Time sync request
> 19:14:36:202 Send Tuya command 0x24, data: 000e6034029c6034029c
> 19:14:40:527 saved node state in 1 ms
> 19:14:40:658 sync() in 130 ms
[... cut ...]
> 19:15:45:589 CTRL got nwk update id 1
> 19:15:45:605 Device TTL 2833 s flags: 0x7
> 19:15:51:546 dev /dev/ttyS0 (/dev/ttyS0)
> 19:15:51:549 GW firmware version: 0x26390500
> 19:15:51:549 GW firmware version is up to date: 0x26390500
> 19:16:14:507 send permit join, duration: 59
> 19:16:14:508 send GP proxy commissioning mode
> 19:16:45:565 Current channel 11
> 19:16:45:581 CTRL got nwk update id 1
> 19:16:45:597 Device TTL 2773 s flags: 0x7

I didn't see a real difference... 
Is that ok for you ?

Smanar, 2021-02-22:
> I didn't see a real difference...

Same, nothing about the "smart plug 4" ....
Instead of manualy resetting the device, have you tried this method ?

>     set phoscon in permit join (add new light)
>     read the basic attribute again in deconz, cluster "basic" 0x0000, press button "read".

Are you using third app ? to check the device JSON ?
Else you can see it using "http" as debug and pressing a button, but can be realy talkative log.
You can too see it using the dev console in your browser.
Or I can explain how to do direclty with the API ?

choss38, 2021-02-23:
Hi,

this method

```
set phoscon in permit join (add new light)
read the basic attribute again in deconz, cluster "basic" 0x0000, press button "read".

```
gave nothing:

> 07:27:15:782 Current channel 11
> 07:27:15:798 CTRL got nwk update id 1
> 07:27:15:814 Device TTL 3433 s flags: 0x7
> 07:27:21:129 ZCL attribute report 0x000425FFFF1E9443 for cluster: 0x0001, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
> 07:27:21:321 remove binding from 0xBC33ACFFFE8B972E cluster 0x2000 due non existing cluster
> 07:27:22:377 Unbind response failed with status 0x88 for 0xbc33acfffe8b972e ep: 0x01 cluster: 0x2000
> 07:27:34:719 Skip idle timer callback, too early: elapsed 830 msec
> 07:27:58:912 created username: D7042DC3A1, devicetype: Phoscon#B1050x886
> 07:28:02:485 New websocket 192.168.0.46:43496 (state: 3) 
> 07:28:04:719 Skip idle timer callback, too early: elapsed 892 msec
> 07:28:07:719 send permit join, duration: 59
> 07:28:07:720 send GP proxy commissioning mode
> 07:28:15:785 Current channel 11
> 07:28:15:801 CTRL got nwk update id 1
> 07:28:15:817 Device TTL 3373 s flags: 0x7
> 07:28:51:731 No thermostat sensor found for 0x00158D000192D15F, endpoint: 0x01
> 07:28:51:731 ZCL attribute report 0x00158D000192D15F for cluster: 0x0201, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
> 07:29:06:719 send permit join, duration: 59
> 07:29:06:720 send GP proxy commissioning mode
> 07:29:15:777 Current channel 11
> 07:29:15:793 CTRL got nwk update id 1
> 07:29:15:809 Device TTL 3313 s flags: 0x7
> 07:29:21:765 dev /dev/ttyS0 (/dev/ttyS0)
> 07:29:21:772 GW firmware version: 0x26390500
> 07:29:21:773 GW firmware version is up to date: 0x26390500
> 07:30:05:719 send permit join, duration: 59
> 07:30:05:720 send GP proxy commissioning mode
> 07:30:15:784 Current channel 11
> 07:30:15:800 CTRL got nwk update id 1
> 07:30:15:816 Device TTL 3253 s flags: 0x7
> 07:30:27:720 Skip idle timer callback, too early: elapsed 799 msec
> 07:31:04:720 send permit join, duration: 0
> 07:31:05:015 Search sensors done
> 07:31:08:719 Skip idle timer callback, too early: elapsed 898 msec

`Are you using third app ? to check the device JSON ?
`

No, I never do that.

With http debug log:

> 07:41:29:870 HTTP API GET /api/config?_=1614066089842 - 192.168.0.2
> 07:41:29:871 {"apiversion":"1.16.0","bridgeid":"00212EFFFF03B2FF","datastoreversion":"93","devicename":"ConBee","factorynew":false,"mac":"dc:a6:32:04:fb:16","modelid":"deCONZ","name":"Phoscon-GW","replacesbridgeid":null,"starterkitid":"","swversion":"2.10.0"}
> 07:41:29:949 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010101000101
> 07:41:29:950 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1
> 07:41:30:274 HTTP client hdr detected:
> GET /api/59C37F236F/sensors/new HTTP/1.1
> Host: 192.168.0.46
> Connection: keep-alive
> Accept: vnd.ddel.v1
> User-Agent: Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.197 Safari/537.36
> X-Requested-With: XMLHttpRequest
> Referer: http://192.168.0.46/pwa/commissioning.html?1608764346000
> Accept-Encoding: gzip, deflate
> Accept-Language: fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7
> 
> 
> 07:41:30:275 HTTP API GET /api/59C37F236F/sensors/new - 192.168.0.46
[... 41 lines cut ...]
> 07:41:32:684 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 00010202000400000294
> 07:41:32:687 Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 660
> 07:41:32:738 http client (127.0.0.1) handled 1 requests
> 07:41:32:740 ~http client 0x0x11f2088
> 07:41:32:787 HTTP client hdr detected:
> GET /api/59C37F236F/sensors/new HTTP/1.1
> Host: 192.168.0.46
> Connection: keep-alive
> Accept: vnd.ddel.v1
> User-Agent: Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.197 Safari/537.36
> X-Requested-With: XMLHttpRequest
> Referer: http://192.168.0.46/pwa/commissioning.html?1608764346000
> Accept-Encoding: gzip, deflate
> Accept-Language: fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7
> 
> 
> 07:41:32:788 HTTP API GET /api/59C37F236F/sensors/new - 192.168.0.46
> 07:41:32:788 ApiMode: 0
> 07:41:32:789 {"lastscan":"active"}
> 07:41:32:847 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 000202020004000002bc
> 07:41:32:847 Tuya debug 5 : Status: 0 Transid: 2 Dp: 514 (0x02,0x02) Fn: 0 Data 700

are you looking for something like that ?

If it's necessary, I can learn how to that with an other method. Maybe it will be easier in french and with voice !

Smanar, 2021-02-23:
In french, and without spamming issue, yes, If you have discord, I m Smanar on the deconz one. For voice, need to find the microphone but can be possible too.

Because It s realy strange, there is nothing on your log about the device, except "Tuya debug 4" but thoses one are tuya specific.
For exemple
```
set phoscon in permit join (add new light)
read the basic attribute again in deconz, cluster "basic" 0x0000, press button "read".
```

On log we can see 
>07:28:07:719 send permit join, duration: 59

But nothing for the read attribute, the command have filled the table on your side ?

And when you are trying to include it manualy, we can see it for sensor part
>19:08:17:728 device announce 0x60A423FFFEEC6B6F (0xAA2B) mac capabilities 0x8E

But realy nothing for "light" part. I think I will add some more log in code for a better trace, because it s seriously something magic for me.

As you don't use third app, I think we will use direclty the API to check device.
For you it s easy, just use a browser with this url (made from your capture)
`http://192.168.0.46/api/59C37F236F/lights`

You will a JSON with all your lights, can you find the dimmer one inside ?

choss38, 2021-02-23:
Here is the json part with the rotary dimmer:

`"4":{"etag":"b9540050ae1c9299a3ddfcef9af4e47c","hascolor":false,"lastannounced":"2021-02-22T19:14:22Z","lastseen":"2021-02-23T18:26Z","manufacturername":"_TZE200_dfxkcots","modelid":"TS0601","name":"Smart plug 4","state":{"alert":"none","bri":17,"on":true,"reachable":true},"swversion":"0xFFFFFFFF","type":"Dimmable light","uniqueid":"60:a4:23:ff:fe:ec:6b:6f-01"}}`

choss38, 2021-02-23:
Did you notice in the previous http debug log 

> Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 620

Data is the numeric value of the brightness.

choss38, 2021-02-23:
I retried the manipulation of this morning.
```
set phoscon in permit join (add new light)
read the basic attribute again in deconz, cluster "basic" 0x0000, press button "read".
```
Nothing in the info log :
> 18:36:11:515 send permit join, duration: 59
> 18:36:11:516 send GP proxy commissioning mode
> 18:36:51:577 Current channel 11
> 18:36:51:594 CTRL got nwk update id 1
> 18:36:51:609 Device TTL 2233 s flags: 0x7
> 18:37:10:515 send permit join, duration: 59
> 18:37:10:516 send GP proxy commissioning mode
> 18:37:51:581 Current channel 11
> 18:37:51:597 CTRL got nwk update id 1
> 18:37:51:613 Device TTL 2173 s flags: 0x7
> 18:38:09:515 send permit join, duration: 59
> 18:38:09:516 send GP proxy commissioning mode
> 18:38:51:585 Current channel 11
> 18:38:51:601 CTRL got nwk update id 1
> 18:38:51:617 Device TTL 2113 s flags: 0x7
> 18:38:57:613 dev /dev/ttyS0 (/dev/ttyS0)
> 18:38:57:618 GW firmware version: 0x26390500
> 18:38:57:619 GW firmware version is up to date: 0x26390500
> 18:39:08:516 send permit join, duration: 0

and the attributes are filled:
[image]

Found this in info2 log:

> 18:42:35:518 Poll light node Smart plug 4
> 18:42:35:881 Poll APS request to 0x60A423FFFEEC6B6F cluster: 0x0008 dropped, cluster doesn't exist
> 18:42:36:011 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x9EE5
> 18:42:36:014 enqueue event attr/lastseen for /lights/4
> 18:42:36:020 Websocket 192.168.0.2:38536 send message: {"attr":{"id":"4","lastannounced":"2021-02-22T19:14:22Z","lastseen":"2021-02-23T18:42Z","manufacturername":"_TZE200_dfxkcots","modelid":"TS0601","name":"Smart plug 4","productid":"Tuya_DIMSWITCH Earda Dimmer","swversion":"0xFFFFFFFF","type":"Dimmable light","uniqueid":"60:a4:23:ff:fe:ec:6b:6f-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"60:a4:23:ff:fe:ec:6b:6f-01"} (ret = 387)
> 18:42:36:515 enqueue event config/localtime for /config/
> ...
> 18:44:01:516 Poll light node Smart plug 4
> 18:44:01:524 Node data 0x60a423fffeec6b6f profileId: 0x0104, clusterId: 0x0000%
> 

I don't know how to transmit to you the http log, should I paste its numerous line here ?

Smanar, 2021-02-23:
Ha nice, we should have started by that

```
{
   "etag":"b9540050ae1c9299a3ddfcef9af4e47c",
   "hascolor":false,
   "lastannounced":"2021-02-22T19:14:22Z",
   "lastseen":"2021-02-23T18:26Z",
   "manufacturername":"_TZE200_dfxkcots",
   "modelid":"TS0601",
   "name":"Smart plug 4",
   "state":{
      "alert":"none",
      "bri":17,
      "on":true,
      "reachable":true
   },
   "swversion":"0xFFFFFFFF",
   "type":"Dimmable light",
   "uniqueid":"60:a4:23:ff:fe:ec:6b:6f-01"
}
```

The integration is perfect
>"bri":17,
>"type":"Dimmable light",

We don't need more, I think if you move the button the "bri" value is changing ?

>Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 620
>Data is the numeric value of the brightness.

Yep exactly from 0 to 1000

>07:41:29:950 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1

And this one if the on/off data = 1 > on 

For me all is fine, the problem is surely from phoscon, so you can stop re-inclusion.
BTW, you don't want to use a domotic application for your automation ^^ ?

Why phoscon see it as plug ..... Probably because of lidl stuff

choss38, 2021-02-23:
> I think if you move the button the "bri" value is changing ?
yes, it changes.
For my domotic I use Home Assistant. 
This device is not from Lidl, but from Aliexpress...

Smanar, 2021-02-23:
Yep, but lidl stuff are tuya device too, it s for that your have "fairy lights" as product.

Home assistant need to work with the device, you can't "link" them with the deconz version you are using for tests ?

choss38, 2021-02-24:
I tried to join HA to this Deconz but it failed: 

> Déverrouillez votre passerelle deCONZ pour vous enregistrer avec Home Assistant.
>     Accédez aux paramètres avancés du système deCONZ
>     Cliquez sur "Déverrouiller la passerelle"

I did unlock the gateway with Phoscon
Result: 
>  Impossible d'obtenir une clé d'API

choss38, 2021-02-24:
Ok, I did it. It was a lan problem.
[image]

Some strange behaviours:
- I can switch off the light with HA but it didn't stay off in HA, it goes back to on (only in HA, the device is really off)
- I can't switch back on from HA
- When I switch off the dimmer with its button, it switches off but HA didn't see it and shows it always on.
- I can change the brightness value from 0 to 100 with HA but I has no effect on the light. If the light was off, changing brightness in HA switches the light on.
- When I play with the brightness knob, from its minimal value to the biggest, HA shows change from 0 to ~22 only and after 22 it starts again from 0 ...
So it seems to be working in a big disorder !

Smanar, 2021-02-24:
>So it seems to be working in a big disorder !

Lol, yep, not sure I have understanding all.

So first the brightness .
>Tuya debug 5 : Status: 0 Transid: 1 Dp: 514 (0x02,0x02) Fn: 0 Data 620
>19:14:30:869 Tuya debug 4 : Address 0x60A423FFFEEC6B6F Payload 000202020004000001f4

This value go from 0 to 1000, I m taking it and make
`level = value * 254 / 1000;`
Because deconz is using value from 0 to 254, so with 620 bri need to be 157

>Payload 0002 0202 00 04 000001f4

In the payload you can see the command in "brut mode" 0202 = 0x02 0x02, 04 is the len and 000001f4 is the data in hexa format

>07:41:29:950 Tuya debug 5 : Status: 0 Transid: 1 Dp: 257 (0x01,0x01) Fn: 0 Data 1

0x01 0x01 is for the on/off, data = 1 mean on.

>09:49:08:099 Tuya debug 5 : Status: 0 Transid: 4 Dp: 515 (0x02,0x03) Fn: 0 Data 10

This one is a mystery for me, It seem the value is always 10 ? I have disabled it on the new code.

If I explain that, it s because, now you are able to see in log received command but send command too (starting by "Send Tuya request"). So the problem will be easy to see.

Have updated too the code a little. Will be faster now

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

choss38, 2021-02-25:
Switching on and off now work.
Changing brightness value in HA has no effect.
When I play with the brightness knob, HA shows change from 0 to ~22 only and after 23 it starts again from 0 ...

Smanar, 2021-02-26:
>Changing brightness value in HA has no effect.

Find the problem, I m using
```
if "on"
{
}
else if "bri"
{
}
```

so if you use "bri" and "on" , "bri was ignored.

>When I play with the brightness knob, HA shows change from 0 to ~22 only and after 23 it starts again from 0 ...

Find it too
I m using a quint8 var, so the values can be only 255 at max, so 255 *255 / 1000 = 65 maximum (not 22 but I think it will be better)

You can git pull the new code.

Smanar, 2021-03-01:
@choss38 Do you have give up ^^ ?

choss38, 2021-03-01:
oh, no ! Sorry I didn't have the time last days. I hope I could try tomorrow.

choss38, 2021-03-03:
Ok, It works perfectly ! Good job !

Each change on the dimmer is read by HA, each change in HA is done by the dimmer. 

I think you can now add your work to deconz api.

Is there a long process ?

Smanar, 2021-03-03:
Depend of the code, some of them are "sensible" so reserved to beta.
It seem we are missed the last beta planned the 01/03, but there is 1 beta and 1 stable every month.

BTW as all is working I have cleaned the coe to make the PR, if you can make a last try to confirm me all is working, before I make the PR ?

choss38, 2021-03-04:
There are some make errors:

```
pi@phoscon:~/deconz-rest-plugin $ git pull
remote: Enumerating objects: 28, done.
remote: Counting objects: 100% (28/28), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 28 (delta 21), reused 28 (delta 21), pack-reused 0
Dépaquetage des objets: 100% (28/28), fait.
Depuis https://github.com/Smanar/deconz-rest-plugin
   5bce1a22..dde87584  tuya_dimmer      -> origin/tuya_dimmer
 * [nouvelle branche]  multisensor      -> origin/multisensor
   7e8cec47..7e14a9b3  orvibo           -> origin/orvibo
   b39550a9..5b87847e  tuya_extender    -> origin/tuya_extender
   3f64f6ed..5048db07  tuya_siren_clean -> origin/tuya_siren_clean
Mise à jour 5bce1a22..dde87584
Fast-forward
 resource.cpp    |  3 ++-
 rest_lights.cpp | 20 ++++++++++----------
 tuya.cpp        | 44 ++++++++++++++++++++++++++++++++------------
 tuya.h          |  4 ++--
 4 files changed, 46 insertions(+), 25 deletions(-)
pi@phoscon:~/deconz-rest-plugin $ qmake && make -j2
make -f Makefile.Release
make[1] : on entre dans le répertoire « /home/pi/deconz-rest-plugin »
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/air_quality.o air_quality.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/authorisation.o authorisation.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/bindings.o bindings.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.10.00\" -DGW_SW_DATE=1614802421 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"dde87584dea7c86afc1fc52a20a36c88721717a9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/ [... cut ...] Port -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/appliances.o appliances.cpp
make[1]: *** [Makefile.Release:1457: release/tuya.o] Error 1
make[1]: *** Attente des tâches non terminées....
make[1] : on quitte le répertoire « /home/pi/deconz-rest-plugin »
make: *** [Makefile:40: release] Error 2
```

Smanar, 2021-03-04:
Corrected, just tested it, it compile now.

choss38, 2021-03-04:
It's ok, everything works.

Smanar, 2021-03-04:
No issue ? can make the PR ?

choss38, 2021-03-04:
I didn't see any issue.

choss38, 2021-03-13:
Hi,
I tried this morning, everything works fine. Thank you for your work.

Smanar, 2021-03-13:
Nice, thx, making PR

ersitan, 2021-04-07:
> Nice, thx, making PR

Hi,

I've the exact same dimmer with the same model name except my dimmer has stepless touch instead. I believe the configurations would be the same for both dimmers. Sorry for my lack of knowledge but I did not quite got what went above :) 

I have the deconz recognized the switch as smart plug as well. I can use it in the HA as on/off only. How would I be able to use the dimming in HA as well? Do I need to bind some attributes in deconz rest?

ersitan, 2021-04-07:
I've also played around with those values and now the physical dimming function (touch) does not respond at all.
[image]

Smanar, 2021-04-07:
The PR is not yet validated, so to test it you need to compile the code.
The tuya cluster from the GUI don't work correctly, it s normal, tuya don't use same "command receipe"

But hard to complile deconz on HA OS.

ersitan, 2021-04-07:
> The PR is not yet validated, so to test it you need to compile the code.
> The tuya cluster from the GUI don't work correctly, it s normal, tuya don't use same "command receipe"
> 
> But hard to complile deconz on HA OS.

Oh, I should wait for it to be validated then.  Well, I have windows OS that I can test the code if you guide me but it has already been done by the author of the post. So I do not know if that would be necessary.

ersitan, 2021-04-07:
@Smanar Is it possibla that forcing the values above in the screenshot would affect the physical dimming function?

ersitan, 2021-04-07:
resetting the power seems like solved the dimming issue

Smanar, 2021-04-07:
Yes, this panel "works" so you can send tuya request, but the format is not good, so the request is corrupted, and results can be hazardous.

closed 2021-04-18 by manup