dresden-elektronik/deconz-rest-plugin #4415: LEDVANCE 73741 Color Temperature Missing

benclaussen, 2021-02-13:
## Describe the question or issue you are having

I have 3x Sylvania/LEDVANCE Model 73741 BR30 Recessed smart bulbs paired. All functions except color **temperature** work properly. Clicking "Color temperature" in Phoscon web UI results in nothing.. no network requests, no log entries, no feedback in browser developer tools. Using the deCONZ app through VNC I can execute `Move to color temperature` commands successfully. But otherwise have no control of temp.

As far as I can tell these are running the latest firmware 0x00102428.

Ideas to expose color temperature?

## Screenshots

## Environment

 - Host system: Ubuntu
 - Running method: Home Assistant deCONZ Add-on (docker container)
 - Firmware version: 26680700
 - deCONZ version: 2.09.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: No

## deCONZ Logs

Can provide if necessary?

## Additional context

Most of my zigbee network is Sylvania/Lightify/LEDVANCE. 

21 routers, 5 end devices. Ikea buttons, eWeLink outlets, Innr outlet, and Sylvania bulbs

Mimiix, 2021-02-15:
@benclaussen Is it posibble to change the CT with HA?

benclaussen, 2021-02-15:
No, HA does not see color temperature at all.

> @benclaussen Is it posibble to change the CT with HA?

benclaussen, 2021-03-09:
I migrated to ZHA and it works 🤷‍♂️

closed 2021-03-09 by benclaussen