dresden-elektronik/deconz-rest-plugin #4416: Problems with Aqara motion sensor and temperature sensor after Deconz 2.09.03

fribse, 2021-02-14:
## Describe the bug

After this update I lost quite a few units, it was clearly updating a lot of IKEA lights, they were flashing like crazy, some wouldn't work after the update (10W led driver that I was running on 12 V, had to solder a resistor on it to get it to wake up again, as it's built to 24V).
But quite a few of my aqara motion sensors 'fell off the network' as well as some temperature and door sensors.
Most of them I've been able to add back into it, but currently two of them are acting up. One is a motion sensor, it doesn't get all the clusters shown, and this even though it is shown as a motion sensor, it doesn't have a motion part in the overview.
It also shows 'ffff' at the top of the window.
[image]
[image]

The other is a temperature sensor that I can't get to join the network at the moment.

## Steps to reproduce the behavior

  1. Upgraded to 2.09.03
  2. A lot of firmware updates going on
  3. Wait a day
  4. Devices seems to drop of the network

## Expected behavior

The devices shouldn't just disappear, or loose network connection.

## Screenshots

## Environment

 - Host system: Intel NUC running PVE
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.09.03)
 - Device: ConBee II
 - Do you use an USB extension cable: USB3 cable, 1 m
 - Is there any other USB or serial devices connected to the host system? AeoTec Z-wave 

## deCONZ Logs

## Additional context

SwoopX, 2021-02-14:
Check out https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

fribse, 2021-02-15:
I will, but everything is extremely unstable at the moment, my IKEA bulbs drop off, the contacts drop off, everything is rather unusable. I powercycle them, and they get back online, but only for a short while.

fribse, 2021-02-15:
This is pure s..., I'm sorry but this is very frustrating, things ping pong on and off the network of the bulbs, aqara sensors everything going to hell, what is going on, now my 1-gang buttons (WXKG06LM) won't talk to the network, they are in the network, but doesn't send anything through the network when I press the button, nothing is received in Home Assistant.

thecem, 2021-02-16:
Try 26670700 i have same problems with 26660700 and 26680700.

fribse, 2021-02-16:
> 26670700

I @thecem Thankyou for the suggestion, I'm going to ask if I can get the Home Assistant Add on updated to that version, not sure that it will be done, I hope so, but you never know.

Mimiix, 2021-02-17:
The firmware is on the stick, not the HA addon.  I suggest following : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually . The addon is currently not capable of updating the firmware.

thecem, 2021-02-17:
I had problems with 2.7.1 and 26680700 after downgrade to 26670700 they are gone (no repairing of aqara), after updating to 2.9.3 in HA (6.7.0) and updating the conbee II on a win10 Maschine to 26680700, they are gone.
@Mimiix is there some Funktion to use the md5 file for checking with GCFFlasher?

fribse, 2021-02-22:
Seems like the network is stabilizing itself, devices are not falling of the network all the time at the moment.
One thing is odd though, my IKEA blinds sometimes 'falls off' the network in the sense that they fall asleep, and doesn't wake up until the buttons on the curtain itself is pressed.

fribse, 2021-02-22:
And it seems like the update frenzy going on of the devices was the culprit of this very unstable network, maybe it would be better to do this in stages?

Mimiix, 2021-02-22:
@thecem Don't ask me , i have no clue how that works.

closed 2021-03-21 by fribse