dresden-elektronik/deconz-rest-plugin #4417: Problem with hvac_modes on Tuya/Moes TRV (TS0601) after update to deCONZ 2.9.3

kuszek, 2021-02-14:
## Describe the bug
After updating from deconz 2.7.1 (HA addon 6.6.2) to deconz 2.9.3 (HA addon 6.7.0) I had encountered a problem with Tuya/Moes TRV not reacting to the changes of hvac_modes "heat" or "off".
hvac_mode is changed in Home Assistant but a TRV doesn't start to heat or doesn't turn off accordingly.

What is more interesting, when I change the preset physically on the valve to auto or holiday (hvac_mode "auto"), it is represented in HA immediately. It doesn't work the other way around (when a change is done in HA).
Changing of the temperature in HA or physically on the valve works fine in both ways.

I tried the following:
- using the service deconz.device_refresh
- restarting deCONZ addon
- restarting Home Assistant
- re-adding of a TRV to deCONZ (turning paring mode on TRV again)
- resetting a TRV (using buttons combination and by changing the batteries)
- removing deCONZ node and re-adding of a TRV a couple of times

After removing the nodes and paring them again I also noticed something strange. They all get the name "Fairy lights" instead of "Thermostat".

## Steps to reproduce the behavior
  1. Change thermostat's hvac_mode to "heat" or "off".
  2. hvac_mode in HA changes but the TRV doesn't respond to the change.

## Expected behavior
hvac_mode or preset changes to "heat" or "off" and the device reacts to the change.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi 4B
 - Running method:  Home Assistant deCONZ Add-on
 - Firmware version: 26680700
 - deCONZ version: 2.09.03 / 12/23/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes, APC UPS

Smanar, 2021-02-14:
Yes, It s my fault.
I have break something on last deconz version, it impact "mode" and "preset" for all tuya stuff that have "TS0601" as model ID.
The corrective is already in PR list, but I don't know yet for when it will be available.

Mimiix, 2021-02-15:
@Smanar What PR is it?

Smanar, 2021-02-15:
This one https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4377
Already merged, but not sure for result yet.

MrHasse, 2021-02-15:
> This one #4377
> Already merged, but not sure for result yet.
Having the same issue with _TZE200_c88teujp/TS0601, tried 2.10.0 but its still the same.

Smanar, 2021-02-16:
Ha ? Not good news.
There is someone with the version 2.10.0, and still have the bug that can share some logs during inclusion ?
Just with "info" and "info_l2" .

Can add more debug line, if you can recompile code too ?

KartoffelToby, 2021-03-12:
Sounds like the same problem: https://github.com/home-assistant/core/issues/47768

fi-sch, 2021-03-12:
Yup, it's still valid and I'm waiting for fix as well.

Smanar, 2021-03-12:
What is your deconz version ?
The problem is corrected now.

fi-sch, 2021-03-12:
> What is your deconz version ?
> The problem is corrected now, you still have problem with "mode" or "preset" ? What is your device manufacture name ?

I'm on latest stable – 2.9.3 using Home Assistant add-on. So I'm waiting for DE to release stable (expected 10th of March) and then add-on to be updated.

Anyways, heating season is ending soon and I'm finally moving to zigbee2mqtt  this spring. So, TBH I personally don't care that much about this stable being released ASAP. I got used to how things are going with deCONZ ;)

kuszek, 2021-03-25:
I updated deconz to 2.10.4 (HA addon 6.8.0) yesterday and the valves started working properly again. Issue can be closed.

Mimiix, 2021-03-25:
Okido!

closed 2021-03-25 by Mimiix