dresden-elektronik/deconz-rest-plugin #4419: Unbind Busch-Jaeger wall-mounted switch (6737/RM01) from conbee coordinator

krombipils, 2021-02-14:
## Describe the question or issue you are having

I'm trying to use a Busch-Jaeger wall-mounted  4-channel transmitter (6737/RM01) in a conbee/deconz controlled zigbee network. On this device it is possible to store and recall a dimmable scene without using a coordinator (see [docu](https://www.busch-jaeger.de/files/files_ONLINE/FB01_6735_6736_6737_ZigBee%20Bedienelement_DE_EN_FR_alte%20Norm.pdf)).

I already used this approach in zigbee2mqtt. Unfortunately the number of bindings is very limited on this devices (see [https://github.com/Koenkk](https://github.com/Koenkk/zigbee-herdsman-converters/pull/879/commits/2993691061517519b3839f45819f2d8fece783c1)) Therefore in zigbee2mqtt you have to remove the bindings to the coordinator for this to work.

Deconz also creates LEVEL_CLUSTER_ID/SCENE_CLUSTER_ID bindings when pairing this device ([https://github.com/dresden-elektronik](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/e8748a6901484462af25b98a542ca1fb7b9703e4/bindings.cpp)).
I tried to remove these binding with the binding dropbox in deconz. Unfortunately this fails with: no entry. Am I doing something wrong? Are there other workaround/solutions?

## Screenshots

## Environment

 - Host system: Intel Nuc
 - Running method:  Marthoc Docker container
 - Firmware version: 0x26340500
 - deCONZ version: 2.09.03
 - Device: ConBee
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

SwoopX, 2021-02-14:
You need to unbind them from the corresponding groups given in the light resources. Please note that you need to use the hex value of the group ID.

krombipils, 2021-03-02:
Thank you @SwoopX!

I used Postman to query the groups and found an 'invisible' group created for my  4-channel transmitter. After unbinding from this group, I was able to programm dimmable scenes directly on the wall switch (on all 4 rows).
This is probably not the best solution, but I like the possibility to control my lights even if the coordinator is down...

closed 2021-03-02 by krombipils