dresden-elektronik/deconz-rest-plugin #4420: Sage doorbell is able to pair but does not stay on the network after (or something like that)

MattL0, 2021-02-14:
## Describe the bug

We just moved to a new place. I reinstalled the sage doorbell, but it was always ''unreachable'' and it didn't send anything to jowihue homeseer plugin.

I tried to pair a brand new sage doorbell but got the same result.

This had been flawless since the almost a year ( I added the sensors  at that time and never touched it) . I do not understand . I also tried to remove any trace of this sensor in the zll.db and repair it again, but no luck. Still not reachable and does not send anything to homeseer when the doorbell is pressed.

anyone have any insight?

## Steps to reproduce the behavior

## Expected behavior

1.Add a new sage doorbell sensors ( seen as a switch and sensors on phoscon btw) . It is added fine each time
2.Ring the doorbell
3.Check the status of the sensors on phonscon sensors tab
4. Poll the sensors with postman  http://ipadress:xxxx/api/Exxxxxxx/sensors/83
which return `{
    "config": {
        "group": "2",
        "on": true,
        "reachable": false
    },
    "ep": 18,
    "etag": "d205cb3a2570b80059b57889cda45a2b",
    "lastseen": null,
    "manufacturername": "Echostar",
    "mode": 1,
    "modelid": "Bell",
    "name": "Doorbell Sensor",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "swversion": "3.2.2",
    "type": "ZHASwitch",
    "uniqueid": "00:08:89:00:01:d9:dc:13-12-0006"
}`

## Screenshots

[image]
[image]

## Environment

 - Host system: Ubuntu 20.04.2 x64
 - Running method: Ubuntu
 - Firmware version: 26680700
 - deCONZ version: 2.09.03 / 12/23/2020 with -minimal tag in systemd .
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Which? Z-wave usb stick, ups , usb bluetooth

## deCONZ Logs

## Additional context

only Philips hue routers (21) and something like 40 end devices ( mostly Aqara door/Aqara motion/hue motion)

MattL0, 2021-02-17:
This was due to drop in voltage. bad wire somewhere or transformer having to much resistance maybe. 

We have two doorbell. wired to the same chime. The weird part is when i press on the first(front) doorbell, I got the code 1002, and same thing for the second(back) doorbell. 

But if I only wire the yellow wire of the sage doorbell, i got 2002 code. And if I wire back the green and remove the yellow, then i got 1002. 
Not sure how the wire diagram is here.. but if everything is wired as the documentation shows, I only got 1002 from doorbell 1 and 2 . So I do not know were the person is ringing. 

I guess I have two option:

1- cut the wires from the back doorbell, make this switch a dry contact with a door sensors . This doorbell is easy too play with, I can't do that with the first one. 

2- Put a aqara button as a back doorbell. 

I close this.

closed 2021-02-17 by MattL0

Mimiix, 2021-02-24:
Thanks for providing the solution! Gonna help some users some day :)

oywino, 2021-03-23:
@MattL0 Hi, I'mt trying to pair my SAGE DoorBell to Phoscon/deConz but I'm having problems. It is visible in both and also in Home Assistant, but not reachable. Have you been able to make it work succesfully?