dresden-elektronik/deconz-rest-plugin #4424: New installation on fresh HA server, previously connected sensors cannot be discovered/reconnected, new sensors are not discovered

DayRunner131, 2021-02-15:
## Describe the question or issue you are having

New to Phoscon/Dconz. I'm running a 64bit HASS.OS installation on a Raspi4 4gb through a mSATA SSD. HA is stable. ConBee II is connected to a USB2 port on a 0.3m USB extension cable. System is running on a 3 amp power supply.

I setup my first network about a week ago and had 3 Aqara Temperature/Humidity sensors & a door sensor connected and reporting successfully to HA. I then attempted to connect an Ikea Symphonisk Control and two INNR SP 224 plugs, for which none could be paired. That's when the trouble started.

I started to notice temperature sensors not reporting. I attempted to change the channel in Phoscon which didn't work. Currently operating on channel 15. I then deleted one of the temperature sensors and reconnected it and this seemed to work up until I tried to connect the SP 224 plugs.

In Deconz, a random router showed up and all my devices started dropping off-line. I then couldn't delete and re-pair them to the network.

Since I was early into my installation, I decided to start over from scratch. I uninstalled Deconz from HA and removed the integrations completely. I rebooted my server without the ConBee II connected, then shut down the server again, plugged in the ConBee II, booted HA and reinstalled DeConz.

With this fresh installation, I got the normal Daylight device and the Phoscon-GW device. The ConBee II shows a firmware 26660700, version 2.09.03 / 12/23/2020, and seems to be functional in dconz. It is again operating on channel 15 and won't budge from that channel even after it says it's been successfully changed in Phoscon. Deconz channel mask shows on channel 15.

At this moment, I cannot connect/add ANY sensor or device, including the 3 temp/humidity sensors and one door sensor that was previously paired.

I'm scratching my head as I cannot on the surface find anything wrong with the installation in any of the debug logs.

I suspect there is some configuration stored in the ConBee II or on my server that did not get wiped/reset when I did the fresh install that is hanging up my new network.

Could I receive some help with troubleshooting this issue?

## Screenshots

## Environment

 - Host system: Raspi4 4gb on 256gb mSata SSD w/3amp power supply
 - Running method: Hass OS 64bit
 - Firmware version: 26660700
 - deCONZ version: 2.09.03
 - Device: ConBee II connected to USB2 
 - Do you use an USB extension cable: Yes, 0.3m
 - Is there any other USB or serial devices connected to the host system? If so: Z-wave dongle on adjacent USB2. SSD controller on USB3.0

## deCONZ Logs

 ************* lines indicated missed log information as it scrolled too fast.

[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] firmware.sh: executing... 
[09:38:43] INFO: GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II 
/dev/ttyACM2     | 0x1CF1 | 0x0030  | DE2411644  | ConBee II 
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 
[cont-init.d] firmware.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] novnc.sh: executing... 
[cont-init.d] novnc.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[09:38:43] INFO: Running the deCONZ OTA updater...
[services.d] done.
[09:38:43] INFO: Running the OSRAM LEdvance OTA updater...
[09:38:43] INFO: Websockify waiting for VNC to start
[09:38:43] INFO: Running the IKEA OTA updater...
[09:38:44] INFO: Starting VNC server (local/yes)...
[09:38:44] INFO: Starting websockify...
WebSocket server settings:
  - Listen on 127.0.0.1:5901
  - Flash security policy server
  - Web server. Web root: /usr/share/novnc
  - No SSL/TLS support (no cert file)
  - proxying from 127.0.0.1:5901 to 127.0.0.1:5900

**********
[... 376 lines cut ...]
09:41:15:443     <modelDescription>Philips hue compatible Personal Wireless Lighting</modelDescription>
09:41:15:443     <modelName>Philips hue bridge 2015</modelName>
09:41:15:444     <modelNumber>BSB002</modelNumber>
09:41:15:445     <modelURL>http://www.dresden-elektronik.de</modelURL>
09:41:15:445     <serialNumber>00212E06C079</serialNumber>
09:41:15:446     <UDN>uuid:62032361-78a4-4315-8c60-68ce8cbed103</UDN>
09:41:15:449     <presentationURL>index.html</presentationURL>
09:41:15:450     <iconList>
09:41:15:450       <icon>
09:41:15:450         <mimetype>image/png</mimetype>
09:41:15:451         <height>48</height>
09:41:15:451         <width>48</width>
09:41:15:451         <depth>24</depth>
09:41:15:451         <url>hue_logo_0.png</url>
09:41:15:452       </icon>
09:41:15:452     </iconList>
09:41:15:452   </device>
09:41:15:452 </root>
09:41:15:454 [INFO] - Found file containing button maps. Parsing data...
09:41:15:457 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
09:41:15:465 [INFO] - Button maps loaded.
09:41:15:467 dlg action: Read binding table
09:41:15:468 found node plugin: libde_rest_plugin.so - REST API Plugin
09:41:15:472 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
## Additional context

Mimiix, 2021-02-15:
I think you have power issues. Can you verify it works with a SD card?

DayRunner131, 2021-02-15:
It would take some time for me to set it up that way at the moment, but in the short term, I did snag some power meter values for your review.

Voltage never dipped below 5.0vdc at both the power supply & USB2 port with ConBee II.

Power from 3.0A USB-C supply to Raspi4 - 5.22V @1.02A max draw
Power from USB2 port to ConBee II - 5.17V @.029A Max draw

DayRunner131, 2021-02-15:
So after further investigation, the issue does have something to do with connecting the SSD to the Raspi4. Without pulling out my scope and looking at the 5vdc USB voltage ripple, I do think there is a power issue but it isn't because the Raspi4 USB power rail is over drawn.

I measured the SSD drawing 250ma, the ConBee II pulling 30ma, and the Zwave dongle pulling 35ma. The voltage stayed above 5vdc. At 315ma totals draw between the 3 connected devices, this is far less than the maximum of 1200ma that the Raspi can output over USB.

Right now all devices are functional with the Raspi running off the SSD, but I am using a powered USB hub to power the ConBee II and Zwave dongles. This provides the 5vdc to the dongles from an isolated separate power source.

At some point I'll pull out my scope and look at this more, but in the meantime, this seems to be a solution for those who run of SSD drives and encounter power issues with the ConBeeII. Perhaps Dresden-Elektronik also needs to examine the potential for more power filtering into the dongle?

Mimiix, 2021-02-17:
@DayRunner131 It could also be the interferrence, but just not showing up at the logs.

closed 2021-03-18 by github-actions[bot]