dresden-elektronik/deconz-rest-plugin #4429: Sunrichter SR-ZG9001K8-DIM

SveLil, 2021-02-16:
## Device

- Product name: Sunrichter Single Color Wall Mounted 4 Groups Zigbee Push Button Remote SR-ZG9001K8-DIM
- Manufacturer: Sunrichter
- Model identifier: ZG2833K8_EU05
- Device type :
  - Switch

## Screenshots

[image]
[image]
[image]
[image]

Note: Battery status is only send after pressing both buttons of group 2:
[image]

Click and long press on each button produces:
```
ZG2833K8_EU05, broadcast to: 0xFF01, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 10
ZG2833K8_EU05, broadcast to: 0xFF01, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0032, zclSeq: 8
ZG2833K8_EU05, broadcast to: 0xFF01, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 9

ZG2833K8_EU05, broadcast to: 0xFF01, endpoint: 0x01, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 11
ZG2833K8_EU05, broadcast to: 0xFF01, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0132, zclSeq: 12
ZG2833K8_EU05, broadcast to: 0xFF01, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 13

ZG2833K8_EU05, broadcast to: 0xFF02, endpoint: 0x02, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 14
ZG2833K8_EU05, broadcast to: 0xFF02, endpoint: 0x02, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0032, zclSeq: 15
ZG2833K8_EU05, broadcast to: 0xFF02, endpoint: 0x02, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 16

ZG2833K8_EU05, broadcast to: 0xFF02, endpoint: 0x02, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 24
ZG2833K8_EU05, broadcast to: 0xFF02, endpoint: 0x02, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0132, zclSeq: 25
ZG2833K8_EU05, broadcast to: 0xFF02, endpoint: 0x02, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 26

ZG2833K8_EU05, broadcast to: 0xFF03, endpoint: 0x03, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 31
ZG2833K8_EU05, broadcast to: 0xFF03, endpoint: 0x03, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0032, zclSeq: 32
ZG2833K8_EU05, broadcast to: 0xFF03, endpoint: 0x03, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 33

ZG2833K8_EU05, broadcast to: 0xFF03, endpoint: 0x03, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 34
ZG2833K8_EU05, broadcast to: 0xFF03, endpoint: 0x03, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0132, zclSeq: 35
ZG2833K8_EU05, broadcast to: 0xFF03, endpoint: 0x03, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 36

ZG2833K8_EU05, broadcast to: 0xFF04, endpoint: 0x04, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 37
ZG2833K8_EU05, broadcast to: 0xFF04, endpoint: 0x04, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0032, zclSeq: 38
ZG2833K8_EU05, broadcast to: 0xFF04, endpoint: 0x04, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 39

ZG2833K8_EU05, broadcast to: 0xFF04, endpoint: 0x04, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 40
ZG2833K8_EU05, broadcast to: 0xFF04, endpoint: 0x04, cluster: LEVEL_CONTROL (0x0008), command: MOVE_WITH_ON_OFF (0x05), payload: 0132, zclSeq: 41
ZG2833K8_EU05, broadcast to: 0xFF04, endpoint: 0x04, cluster: LEVEL_CONTROL (0x0008), command: STOP_WITH_ON_OFF (0x07), payload: None, zclSeq: 42
```

## Picture
[image]

MaartenvMegen, 2021-02-16:
I have the same device. In a previous deconz version it would expose button events over the API and I was able to use it in home assistant. Deconz version 2.9.3 does not seem to do this. Perhaps something got more strict in checking models ID's? Because there is a model ID "ZG2833K" in button_maps.json which looks identical in configuration at first glance.

SveLil, 2021-02-16:
Yes, looking at the source code, ZG2833K is in there. But I can't use it in the Phoscon Webapp either.

phasperhoven, 2021-02-18:
Same here. I have two of those mounted to the wall that used to work but now are broken.

I raised an issue (#4439). I probably should have seen this issue one before, although in my opinion it's not really a Device Request but rather a Bug as it used to work before:-).

SwoopX, 2021-02-18:
@SveLil I can't spot anything odd based on the given information. However, from the debug output, you removed the most important part at the beginning of the line.

SveLil, 2021-02-18:
The front part is:
`[INFO] - No button map for: `

SveLil, 2021-02-18:
If I should debug anything, please let me know

SwoopX, 2021-02-18:
The fix should be fairly simple: In `button_maps.json`, find the entry `ZG2833K` and replace it with your exact model ID from the basic cluster. After a deconz restart, it should all go back to normal.

I'm a bit puzzled tbh that it doesn't work at all since I was under the impression, a loose check on the modelID would follow the exact check when no match is found.

SveLil, 2021-02-18:
Yes, this fixes the switch in openHAB (and my guess is, in HA, too).
It still doesn't show up in the Phoscon "Switch Editor", even though it is registered as a switch

phasperhoven, 2021-02-19:
Yes, this fixes it in HA too. 

See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4439#issuecomment-781724968. 

But it is more a workaround than a fix IMHO. It would make sense to have a loose check as SwoopX suggests.

SveLil, 2021-02-19:
@SwoopX Do you want to keep this issue or #4439 (as it is a bug and not a new device request)?

SwoopX, 2021-02-20:
I'd say let's close this one here

Mimiix, 2021-02-22:
@SwoopX This one?

closed 2021-02-22 by SwoopX