dresden-elektronik/deconz-rest-plugin #4432: Develco siren SIRZB-110

tyr84, 2021-02-16:
## Describe the bug

I've added the siren, and it shows up as a light. However, I can only activate the siren through the deconz interface:
[image]

I've tried the same method as the develco smoke alarms, setting state-alert to select or lselect, both in the api, and through the deconz.configure service in home assistant, but there is no response from the siren.
[image]
[image]

Am I doing something wrong, also is it possible to reach the deconz functionality shown in the first screenshot through the api? (selecting warning mode etc)

By further testing and referencing #3232 in addition to the tech-manual referenced above, I've come to the conclusion that the bit mapping is off for this device

Warning mode 2 and 3 seems to act as the 2-bits controlling siren level, while siren level 0, 1 and strobe is somehow mapping to the 4-bits that control warning mode. The exact mapping is unclear for me

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: 26580700
 - deCONZ version: 2.7.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Aeotec Z-wave

## deCONZ Logs

## Additional context

SwoopX, 2021-02-16:
Hm, it seems that the correct settings for Develco devices somehow got lost almost a year ago... Gotta check that on my smoke sensors as well.

The mapping of any bits isn't that hard as the specs define it (see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1593#issuecomment-504700737). However, finding out what the device supports is a bit challenging.

Can you pleae check if Warning Mode 1 + Strobe + Siren 1 makes "some noise"?

tyr84, 2021-02-18:
No sound, just an error (see screenshot)

[image]

SwoopX, 2021-02-18:
I assume this is due to a duration of 0 seconds?

tyr84, 2021-02-19:
Doesn't seem so:
[image]

Just to make sure there is nothing else wrong. This works (makes sound):
[image]

SwoopX, 2021-02-20:
Ok, so I just checked the SMSZB smoke detector. The "old" version back in the days seemed to be rather sensitive on the combination to pick so that any audio alert goes off. Now I only have the newer ones and they pretty much work with an arbitraty combination.

The siren apparently is also a bit sensitive. Could you check once again if this combination gives you some audio?

[image]

tyr84, 2021-02-20:
Yep, it gives some sort of siren, pretty low "taaa-daaa-taaa-daaa" kinda sounds like a fire truck siren :-)

Warning mode seems to control siren level, Selecting Warning mode 2 and 3 seems to give high level
This seems to be in accordance with the tech manual:

[image]
[https://www.elektroimportoren.no/docs/lib/4514809-Manual-18.pdf]

As far as I understand your code, the bitmap is controlled by:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/5b8d1a64180b98daa17414671a90af7eef3b7df9/general.xml#L3049-L3067

Which seems to flip the Warning mode and siren level

As far as I can tell Develco implements the zigbee standard for this command:
[image]

Disclaimer: I might be totally wrong, since I can't find any info on the bmp8 format

SwoopX, 2021-02-20:
@tyr84 
Gee, the first documentation from them I haven't checked for those details, as none of them has a solid info about that and then it turns out to be decent... first-world problems. However, they still got those challenges all over their docs ;)

REgardless, so it seems like the device doesn't support strobe and is allergic to using it. What actually controls the loudness is the siren level and not the warning mode (-> error in Develco documentation). So I'd let it go of on Burglar on highest audio level then.

tyr84, 2021-02-21:
Cool!

While on the subject, has there been any discussion to expanding the "alert" parameter?

It bugs me that this implementation "dumbs down" the siren to only one setting with short and long duration.

Given the IAS WD spec something like:
`
alert: {
  warning_mode: (int),
  strobe: (bool),
  siren_level: (int),
  warning_duration: (int),
  strobe_duty_cycle: (int),
  strobe_level: (int),
}
`
or if complexity of classes is an issue, mapping warning mode to brightness, and siren level to saturation or something similar.

ebaauw, 2021-02-21:
> As far as I understand your code, the bitmap is controlled by:

The definition of the payload defines _Warning Mode_, _Strobe_, and _Siren Level_ as separate `uint4` and `uint2` attributes, packed into a `map8`.  There's more places in the ZCL standard where they do this, but the GUI doesn't support it.  Hence I decided to expose the individual bits, so you get to play assembler yourself.

Note that Bitron devices seem to implement the bit order backwards (e.g. _Siren Level_ is bits 6 and 7 instead of 0 and 1).

Different devices might define _Siren Level_ values differently (particularly whether 0 is no sound vs low sound).

> While on the subject, has there been any discussion to expanding the "alert" parameter?

My bad, today I would probably have exposed the sound as `state.howl` and the light as `state.strobe`, rather than re-(ab-)using `state.alert`.  _Duration_ can be set through `state.ontime`.  I have yet to see a siren that uses different sound patterns for different _Warning Mode_ values, and different sound volumes for different _Siren Level_ values, hence no effort to expose these.  The Heiman siren supports _Stobe Duty Cycle_, but I haven't seen any others that do, and mine broke.

You can set the following `state.alert` values:

Value | _Warning Mode_ | _Strobe_ | _Siren Level_ | Comments
--- | --- | --- | --- | ---
`none` |  0 | 0 | 0 | Alarm off
`select` | 1 | 1 | 3 | Alarm for 1 sec
`lselect` | 1 | 1 | 3 | Alarm for 5 minutes*.
`blink` | 0 | 1 | 0 | Blink for 5 minutes*.
`all` | 1 | 1 | 3 | Alarm for 1 sec*, broadcasted to all devices.

*) Use `state.ontime` to set a different _Duration_.

tyr84, 2021-02-22:
Thanks for the reply!

Actually, playing around with the GUI settings on the Develco siren i get at least 3 different sounds, but since either the Develco device or GUI is not complying with ZCL (bits seems reversed, with siren level controlling warning mode and vice versa) it's non-trivial to exhaust the modes specified in the Develco manual.

Do you know if the GUI is compliant with ZCL? Looks like the Develco manual is. My impression is that warning mode 1-4 should represent individual bits in the uint4?

tyr84, 2021-04-21:
This was not fixed.

Currently running 2.10.4, which should include the bugfix.

Current state is that "select" gives of a short beep (not loud), "lselect" makes the beep last a second or so longer, with the same volume.

Mimiix, 2021-05-10:
Re-opened on request of @tyr84

closed 2021-06-09 by github-actions[bot]

jarlesb, 2021-08-17:
The Develco Siren is still not working any chance you could look at it again?

This seems to be better aligned with the Develco spec, or did I miss something?

<command id="0x00" dir="recv" name="Start warning" required="m">
	<payload>
		<attribute id="0x00" type="bmp8" name="Options" required="m">
			<value name="Warning mode 0" value="0"></value>
			<value name="Warning mode 1" value="1"></value>
			<value name="Warning mode 2" value="2"></value>
			<value name="Warning mode 3" value="3"></value>
			<value name="Strobe mode 0" value="4"></value>
                        <value name="Strobe mode 1" value="5"></value>
			<value name="Siren level 0" value="6"></value>
			<value name="Siren level 1" value="7"></value>
		</attribute>
		<attribute id="0x01" type="u16" name="Warning duration" required="m"></attribute>
		<attribute id="0x02" type="u8" name="Strobe duty cycle" required="m" default="0x00"></attribute>
		<attribute id="0x00" type="enum8" name="Strobe level" required="m">
			<value name="Low" value="0x00"></value>
			<value name="Medium" value="0x01"></value>
			<value name="High" value="0x02"></value>
			<value name="Very high" value="0x03"></value>
		</attribute>
	</payload>
</command>
<command id="0x01" dir="recv" name="Squawk" required="m">
	<payload>
		<attribute id="0x00" type="bmp8" name="Options" required="m">
			<value name="Sqawk mode 0" value="0"></value>
			<value name="Sqawk mode 1" value="1"></value>
			<value name="Sqawk mode 2" value="2"></value>
			<value name="Sqawk mode 3" value="3"></value>
			<value name="Sqawk level 1" value="6"></value>
			<value name="Sqawk level 2" value="6"></value>
		</attribute>
	</payload>
</command>