dresden-elektronik/deconz-rest-plugin #4433: Axis Gear

pauljchristy, 2021-02-16:
## Describe the bug
I have a number of Axis Gear Shade Controllers. Cluster 8 (Level Control) id=0x0000 seems to report every so often incorrectly. I will get an report showing that the device level (bri) has changed even though it hasn't. Example:

I closed the shade - By setting lift to 100

"etag": "b7f2c8419440cd79e670bb7ab658887f",
    "hascolor": false,
    "lastannounced": "2021-02-15T16:22:42Z",
    "lastseen": "2021-02-16T19:08Z",
    "manufacturername": "AXIS",
    "modelid": "Gear",
    "name": "Shade BM2",
    "state": {
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": "102-5.3.5.1125",
    "type": "Window covering device",
    "uniqueid": "00:24:46:00:00:14:61:b7-01"
}
Then this a minute later:
{
    "etag": "32091993e9722947ab00b4cea1de4f82",
    "hascolor": false,
    "lastannounced": "2021-02-15T16:22:42Z",
    "lastseen": "2021-02-16T19:09Z",
    "manufacturername": "AXIS",
    "modelid": "Gear",
    "name": "Shade BM2",
    "state": {
        "bri": 0,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": "102-5.3.5.1125",
    "type": "Window covering device",
    "uniqueid": "00:24:46:00:00:14:61:b7-01"

Then this
    "etag": "0c7cf0c7473ba37aec21ad680552bc7f",
    "hascolor": false,
    "lastannounced": "2021-02-15T16:22:42Z",
    "lastseen": "2021-02-16T19:13Z",
    "manufacturername": "AXIS",
    "modelid": "Gear",
    "name": "Shade BM2",
    "state": {
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": "102-5.3.5.1125",
    "type": "Window covering device",
    "uniqueid": "00:24:46:00:00:14:61:b7-01"
}
  
The "lift is always correct. This cycle will repeat every 5 minutes.  I have remove Binding and removed reporting on all clusters except 102 and reporting works correctly until deconz is restarted or the reporting is automatically set to 1/300.  I have defaulted everything and then reinstalled.  

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

Version    2.09.01 / 12/23/2020
Firmware 26680700

Host system: Windows 10 Pro
Running method: Windows
Device: ConBee II
Do you use an USB extension cable: yes 
Is there any other USB or serial devices connected to the host system?  No 

## deCONZ Logs

## Additional context

Smanar, 2021-02-16:
Ha yep, realy, strange.
Easy to by pass, we can just ignore the cluster 0x0008 for this device, but the working mode is still strange.

You can 't catch some log just before this event ? With info and APS ?

> I have remove Binding and removed reporting on all clusters except 102 and reporting works correctly until deconz is restarted or the reporting is automatically set to 1/300

Mean it works after you have do that, but only till the restart ?

pauljchristy, 2021-02-16:
_Mean it works after you have do that, but only till the restart? Yes, It last until restart or approx 8 hours.

I will get the logs with info and APS.

pauljchristy, 2021-02-16:
Here is a log. Hopefully this what you are looking for

0x0024460000155732 Is a window shade

16:54:02:766 APS-DATA.request id: 128, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:54:02:813 APS-DATA.confirm id: 128, status: 0x00 SUCCESS
16:54:02:813 APS-DATA.confirm request id: 128 -> confirmed, timeout 1613512442
16:54:02:829 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
16:54:02:829 APS-DATA.indication request id: 128 -> finished
16:54:02:829 APS-DATA.request id: 128 erase from queue
16:54:08:125 APS-DATA.indication srcAddr: 0x576a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0102, lqi: 255, rssi: -58
16:54:08:125 [INFO] - No button map for: Gear endpoint: 0x01 cluster: WINDOW_COVERING (0x0102) command: 0x0A payload[0]: 008
16:54:08:125 ZCL attribute report 0x0024460000144E07 for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:54:08:735 APS-DATA.indication srcAddr: 0x8c93, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 183, rssi: -71
16:54:08:739 [INFO] - No button map for: Gear endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: 0x0A payload[0]: 000
16:54:08:740 ZCL attribute report 0x0024460000155732 for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:54:09:299 APS-DATA.indication srcAddr: 0x8c93, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0102, lqi: 183, rssi: -71
16:54:09:303 [INFO] - No button map for: Gear endpoint: 0x01 cluster: WINDOW_COVERING (0x0102) command: 0x0A payload[0]: 008
16:54:09:304 ZCL attribute report 0x0024460000155732 for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:54:09:516 APS-DATA.request id: 160, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:54:09:563 APS-DATA.confirm id: 160, status: 0x00 SUCCESS
16:54:09:563 APS-DATA.confirm request id: 160 -> confirmed, timeout 1613512449
16:54:09:579 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 183, rssi: -71
16:54:09:579 APS-DATA.indication request id: 160 -> finished
16:54:09:579 APS-DATA.request id: 160 erase from queue
16:54:11:766 APS-DATA.request id: 171, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:54:11:813 APS-DATA.confirm id: 171, status: 0x00 SUCCESS
16:54:11:813 APS-DATA.confirm request id: 171 -> confirmed, timeout 1613512451
16:54:11:829 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 183, rssi: -71
16:54:11:829 APS-DATA.indication request id: 171 -> finished
16:54:11:829 APS-DATA.request id: 171 erase from queue
16:54:14:016 APS-DATA.request id: 182, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:54:14:063 APS-DATA.confirm id: 182, status: 0x00 SUCCESS
16:54:14:063 APS-DATA.confirm request id: 182 -> confirmed, timeout 1613512454
16:54:14:078 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 183, rssi: -71
16:54:14:078 APS-DATA.indication request id: 182 -> finished
16:54:14:078 APS-DATA.request id: 182 erase from queue
16:54:16:266 APS-DATA.request id: 193, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:54:16:313 APS-DATA.confirm id: 193, status: 0x00 SUCCESS
16:54:16:313 APS-DATA.confirm request id: 193 -> confirmed, timeout 1613512456
16:54:16:329 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 183, rssi: -71
16:54:16:329 APS-DATA.indication request id: 193 -> finished
16:54:16:329 APS-DATA.request id: 193 erase from queue
16:54:18:516 APS-DATA.request id: 204, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:54:18:563 APS-DATA.confirm id: 204, status: 0x00 SUCCESS
16:54:18:563 APS-DATA.confirm request id: 204 -> confirmed, timeout 1613512458
16:54:18:579 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 183, rssi: -71
16:54:18:579 APS-DATA.indication request id: 204 -> finished
16:54:18:579 APS-DATA.request id: 204 erase from queue
16:54:19:750 APS-DATA.indication srcAddr: 0x474e, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 159, rssi: -74
16:54:19:750 APS-DATA.request id: 212, addrmode: 0x03, addr: 0x00244600001461b7, profile: 0x0104, cluster: 0x0019, ep: 0x01 -> 0x01 queue: 0 len: 4 tx.options 0x04
16:54:20:766 APS-DATA.request id: 217, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
16:54:20:813 APS-DATA.confirm id: 217, status: 0x00 SUCCESS
16:54:20:813 APS-DATA.confirm request id: 217 -> confirmed, timeout 1613512460
16:54:20:829 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 143, rssi: -76
16:54:20:829 APS-DATA.indication request id: 217 -> finished
16:54:20:829 APS-DATA.request id: 217 erase from queue
[... cut ...]
16:54:47:766 APS-DATA.request id: 89, addrmode: 0x03, addr: 0x00212effff0482d1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:54:47:813 APS-DATA.confirm id: 89, status: 0x00 SUCCESS
16:54:47:813 APS-DATA.confirm request id: 89 -> confirmed, timeout 1613512487
16:54:47:829 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
16:54:47:829 APS-DATA.indication request id: 89 -> finished
16:54:47:829 APS-DATA.request id: 89 erase from queue

Smanar, 2021-02-17:
Not realy, but don't worry

> 16:54:08:125 ZCL attribute report 0x0024460000144E07 for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000
> 16:54:08:740 ZCL attribute report 0x0024460000155732 for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000
> 16:54:09:304 ZCL attribute report 0x0024460000155732 for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000

>_Mean it works after you have do that, but only till the restart? Yes, It last until restart or approx 8 hours.

It s like you have a covering that use the 2 cluster and the other only one.

But Something easy to test, and not seem problematic is prevent deconz making reporting for the cluster 0x0008. For me it s useless the covering cluster is better.
Can you try this modification ?

```
sudo apt install deconz-dev
git clone --branch Gear_test https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

pauljchristy, 2021-02-17:
I’m on windows.

Thank you,

[/Users/paulchristy/Library/Containers/com.microsoft.Outlook/Data/Library/Caches/Signatures/signature_1555321239]
Paul Christy – Systems Specialist
248-835-7100

Remote Technical Support | IT Support
[cid:image002.jpg@01D70542.C19A3C10]<http://www.mistar.us/>  [cid:image003.jpg@01D70542.C19A3C10] <mailto:paul@mistar.us>

pauljchristy, 2021-02-17:
My Deconz is connected to a window 10 machine.

Smanar, 2021-02-17:
Arf, np, will wait a little to see if someone else can make tries, not the first time I m seing this device.

Smanar, 2021-02-23:
@seafoodrice , no problem on your side with this device ?

seafoodrice, 2021-02-23:
Hi Smanar,

It is good. Thanks Smanar.

closed 2021-03-24 by github-actions[bot]