dresden-elektronik/deconz-rest-plugin #4434: Unable to access to Web App

markicos, 2021-02-17:
I have been trying to get into the Web APP by using the User and Password but it seems that the user or password are not the correct one. i have no problem when getting into Phoscon App when i am request only for the Password, so i consider the problem is the user, any of you have an idea of the user i haver to use?

## Screenshots

[image]

## Environment

 - Host system: (Raspberry Pi 4 and PC  )
 - Running method: (Raspbian and Windows )
 - Firmware version: 26660700
 - deCONZ version: 2.09.03 / 23/12/2020
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## Additional context

The final aim of my development is to avoid get into phocon app to get an apiKey

SwoopX, 2021-02-17:
The username is `delight`. By default,  it's also the password unless you haven't set your own.

markicos, 2021-02-17:
Yes it is!! thank you very much

closed 2021-02-17 by markicos