dresden-elektronik/deconz-rest-plugin #4436: Provide API responses as JSON arrays and JSON lines

WTell, 2021-02-17:
## Feature request type

This is a request for responses from the REST API as JSON arrays and JSON lines, as opposed to the nested JSON object structure provided today.

## Description

Today, the API responses provided over HTTP (for example from `/api/KEY/sensors`) take the following form (prettified):

```
{
  "1": {
    "lastseen": "2021-02-15T05:02Z",
    "name": "Temperature Sensor A",
    "state": {
      "alert": "none",
      "level": 152,
      "on": true,
      "reachable": true
    }
  },
  "3": {
    "lastseen": "2021-02-15T05:01Z",
    "name": "Temperature Sensor B",
    "state": {
      "alert": "none",
      "level": 13,
      "on": false,
      "reachable": true
    }
  },
  "4": {
    "lastseen": "2021-02-15T05:00Z",
    "name": "Temperature Sensor Backup",
    "state": {
      "alert": "none",
      "level": 76,
      "on": true,
      "reachable": true
    }
  }
}
```

This structure makes sense when accessing the devices directly by ID (ie. `$.1.name`, `$.3.state.level etc.`), but it makes lense sense when performing iterative operations that will access the data in sequence, as these data engines typically expect the data to be presented as an array.

Many modern data ingestion engines and ETL (Extract, Transform and Load) tools expect JSON arrays or JSON lines as their input. In particular, the format presented today is incompatible with popular tools like Apache Druid, influxDB and Telegraf, Pentaho Data Integration (formerly Kettle) etc.

The basic principle of these ingestion tools is that data that has a repeating or iterative format – which is the case with the deCONZ REST API – should take the form of an array.

I suggest that as an URL parameter, the data could be requested presented as a JSON array, for by appending `?format=jsonArray` to the URL string (`/api/KEY/sensors?format=jsonArray`). This necessitates that the ID attribute is moved into the array object, which anyway makes sense as "ID" is a logically an identifier at the same hierarcical level as "name", "type", "uniqueid" and others.

An example of the data format that `?format=jsonArray` would provide is (again prettified for ease of reading):

```
[
  {
    "id": 1,
    "lastseen": "2021-02-15T05:02Z",
    "name": "Temperature Sensor A",
    "state": {
      "alert": "none",
      "level": 152,
      "on": true,
      "reachable": true
    }
  },
  {
    "id": 3,
    "lastseen": "2021-02-15T05:01Z",
    "name": "Temperature Sensor B",
    "state": {
      "alert": "none",
      "level": 13,
      "on": false,
      "reachable": true
    }
  },
  {
    "id": 4,
    "lastseen": "2021-02-15T05:00Z",
    "name": "Temperature Sensor Backup",
    "state": {
      "alert": "none",
      "level": 76,
      "on": true,
      "reachable": true
    }
  }
]
```

This feature would make ingestion of deCONZ REST API responses a lot simpler to deal with in many popular and modern data tools.

Further, many tools in the same category – data ingestion and ETL – work very well with, or expect, the data to be fed as individual lines logically equivalent to database rows. A popular JSON implementation of this is the [JSON Lines](https://jsonlines.org) format. In JSON Lines (also known as JSON-per-line), each line (`'\n'`) in the data feed is a logically separate JSON object in and of itself. JSON Lines would build on the same structure as the array format above, with the ID attribute inside the data object, but without square brackets.

I suggest that this format is implemented at the same time as the change suggested above, as they are very near each other in code and functionality. Similarly to the request attribute suggested above, I would put forth that this could be achieved by appending `?format=jsonLines` to the URL string (`/api/KEY/sensors?format=jsonLines`).

An example response data provided as JSON Lines would be:

```
{"id": 1,"lastseen": "2021-02-15T05:02Z","name": "Temperature Sensor A","state": {"alert": "none","level": 152,"on": true,"reachable": true}}
{"id": 3,"lastseen": "2021-02-15T05:01Z","name": "Temperature Sensor B","state": {"alert": "none","level": 13,"on": false,"reachable": true}}
{"id": 4,"lastseen": "2021-02-15T05:00Z","name": "Temperature Sensor Backup","state": {"alert": "none","level": 76,"on": true,"reachable": true}}
```

## Considered alternatives

The existing implementation prevents the deCONZ API from being compatible with several popular data ingestion and ETL tools without proxying or transforming the data before processing.

## Additional context

Additional context provided in the description above.

Dazanus, 2021-02-18:
Hi.
Came across this thread while looking on Google for a solution to this problem. I’m really looking for this as well, as the output from the API is not suited for import into my logging toolchain. The “?format=jsonLines” solution suggested above would be perfect!

Mimiix, 2021-02-18:
Dropped it into the Dev chat. I do like this proposal and i do see your problem (as i'm working with API's and ETL tools all day).

Wondering about your specific usecase though :)

WTell, 2021-02-20:
> Dropped it into the Dev chat. I do like this proposal and i do see your problem (as i'm working with API's and ETL tools all day).

Awesome! Thanks a bunch. I hope this is something that can be implemented, as I don't like the lack of elegance in hosting proxy-services for data transformation between implementation of standards (like JSON) that really _ought to be_ directly transferrable.

> Wondering about your specific usecase though :)

So my specific use case is ingestion of sensor data (room temperature and room occupancy) and consumption data (mainly electricity), for the purpose of analysis together with historical weather data, weather predictions, electricity prices (past and near future) and scheduling sources (calendars).

I've always been the guy that laughs when people drop terms like "AI" and "machine learning" into a conversation, but perhaps I'll have to eat my own words on this journey. But regardless of where the journey ends, it certainly begins with data!