dresden-elektronik/deconz-rest-plugin #4439: Sunricher 8-button switches stopped working (reformatted)

phasperhoven, 2021-02-17:
## Describe the bug
After moving my Homeassistent install to a new machine, I no longer receive deconz_event's from them in HA. I was running them on a Raspberry Pi but moved it all to a Supervised Install on Debian. I took a backup from deconz and restored that to the new install. Everything comes back but these events.

The switches are seen by deconz and by HA, but that only sees a battery level sensor. That is the same as before. Only the events are missing.

Another post here suggests it might have to do with the software version. I did not actively upgrade the firmware, but as I switched it from one machine to the other, it may have been running a different version before. I might have done the upgrade some time ago and not reboot, but I have no memory of that.

## Steps to reproduce the behavior
When in HA, I could go to Developer Tools->Events->"deconz_event"->Start Listening. This is how I found out what event codes to use in the first place. Now I get all other events like from some Aqara and Ikea switches but none from the Sunrichers.

## Expected behavior
I'd expect events coming in just like from other devices.

## Screenshots

## Environment
 - Host system: (old) Mac Mini, Debian 10, Supervised HA
 - Running method:  Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version:  2.09.03 / 23/12/2020
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? Yes: DSMR smart power meter.

## deCONZ Logs
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] firmware.sh: executing... 
[23:00:29] INFO: GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2131958  | ConBee II 
/dev/ttyUSB0     | 0x0403 | 0x6001  | P11I9SVS   | Generic FTDI 
[cont-init.d] firmware.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] novnc.sh: executing... 
[cont-init.d] novnc.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[23:00:29] INFO: Running the IKEA OTA updater...
[23:00:29] INFO: Running the deCONZ OTA updater...
[23:00:29] INFO: Websockify waiting for VNC to start
[23:00:29] INFO: Running the OSRAM LEdvance OTA updater...
[23:00:29] WARNING: Your direct VNC access is not protected!
[23:00:29] INFO: Starting VNC server (local/no)...
[23:00:30] INFO: Starting websockify...
WebSocket server settings:
  - Listen on 127.0.0.1:5901
  - Flash security policy server
  - Web server. Web root: /usr/share/novnc
  - No SSL/TLS support (no cert file)
  - proxying from 127.0.0.1:5901 to 127.0.0.1:5900
[23:00:32] INFO: deCONZ waiting for VNC to start
[23:00:33] INFO: Starting the deCONZ gateway...
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
23:00:33:795 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
23:00:33:800 CTRL. 3.27.223:00:33:932 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2131958-if00
23:00:33:961 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2131958-if00 / serialno: DE2131958, ConBee II
23:00:33:963 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
23:00:34:028 parent process s6-supervise
23:00:34:029 gw run mode: docker/hassio
23:00:34:029 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
23:00:34:030 DB sqlite version 3.27.2
23:00:34:030 DB PRAGMA page_count: 39
23:00:34:031 DB PRAGMA page_size: 4096
23:00:34:031 DB PRAGMA freelist_count: 0
23:00:34:031 DB file size 159744 bytes, free pages 0
23:00:34:031 DB PRAGMA user_version: 7
23:00:34:031 DB cleanup
23:00:34:032 DB create temporary views
23:00:34:041 started websocket server at port 8081
23:00:34:043 [INFO] - Found file containing button maps. Parsing data...
23:00:34:044 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
23:00:34:049 [INFO] - Button maps loaded.
23:00:34:049 dlg action: Read binding table
23:00:34:050 found node plugin: libde_rest_plugin.so - REST API Plugin
23:00:34:051 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
[23:00:34] INFO: Starting Nginx...
2021/02/17 23:00:34 [notice] 308#308: using the "epoll" event method
2021/02/17 23:00:34 [notice] 308#308: nginx/1.14.2
2021/02/17 23:00:34 [notice] 308#308: OS: Linux 4.19.0-14-amd64
2021/02/17 23:00:34 [notice] 308#308: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2021/02/17 23:00:34 [notice] 308#308: start worker processes
2021/02/17 23:00:34 [notice] 308#308: start worker process 1192

## Additional context
When I enter the container running the deconz addon, I find the following in the file /usr/share/deCONZ/devices/button_maps.json:
        "sunricherCCTMap": {
            "vendor": "Sunricher",
            "doc": "CCT remote",
            "modelids": [],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR_TEMPERATURE", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Move to color temperature"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_COLOR_TEMPERATURE", "0x013C", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Move color temperature up"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_COLOR_TEMPERATURE", "0x033C", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Move color temperature down"],
                [1, "0x01", "COLOR_CONTROL", "STOP_MOVE_STEP", "0", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_"]
            ]
        },

That empty array of modelids seems to be the problem, but I'm not sure if that is the cause of this and how to fix that...

TIA,
Peter

phasperhoven, 2021-02-19:
OK, I found a fix/workaround. I needed to add the model code to the button_maps.json inside the image (not the running container!). 

That means:
- stop the deconz container: docker stop addon_core_deconz
- find the button_map-file: find / -name button_map.json -print. (This should only give one result. When there are more the container is still running.
- cd /var/lib/docker/overlay2/<some long string of numbers>/diff/usr/share/deCONZ/device
- open the file in your favorite editor and find the sunricherMap-entry.
- In the list of modelID's is the code ZG2833K, replcae that with ZG2833K8_EU05 (or simply add it to the list)
- start the container again: docker start addon_core_deconz

That's it. Of course, if the image get overwritten by an upgrade, and this issue is not fixed, you'll have to go trough this procedure again.

If the ID is added to the running container, it does not work, as the button_map is read at startup and restarting builds a new container from the image.

As stated above, this is running hassio in supervised mode on linux, so you can get to the images fairly easily. I don't know if this applies to running hassos on raspberry pi (or even the problem even exists there.

I am new to this forum, so I am not sure whether to close this issue (it is fixed for me) or leave it open so it can be addressed in the next release.

closed 2021-03-03 by manup

mstormi, 2021-04-28:
sorry for being off topic but where did you obtain the Sunricher products from ?
Last time I tried getting one it was only possible directly from the vendor. Thanks!