dresden-elektronik/deconz-rest-plugin #4440: Device name

theCroh, 2021-02-18:
## Device

- Product name: Aqara Water Leak Sensor T1 SJCGQ12LM
- Manufacturer: LUMI
- Model identifier: lumi.flood.agl02
- Device type : 
  - Sensor

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Mimiix, 2021-02-18:
@SwoopX @Smanar Is this a easy add? I'd like it in for next beta as i suspect a lot of interest.

SwoopX, 2021-02-18:
@Mimiix The day Manu merges my door/windows sensor PR for the T1, I can easily add this one here as well. I'll reach out to him as I got several other dependencies on top of that.

closed 2021-03-03 by manup