dresden-elektronik/deconz-rest-plugin #4442: Double Tab Problem with Aqara Cube

bruesche, 2021-02-18:
## Describe the bug
  Hi,

today my Aquara Cube arrived. Connecting to Deconz work without any problems. The I recognized that the double tap action and the rotation action doesn't work. I found here a problem with the pairing and that I should shake and rotate the cube during the pairing. After that the rotation action was working. But the the double tap didn't work.
Can anyone give me a hint to solve "the problem"?

thx for help

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment
My Deconz version is 2.10.00 and the cube is shown in deconz with product number MFKZQ01LM.

## deCONZ Logs

## Additional context

SwoopX, 2021-02-18:
Please don't tell me you tapped your finder on it twice... :) You need to "smash" the cube itself twice.

bruesche, 2021-02-18:
Shame on me. 🤣

closed 2021-02-18 by bruesche