dresden-elektronik/deconz-rest-plugin #4443: Tuya ZigBee Signal Repeater - How to rename device in DeCONZ?

Sjonnie2018, 2021-02-18:
## Describe the question or issue you are having
I have a Tuya ZigBee 3.0 Signal Repeater which is working fine but in DeCONZ it shows as 0x47AE and I would like to change that to a more comprehensible name. Especially since I have another one ordered which is supposed to arrive soon. I have been looking at all cluster info screens (added below) but can't figure out if and how I can change that.

## Screenshots

Node info:
[image]

Cluster info:
[image]
[image]
[image]
[image]

## Environment

 - Host system: Raspberry Pi 3 Model B rev 1.2
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.09.03
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? A SSD on a seperate board (X820?) with own powersupply

## deCONZ Logs
N/A

## Additional context
N/A

Smanar, 2021-02-18:
Oups I have forget it https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4175

I think it s only possible if the device is in the API.

The modification is fast, do you have a linux machine to test a code modification ?

Sjonnie2018, 2021-02-18:
Hello Smanar,

I have a Linux machine as my main computer but it is not connected to the Conbee II stick. If I pull that out of the Raspberry Pi I have serious WAF issues tonight ;)

I read 4175 and have to say I don't need it in the API (a.k.a. show up in Domoticz). I would only like to rename it in DeCONZ.

Greetz,

Sjonnie

Smanar, 2021-02-20:
Ok so I have made a code modification, but the other user can't test it ^^, only windows machine.

So I can make the PR in blind mode, but If you can test it ?
It will make the device visible in the API, so with the name modifiable, based on last code version from official repo.

Sjonnie2018, 2021-02-21:
I can test for you but you have to help me there. Let me know what I should do.

Greetz,

Sjonnie

Smanar, 2021-02-21:
Ok so first to prevent WAF issue, make a backup of the file
`/usr/share/deCONZ/plugins/libde_rest_plugin.so`
In another folder, if you have problem, just replace this file at the same place.

So the code is based on the last deconz version, for explanation just go here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch tuya_extender https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then just re-include your device.

Sjonnie2018, 2021-02-21:
Hi Smanar,

Well my Pi has been compiling for over an hour and due to the WAF I had to reboot the device. Is it supposed to take so long?

Didn't get a specific error or such. The Pi was just sitting there "compiling" I think. I should note that the Pi also serves as my DNS/DHCP server (PiHole) so if it is compiling and trying to get information/data from the Internet that would explain.

Will try to start the compilation in the night so it has enough time to finish the job.

Bit strange though... that it would take so much time.

Smanar, 2021-02-21:
No, 1h is too much, but depend of the PI too.
If you have a smal pi remove the "-j2" 
`qmake && make`

On my Raspberry 3, need to use the command without "-j2" else it freeze totaly the system, but can interrupt using ctrl+C (don't need to reboot).

But don't destroy your OS for that ^^, if nothing on display after a time, give up, I can make the pr in blind mode, It can not work, but I don't think it can break something.

Sjonnie2018, 2021-02-22:
I will have a go at it this evening. If it fails we can try your compiled version; see what happens.

I couldn't cancel because the Pi (3B) was totally unresponsive ;)

Smanar, 2021-02-22:
Ok, so same problem than me, the option "-j2" monopolize too much the CPU.

Sjonnie2018, 2021-02-22:
Hi Smanar,

Compiling failed. I am not sure if it is at the last stage of the compilation so I grabbed the output for you to investigate:

`rm -f libde_rest_plugin.so
g++ -Wl,-O1 -shared -o libde_rest_plugin.so release/air_quality.o release/authorisation.o release/bindings.o release/change_channel.o release/connectivity.o release/colorspace.o release/database.o release/daylight.o release/device_setup.o release/diagnostics.o release/discovery.o release/de_web_plugin.o release/de_web_widget.o release/de_otau.o release/event.o release/event_queue.o release/fan_control.o release/firmware_update.o release/gateway.o release/gateway_scanner.o release/green_power.o release/group.o release/group_info.o release/gw_uuid.o release/ias_ace.o release/ias_zone.o release/identify.o release/json.o release/light_node.o release/poll_control.o release/poll_manager.o release/read_files.o release/resource.o release/resourcelinks.o release/rest_configuration.o release/rest_devices.o release/rest_gateways.o release/rest_groups.o release/rest_lights.o release/rest_node_base.o release/rest_resourcelinks.o release/rest_rules.o release/rest_sensors.o release/rest_schedules.o release/rest_touchlink.o release/rest_scenes.o release/rest_info.o release/rest_capabilities.o release/rule.o release/thermostat_ui_configuration.o release/upnp.o release/permitJoin.o release/scene.o release/sensor.o release/thermostat.o release/time.o release/tuya.o release/basic.o release/appliances.o release/reset_device.o release/rest_userparameter.o release/zcl_tasks.o release/window_covering.o release/websocket_server.o release/xmas.o release/moc_de_web_plugin.o release/moc_de_web_plugin_private.o release/moc_de_web_widget.o release/moc_gateway.o release/moc_gateway_scanner.o release/moc_poll_manager.o release/moc_rest_devices.o release/moc_websocket_server.o  -L../../release -L../.. -ldeCONZ -lcrypt -lsqlite3 -lQt5Widgets -lQt5Gui -lQt5SerialPort -lQt5WebSockets -lQt5Network -lQt5Core -lGLESv2 -lpthread -latomic  
/usr/bin/ld: release/bindings.o: file not recognized: file truncated
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile.Release:379: ../libde_rest_plugin.so] Error 1
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'
make: *** [Makefile:40: release] Error 2
pi@RPi3:~/deconz-rest-plugin $ 
`
Maybe we'd be better of you compiling the file and me copying it into the appropriate folder?

Greetz,

Sjonnie

Smanar, 2021-02-22:
Can try, but it s not realy secure for you, and the compilation is device dependent (but we use both raspberry 3)
I have put the file here https://cjoint.com/c/KBwszZuDCa6

Sjonnie2018, 2021-02-22:
OK, the so file was accepted and deCONZ works as before. The only thing is: I still can't change the name. Looked at all the cluster info screens and modified the NWK data to reflect the name I want but it simply refuses to change the name in deCONZ GUI. Is there something I am missing?

Greetz,

Sjonnie

Smanar, 2021-02-22:
>Looked at all the cluster info screens and modified the NWK data to reflect the name I want but it simply refuses to change the name in deCONZ GUI

Yep, try with phoscon, If I m right the device is not visible in the API, so you can edit value in it (and they will be used by deconz too)

If the device is not visible, instead of re-including it
- set phosocn in permit join ( add new light)
- Read again the basic attribute, cluster 0x0000, button "read"

Sjonnie2018, 2021-02-23:
Hi Smanar,

That works! Phoscon identifies the repeater with the name "Range Extender 12" and as type water sensor. The fact that it shows as a water sensor is fine by me. I will only use it in deCONZ.

Thanks for your relentless support (as always :) )

Greetz,

Sjonnie

Smanar, 2021-02-23:
Lol, a water sensor ? Seriously ?
It s a bug from phoscon, or deconz ? If you can take a look on the device JSON ? You can see it in domoticz/custom/deconz/lights and on the right icons ?

Sjonnie2018, 2021-02-23:
Hi Smanar,

In Domoticz it shows as follows:

`{
	"ActTime" : 1614103481,
	"AstrTwilightEnd" : "20:02",
	"AstrTwilightStart" : "05:46",
	"CivTwilightEnd" : "18:43",
	"CivTwilightStart" : "07:04",
	"DayLength" : "10:30",
	"NautTwilightEnd" : "19:22",
	"NautTwilightStart" : "06:25",
	"ServerTime" : "2021-02-23 19:04:41",
	"SunAtSouth" : "12:54",
	"Sunrise" : "07:39",
	"Sunset" : "18:09",
	"app_version" : "2020.2",
	"result" : 
	[
		{
			"AddjMulti" : 1.0,
			"AddjMulti2" : 1.0,
			"AddjValue" : 0.0,
			"AddjValue2" : 0.0,
			"BatteryLevel" : 255,
			"CustomImage" : 17,
			"Data" : "On",
			"Description" : "",
			"DimmerType" : "none",
			"Favorite" : 0,
			"HardwareID" : 11,
			"HardwareName" : "ConBee2",
			"HardwareType" : "deCONZ plugin",
			"HardwareTypeVal" : 94,
			"HaveDimmer" : true,
			"HaveGroupCmd" : true,
			"HaveTimeout" : false,
			"ID" : "14:b4:57:ff:fe:36:4b:a2-01",
			"Image" : "ComputerPC",
			"IsSubDevice" : false,
			"LastUpdate" : "2021-02-23 17:48:30",
			"Level" : 0,
			"LevelInt" : 0,
			"MaxDimLevel" : 100,
			"Name" : "Range extender Schuur",
			"Notifications" : "false",
			"PlanID" : "0",
			"PlanIDs" : 
			[
				0
			],
			"Protected" : false,
			"ShowNotifications" : true,
			"SignalLevel" : "-",
			"Status" : "On",
			"StrParam1" : "",
			"StrParam2" : "",
			"SubType" : "Switch",
			"SwitchType" : "On/Off",
			"SwitchTypeVal" : 0,
			"Timers" : "false",
			"Type" : "Light/Switch",
			"TypeImg" : "lightbulb",
			"Unit" : 103,
			"Used" : 0,
			"UsedByCamera" : false,
			"XOffset" : "0",
			"YOffset" : "0",
			"idx" : "313"
		}
	],
	"status" : "OK",
	"title" : "Devices"
}
`

So Domoticz sees it as a light switch / light bulb and Phoscon thinks it is a water sensor. Both are mistaken as it is only a repeater but as I said: fine by me! I have no need for programming or adding the repeater to a flow. It just needs to amplify the signal so my outdoor sensors can be read.

Greetz,

Sjonnie

Smanar, 2021-02-23:
Ok so I think the problem is from phoscon.
A last question, and I stop disturb you ^^ ! It works fine ?

Sjonnie2018, 2021-02-23:
You can never disturb me;)

Yes the repeater is working fine as a router. All connected sensors report values so all in order!

Greetz,

Sjonnie

closed 2021-02-24 by Sjonnie2018

javiges, 2021-09-03:
> Nunca me puedes molestar;)
> 
> Sí, el repetidor funciona bien como enrutador. Todos los sensores conectados informan valores, ¡así que todo en orden!
> 
> Saludos,
> 
> Sjonnie

There is no way that it is shown in another way? I am seriously thinking about removing everything from deconz and putting it in z2mqtt, since it seems that they are better, they give more attributes for devices and they are up to date and phoscom is not up to par

Smanar, 2021-09-03:
The inclusion for tuya device is problematic yes, but will be improved on future version.
Idk if this device can be included just with phoscon.

>they give more attributes for devices

Wich one attribute you want for this repeater ?

>phoscom is not up to date

Phoscon is just a basic third app and don't support all deconz device, use any other one

SwoopX, 2021-09-03:
@javiges First of all, the following isn't meant offensive in any way nor should it be any blaming. However, it somehow shows once more that DE should provide some more clarity on their eco system and explain to users what's out there and what to expect.

If you feel like z2m is the better choice for your needs, then you should consider moving over. We won't stop you. But there's some points to elaborate on.

1. As Smanar already said, Phoscon is just a REST API client like many others out there and most importantly, it's NOT deconz. So you can use it to interact with deconz' API
2. Yes, from a supported device perspective, Phoscon is running quite behind. However, deconz supports by far more devices than Phoscon does.
3. I'm certain z2m has some strengths over deconz, where adding easy device support is one of them. With the next version, DDFs (Device Definition Files) will be introduced so that everybody will be put in the position to add devices without any programming skills. Presumably, this significant change will be a bit bumpy but give greater flexibility.
4. I've to admint I'm a bit biased on this one and this is my very personal opinion, but regardless. What about the attributes? deconz GUI is the most powerful tool out there giving you access to literally everything which can be configured on a device or read in terms of information. And this is way more than is exposed via deconz' API or any other API.

But, as said, you need to collect some more background info on deconz to unleash the true power 😉 Up to you to make the choice.