dresden-elektronik/deconz-rest-plugin #4446: Zemismart M515EGBZTN

DecoBook, 2021-02-19:
## Device

- Product name: Zemismart M515EGBZTN
- Manufacturer: _TZE200_xuzcvlku
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Other: Roller blinds motor

## Screenshots

Mimiix, 2021-02-19:
Missing all screenshots.

Smanar, 2021-02-19:
Duplicate https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3348 ?

DecoBook, 2021-02-19:
I tried everything suggested in #3348, did not work.
Last try was with lates Beta 2.10.0

Smanar, 2021-02-19:
Yep but 
```
Ok, so it seem there is a new model for this device, but this one is not a router but an end device.
Problem, the code don't support it yet, no one have problem to include it ?
```

Do you have a node capture from deconz ?

Smanar, 2021-02-19:
> Ok, so it seem there is a new model for this device, but this one is not a router but an end device.
> Problem, the code don't support it yet, no one have problem to include it ?

It s exactly your device ^^.
You have a linux machine to test modifications ?

DecoBook, 2021-02-20:
Yes, Conbee ist connected to a Raspberry PI

Smanar, 2021-02-20:
In fact I have already make the PR, and already tested it with another user.
But if you want to try it

So you need to instal deconz then take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch M515EGB https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then re-include the device.

DecoBook, 2021-02-21:
Thank you, I followed the instructions to compile the rest-plugin and restarted deconz.
Then I went to Phoscon app and searched for lights, twice - device did not show.
I deleted the node in deCONZ GUI, went back to Phoscon and searched for lights twice - device did not show up.

Is there some kind of log I can look at, maybe I can set the logging level to debug somewhere to get extensive information?

Smanar, 2021-02-21:
I have problem ATM with the same kind of device, battery powered covering, and the code is already in beta, so we are already seraching the problem > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4223

stevewull, 2021-02-21:
I have the same Zemismart _TZE200_xuzcvlku model TS0601 covers. 

Using the Roanoke release, deCONZ detected it properly. It appears in the API as a "Window covering device".  Also was automatically detected as a Cover in Home Assistant using the deCONZ integration.  This is great! Thank you for your hard work.

One question: how to I reverse the direction setting in deCONZ?  In HA selecting up causes the cover to go down and vice versa. 
I realize that the Cover hardware has a setting to reverse the direction, but doing this would cause the supplied remote control to have the wrong direction.

Smanar, 2021-02-21:
>One question: how to I reverse the direction setting in deCONZ? In HA selecting up causes the cover to go down and vice versa.
I realize that the Cover hardware has a setting to reverse the direction, but doing this would cause the supplied remote control to have the wrong direction

IDK there is lot of combinaisons possibles.

- From the code (the direction is not reversed by defaut ATM).
- You can reverse the wiring.
- There is an hidden  zigbee command, available using the api 

PUT, PATCH `/api/<apikey>/lights/<id>` with {"reverse":true}

stevewull, 2021-02-21:
Thanks Smanar

closed 2021-02-22 by manup

stevewull, 2021-03-01:
Hi Smanar, this topic came up on the Home Assistant forum: https://github.com/home-assistant/core/issues/46525

I executed the put command with {"reverse":true}. This corrected the direction setting selectin Up in HA now makes the cover go up. However, the % open, which was correct before is now reversed. It shows 100% open when the cover is fully closed. I'm not sure whether this is an issue with the API or with the deConz integration for Home Assistant.

Smanar, 2021-03-01:
Ha ? I haven't imagined this situation.

So without the command all is reversed.
But with this command, the physical side is good, but the return are reversed ?

BTW, can you share a device JSON, with the command {"reverse":true} and one other with {"reverse":false} ?

stevewull, 2021-03-06:
How do I capture the device JSON when issuing the commands?  (that was for the M515EGB (ZB) with external usb zigbee. It is working well overall. I can control it from HA)

Thus, I purchased some more curtain motors and was shipped the new, M515EGBZTN, version with internal zigbee. Using version 2.10.1 of deCONZ, I paired the new curtain motor. It shows up in the deCONZ gui. Same cluster info as posted by DecoBook. But it does not appear in Phoscon App or the API.

Smanar, 2021-03-06:
I know it s possible with HA log, but I don't know the procedure.
Else you can just use

`http://IP:PORT/api/KEY/lights`

IP and PORT are the same you are using for phoscon, and key is an api key, if you haven't one

`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

Kane610, 2021-04-09:
> How do I capture the device JSON when issuing the commands? (that was for the M515EGB (ZB) with external usb zigbee. It is working well overall. I can control it from HA)
>
> Thus, I purchased some [...]

You can always enable debug logs in HASS for the deCONZ integration, you will get all data in the logs

PaulBiod, 2023-03-22:
> IDK there is lot of combinaisons possibles.
>
> * From the code (the direction is not reversed by defaut ATM).
> * You can reverse the wiring.
> * There is an hidden  zigbee command, available using the [...]

Hi, could you explicit the exact command please, cause I did : 

curl -H 'Content-Type: application/json' -X PUT -d '{"reverse": true}' http://172.30.33.4:40850/api/<APIKEY>/lights/9

and it says 
[{"success":{"/lights/9/reverse":true}}]
but it doesn't change anything, my buttons are still inverted...
thanks

PaulBiod, 2023-03-22:
> > > One question: how to I reverse the direction setting in deCONZ? In HA selecting up causes the cover to go down and vice versa.
> > > I realize that the Cover hardware has a setting to reverse the direction, but doing this would cause the supplied remote control to have the wrong direction
> > 
> > 
> > IDK there is lot of combinaisons possibles.
> > 
> > * From the code (the direction is not reversed by defaut ATM).
> > * You can reverse the wiring.
> > * There is an hidden  zigbee command, available using the api
> > 
> > PUT, PATCH `/api/<apikey>/lights/<id>` with {"reverse":true}
> 
> Hi, could you explicit the exact command please, cause I did :
> 
> curl -H 'Content-Type: application/json' -X PUT -d '{"reverse": true}' http://172.30.33.4:40850/api/APIKEY/lights/9
> 
> and it says [{"success":{"/lights/9/reverse":true}}] but it doesn't change anything, my buttons are still inverted... thanks

I respond to myself, it worked with reverse: false (don't know why) !!