dresden-elektronik/deconz-rest-plugin #4447: Locating a device

dergreg, 2021-02-19:
## Describe the question or issue you are having

This is probably the most stupid question ever asked here, but here it goes 🙈

I have an Aqara temperature and humidity sensor connected to my Raspberry with deconz board in my house.

It reports everything well and signal strength is good (light green in deconz).

There problem is: I don't know where it is 😂😂😂

Is there any way, say walking around the house with the Raspberry, and continously polling for and measuring the signal strength to the Aqara sensor?

The signal strength colors in the deconz application seem to update rather infrequently.

## Screenshots

n/a

## Environment

Raspberry 3B+ with deconz gpio board, latest firmware and Raspbian Stretch. 

## deCONZ Logs

n/a

## Additional context

n/a

Smanar, 2021-02-19:
Lol, thoses device are lazy, so realy quiet.

Try to heat your house, but only some part ? you will have more report if the temperature change.

Mimiix, 2021-02-22:
I've had this issue with someone before. The guy eventually found it on a chair. 

I suggested the same as @Smanar said, but then with a second sensor. Try to put it in a room and see if it gets to almost the same readings.

SwoopX, 2021-02-22:
Generally, if a device has the identify cluster available, you use this one to make the give whatever type of signal (turn on/off, blink, audio, etc.).

The Aqaras do not support this, so there's barely any chance to find it, except for the given suggestions.

Mimiix, 2021-02-23:
> Generally, if a device has the identify cluster available, you use this one to make the give whatever type of signal (turn on/off, blink, audio, etc.).
> 
> The Aqaras do not support this, so there's barely any chance to find it, except for the given suggestions.

Even if it did blink, that sensor doesn't have a bright light.

closed 2021-03-24 by github-actions[bot]