dresden-elektronik/deconz-rest-plugin #4450: 2.10.00-Beta: Lidl/Silvercrest TS0211 not recognized as button

MartinAirN, 2021-02-19:
Please note: The device here is named either ts0211_5 or ts0211_6. That is due to re-pairing, after deleting the device through the rest-api.

## Describe the bug
The Lidl/Silvercrest TS0211 should now in deconz (not Phoscon) be supported after Pull Request #4282.

It connects to the network, and is named as expected. If I then understand how a device that is not yet supported by Phoscon should work, is that it should be working in 3rd party apps as just a button in this case. That it does not.

## Steps to reproduce the behavior
1. Make sure the device is out of the network if it was there before
2. In Phoscon click "Add new switch" and follow the "Other" steps
3. Press connect button on TS0211 3 times within 2 seconds
4. It should then appear on the Deconz network as connected (see below for screenshot) (Phoscon doesnt see it)
5. Go to 3rd party app, such as Home-Assistant (HA)
6. TS0211 should be listed as a device
7. Open device details -> No battery status
8. Try to link device to automation -> No Button options
9. Wait X time (30-60min I have seen)
10. HA -> Open device details -> battery status
11. HA -> Try to link device to automation -> No Button options

ps. Instead of using HA, the rest-api can also be used. This will not directly list the battery status and will list buttonevent as "null"

listening to the deconz_event in HA lists a button event on button press:
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "ts0211_5",
        "unique_id": "84:2e:14:ff:fe:f5:ba:31",
        "event": null,
        "device_id": "0353c501997e6ca6418c159aaddfb112"
    },
    "origin": "LOCAL",
    "time_fired": "2021-02-19T10:30:10.348084+00:00",
    "context": {
        "id": "5c5e5baa12933becbcc6f019b6e7099d",
        "parent_id": null,
        "user_id": null
    }
}
```

## Expected behavior
Phoscon should see the device as pairing as a Switch device, it does see it if pairing as a sensor device. (not in the UI just the pairing action)
Step 7 & 8 should have battery status & button options

## Screenshots
Screenshot of device info in Deconz:
[image]

Screenshot of Home-Assistant reading the device:
[image]
Note there is no battery level listed. Nor in HA is the fact it should be a button recognized.

Rest-API sensor read (after waiting for the battery status to show):
```
{
  "config": {
    "battery": 100,
    "enrolled": 1,
    "on": true,
    "pending": [],
    "reachable": true
  },
  "ep": 1,
  "etag": "38dfea563aa02567c8c3dfd7b8f9fc56",
  "lastseen": "2021-02-19T16:37Z",
  "manufacturername": "SILVERCREST",
  "mode": 1,
  "modelid": "TS0211",
  "name": "TS0211 6",
  "state": {
    "buttonevent": null,
    "lastupdated": "2021-02-19T16:37:42.966",
    "lowbattery": false,
    "tampered": false
  },
  "swversion": "2020.8.4",
  "type": "ZHASwitch",
  "uniqueid": "84:2e:14:ff:fe:f5:ba:31-01-0500"
}
```

## Environment
 - Host system: NAS
 - Running method: Marthoc Docker container
 - Firmware version: (26680700)
 - deCONZ version: (2.10.00)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system?: no

## deCONZ Logs
Click on button event:
```
17:37:41:955 APS-DATA.indication srcAddr: 0x3e47, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -26
17:37:41:955     asdu: 097d00050000640000
17:37:41:956 APS-DATA.request id: 65, addrmode: 0x02, addr: 0x3e47, profile: 0x0104, cluster: 0x0500, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
17:37:41:956 [INFO] - No button map for: TS0211, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 050000640000, zclSeq: 125
```

## Additional context
Other devices in the network:
[image]

Smanar, 2021-02-19:
```
{
  "config": {
    "battery": 100,
    "enrolled": 1,
    "on": true,
    "pending": [],
    "reachable": true
  },
  "ep": 1,
  "etag": "38dfea563aa02567c8c3dfd7b8f9fc56",
  "lastseen": "2021-02-19T16:37Z",
  "manufacturername": "SILVERCREST",
  "mode": 1,
  "modelid": "TS0211",
  "name": "TS0211 6",
  "state": {
    "buttonevent": null,
    "lastupdated": "2021-02-19T16:37:42.966",
    "lowbattery": false,
    "tampered": false
  },
  "swversion": "2020.8.4",
  "type": "ZHASwitch",
  "uniqueid": "84:2e:14:ff:fe:f5:ba:31-01-0500"
}
```

>"buttonevent": null,

Ok there is probably an issue, but the device is clearly a switch from the JSON.

MartinAirN, 2021-02-25:
If the device missing from button_maps.json could be the reason for the switch not being usable as a switch perhaps @jurriaan has these changes at the ready based on the original PR.

jurriaan, 2021-02-25:
Not sure how this is supposed to work, I just listen to the button events and that works fine here.

Should a pushbutton even be registered as a switch?

SwoopX, 2021-02-25:
> 
> 
> If the device missing from button_maps.json could be the reason for the switch not being usable as a switch perhaps @jurriaan has these changes at the ready based on the original PR.

Yes, that indeed seems to be the case.

knutbrecht, 2021-03-15:
ist there any solution for that?
I got the exact same behaviour - device is not recognized as a butten. It is recognized as a sensor (why?) but not listed in Phoscon.
It is listed in deconz objects (iobroker in my case) but no correct behavior when buuton is pushed. There is someting happening but the "button" object is not changing.

Has anyone succeeded integrating the doorbell correctly?
[image]

MartinAirN, 2021-03-16:
> 
> 
> ist there any solution for that?

I have been trying to get a button_maps.json entry going for this one. But as its my first time trying, no clue if/when I will figure out what the contents should be.

So if anyone has a manual or list of logs he/she needs from me feel free to comment/ask.

Typ1er, 2021-03-21:
I have the same problem in FHEM, the readings always stay at "zero", but the timestamp is rewritten. So it does not really work unfortunately.

SwoopX, 2021-03-22:
Just had a look at the PR, the implementation is wrong and it should have never exposed as a switch. Correct would have been to expose it as alarm sensor.

Mimiix, 2021-03-23:
@SwoopX What would be the next steps?

SwoopX, 2021-03-24:
The resource type should just be changed from ZHASwitch to ZHAAlert so the `alert` resource gets exposed, which then can be used to take whatever action.

I guess I can do that somewhere next week.

MartinAirN, 2021-04-17:
This issue is currently waiting for the Merge of the fix #4652.

MartinAirN, 2021-05-10:
As #4652 has been closed by @SwoopX this is still an issue.

Using the latest deconz version and re-pairing the TS0211 Silvercrest Doorbell it is only reporting its battery status to other clients and not behaving as a button nor alarm.

T0mWz, 2021-05-10:
However, pull request https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4652 is closed and not merged, so it's not clear to me of it's now fixed or still open. Will try myself too, to see what's happens now

MartinAirN, 2021-05-10:
@T0mWz Indeed, it is closed that is why the issue is still open and I confirmed on the latest version the behavior was not fixed by another change.

SwoopX, 2021-05-30:
Can someone please check with debug enabled if such an event gets generated on press and release (and please also share the lines)?

`[INFO] - No button map for: TS0211, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 050000640000, zclSeq: 125`

T0mWz, 2021-05-31:
Hi @SwoopX ,

Have started deconz with `DEBUG_INFO=2`. See not more logging information as already provided by @MartinAirN .
```
11:17:44:784 [INFO] - No button map for: TS0211, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 050000640000, zclSeq: 105
11:17:44:785 Websocket 192.168.112.3:60166 send message: {"e":"changed","id":"43","r":"sensors","state":{"buttonevent":0,"lastupdated":"2021-05-31T09:17:44.782","lowbattery":false,"tampered":false},"t":"event","uniqueid":"84:2e:14:ff:fe:f9:4a:ca-01-0500"} (ret = 198)
```

T0mWz, 2021-06-08:
Hi @Kane610,

Do think that also for the Lidl TS0211 support is required within HA?  Here we encounter the issue as with the Ikea Shortcut Button, were  the device registered without any further option within HA?

Thanks for your help!

Kane610, 2021-06-08:
> Hi @Kane610,
> 
> Do think that also for the Lidl TS0211 support is required within HA? Here we encounter the issue as with the Ikea Shortcut Button, were the device registered without any further option within HA.
> 
> Thanks for your help!

Yeah it's not supported today. Supported units can be seen here https://github.com/home-assistant/core/blob/eb687b733233c8aa97b7963b9b522808ed65d711/homeassistant/components/deconz/device_trigger.py#L525

Open an issue and be as detailed as possible. But not deconz logs but events from within hass

T0mWz, 2021-06-08:
I'm sorry, but I doesn't get any event in HASS 😅

Kane610, 2021-06-09:
> Is this implementation correct enough for HA @Kane610 ? I just started with HA, so I have not yet an idea of the design principes 😅

The HA event is what I need, but it needs to contain "buttonpress": 100x in them to be relevant, I need a mapping between that event and what you did to generate it e.g. short press -> 1001, release after short press -> 1002, etc.

I also need a modelid

T0mWz, 2021-06-09:
@SwoopX can you help out here ? 😅 Can you make a device mapping somehow? 

Some HA logs already;
```
2021-06-09 13:03:05 DEBUG (MainThread) [pydeconz.websocket] {"e":"added","id":"51","r":"sensors","sensor":{"config":{"battery":null,"enrolled":1,"on":true,"pending":[],"reachable":true},"ep":1,"etag":"738cd5d0da54d7348aa4db531a82df4e","id":"51","lastseen":"2021-06-09T11:03Z","manufacturername":"LIDL Silvercrest","mode":1,"modelid":"HG06668","name":"Switch 51","state":{"buttonevent":null,"lastupdated":"2021-06-09T11:02:54.994","lowbattery":false,"tampered":false},"type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:f9:4a:ca-01-0500"},"t":"event","uniqueid":"84:2e:14:ff:fe:f9:4a:ca-01-0500"}

2021-06-09 13:03:23 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"51","r":"sensors","state":{"lastupdated":"2021-06-09T11:03:23.044","lowbattery":false,"tampered":false},"t":"event","uniqueid":"84:2e:14:ff:fe:f9:4a:ca-01-0500"}
2021-06-09 13:03:24 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"51","r":"sensors","state":{"lastupdated":"2021-06-09T11:03:24.881","lowbattery":false,"tampered":false},"t":"event","uniqueid":"84:2e:14:ff:fe:f9:4a:ca-01-0500"}
```

Kane610, 2021-06-09:
I'll unsubscribe to this issue. Create an issue on https://github.com/Kane610/deconz once deCONZ support is verified to work.

closed 2021-06-13 by manup