dresden-elektronik/deconz-rest-plugin #4451: Xiaomi motion sensor: etag does not change when presence changes from true to false

hbkoenig, 2021-02-19:
## Describe the bug

using the LUMI lumi.sensor_motion.aq2 sensor I noticed that etag and lastseen timestamp only changes when presence changes from false to true, but they stay the same when presence changes from true to false.

for Tuyatec RH3040 motion sensor, etag changes for both transitions of presence (as I and my code would expect).

example data:

LUMI lumi.sensor_motion.aq2:
```
        UTC                     etag                                    lastseen                lastupdated                presence
        2021-02-19T17:38:38     46305a4902491dd946ea0c6c9f0e36a4        2021-02-19T17:25Z       2021-02-19T17:26:32.506    false
        2021-02-19T17:52:51     51948b7fd85725b42c62610bb4dc0218        2021-02-19T17:52Z       2021-02-19T17:52:51.153    true
        2021-02-19T17:54:22     51948b7fd85725b42c62610bb4dc0218        2021-02-19T17:52Z       2021-02-19T17:54:22.007    false
        2021-02-19T17:54:45     4db875fa14a001e363794ba8898d66f6        2021-02-19T17:54Z       2021-02-19T17:54:44.498    false
```

Tuyatec RH3040:

```
        UTC                     etag                                    lastseen                lastupdated                presence
        2021-02-19T17:41:00     88b2b30b6cbda4aa404c4c9492f2ad05        2021-02-19T17:40Z       2021-02-19T17:26:02.685    false
        2021-02-19T17:41:02     88b2b30b6cbda4aa404c4c9492f2ad05        2021-02-19T17:41Z       2021-02-19T17:26:02.685    false
        2021-02-19T17:52:54     2f09a9276d6a1d7dbf8e261a53149792        2021-02-19T17:52Z       2021-02-19T17:52:52.806    true
        2021-02-19T17:53:54     ef315fc4ebee34bf0e7e13f0563eb3b9        2021-02-19T17:53Z       2021-02-19T17:53:53.230    false
        2021-02-19T18:03:11     06f4dcd54611802b2d0a000c1d7da046        2021-02-19T18:03Z       2021-02-19T17:53:53.230    false
```

## Steps to reproduce the behavior
check REST data

## Expected behavior

etag and lastseen should change when sensor value/measurement changes (as it does for Tuyatec RH3040).
next I'll check and use LIDL/Sivlercrest motion sensors too...

## Screenshots

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: 26680700
 - deCONZ version: 2.10.00 (just updated from 2.09.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

available on demand

## Additional context

SwoopX, 2021-02-20:
Root cause seems to be rather simple: deconz changes the etag on transmissions. However, the Xiaomi's do not send occupancy false, so this switch is handled in software...

Guess I already found the place where this is missing.

hbkoenig, 2021-02-22:
> Root cause seems to be rather simple: deconz changes the etag on transmissions. However, the Xiaomi's do not send occupancy false, so this switch is handled in software...
> 
> Guess I already found the place where this is missing.

cool, thanks!!

did I get you right that some sensors like that RH3040 handle the timeout (not seeing any motion) in the sensor and transmit a timeout "no presence" zigbee packet, while for others like that Xiaomi Aqara that timeout handled in deconz ?

interesting;)

SwoopX, 2021-02-22:
Yup. Sad, but true. One more reason for me to go for the more pricy (and compliant) stuff...

closed 2021-03-03 by manup