dresden-elektronik/deconz-rest-plugin #4469: Support Sengled SMART LED Z01-A19NAE26

deviantintegral, 2021-02-23:
## Device

- Product name: SENGLED SMART LED: https://www.amazon.ca/Sengled-2700K-6500K-Required-Daylight-Equivalent/dp/B08KZPDK1J/ref=sr_1_7
- Manufacturer: sengled
- Model identifier: Z01-A19NAE26 (phoscon shows version 0x0000002E as well)
- Device type :
  - Light

## Screenshots
[image]

[image]

[image]

As is, I can control brightness in Phoscon. I can control light temperature in deconz, though it's showing the bulb as if it supports full RGB. Those commands error out if tried.

From the REST API (the light is currently off at the switch)

```
  "5": {
    "etag": "adcc136c87b705f2f20177919eba6fa3",
    "hascolor": true,
    "lastannounced": null,
    "lastseen": "2021-02-23T01:27Z",
    "manufacturername": "sengled",
    "modelid": "Z01-A19NAE26",
    "name": "Kitchen Hallway 1",
    "state": {
      "alert": "none",
      "bri": 254,
      "colormode": "ct",
      "on": false,
      "reachable": false
    },
    "swversion": "0x0000002E",
    "type": "Dimmable light",
    "uniqueid": "b0:ce:18:14:03:00:f9:92-01"
  },
```

Mimiix, 2021-02-23:
Missing screenshots from other clusters.

deviantintegral, 2021-02-23:
Done! I think I misunderstood what screenshots were required from the issue template. I'll file a followup.

deviantintegral, 2021-02-23:
An aside for anyone else reading this; while temperature controls don't show up in Phoscon or Home Assistant's UI, I am able to change the colour by calling the `light.turn_on` service with `kelvin` set. I just verified that the lights work with the `flux` integration too. As long as you don't need manual control of temperature, I'd say these lights are good enough to purchase given their low price in bulk at $14.25 CDN / bulb.

stevewull, 2021-03-09:
@Mimiix: I posted the screenshots of all clusters, including all of the attributes for the color control cluster, in issue #4536. Hopefully, it provides the info you requested.

I confirm that behavior is the same for me as deviantintegral: temperature controls don't show up in Phoscon or Home Assistant's UI, but I am able to change the color by calling the light.turn_on service.

stevewull, 2021-04-01:
I just installed v.2,10.4 Tortuga.  This issue is not resolved. API still does not have the ct value:

[image]

closed 2021-04-22 by manup

kaisamng, 2021-06-27:
The pull request for this only adds a kWh metering element for Z01-A19NAE26, not an element for adjusting color temperature manually. Unfortunately, the workaround that @deviantintegral on calling the service manually fails. 

Currently through Phoscon, adjusting the brightness changes _both_ the color temperature and the brightness of these bulbs. 

Is there any plan to support a color temperature option, separate from brightness, for these bulbs?