dresden-elektronik/deconz-rest-plugin #4471: inNode homekit o

TheNON75, 2021-02-23:
## Feature request type

## Description

## Considered alternatives

## Additional context

TheNON75, 2021-02-23:
Dear All,

I would like to recommend to implement the possibility of switching off the sending of payload on the HomeKit output on start up/restart.
for example if the in node is dimmer and the last message was on:true then the light or the controlled accessory will be turned on or toggled, depending on how the payload was used.
The current method may cause issues, when performing a maintenance or simply performing a full deployment during the night as some lights may turn on based on the dimmer.
Typically this happens in the following cases:
- the dimmer (or any kind of switch) is acting as a toggle on/off
- there are more dimmers (switches) that are controlling an accessory, therefore one of the actors' last message will be with high probability on:true
- the accessories are controlled also from an application such as home app

thank you for considering

Mimiix, 2021-02-23:
Hi.

This is not how the Template works...

Please use the template accordingly. I do not appreciate going around it like this.

closed 2021-02-23 by Mimiix

TheNON75, 2021-02-23:
sorry Mimiix, wrong place, I will delete it

Mimiix, 2021-02-23:
No need to say sorry 😉.