dresden-elektronik/deconz-rest-plugin #4476: Moes / Tuya TRV not working

swieprecht, 2021-02-24:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: Raspberry Pi 3b+
 - Running method: Raspian
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 23/12/2020
 - Device: ConBee II
 - Do you use an USB extension cable: Yes - 3m
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

easybeat, 2021-02-24:
Hi

First I would recommend to updated deconz to latest beta. 

Then you need to do the inclusion for each device 2x in a row. The first time it will show up in Deconz as unknown device. Then run it again and it will update to a proper device.

Be aware not to put these devices too far away from the box. There seems to be a connection issue with them.

swieprecht, 2021-02-24:
Hi, thank you for the advice,
Sorry I can't have explained it well enough - I have done inclusion twice in a row, which didn't work, then once, reset, and again and didnt work, and then the read information from Deconz GUI trick. 
The only thing I haven't done is update deconz to latest beta - is that recommended from a stability POV? And will doing this actually improve Tuya device recognition?
:)

easybeat, 2021-02-24:
If you want to use the TRV I do reccommend to update....I'm on 2.9.3 and there is no difference in stability.

swieprecht, 2021-02-24:
Oh okay amazing, I didnt know if a beta update would help but if it does ill definitely update now :)

Smanar, 2021-02-24:
If you have the GUI, will be easier.
1 - check if the device is visible on it.
2 - Set phoscon (and not deconz) in permit join (add new sensor)
3 - Read the basic attribute on deconz (cluster 0x0000, press "read" button).

It will trigger a new inclusion.

If it s still not working (remember not all device are visible on phoscon, need to use the API or third app), can you share the manufacture name see on on step 3.
And if you can the bad entry in API, the device JSON.

swieprecht, 2021-02-24:
Thank you all... it's now working after an upgrade to beta and a reboot.
The devices are oddly called Fairy Lights in the GUI not Tuya TRV, but it's all working in Home Assistant

closed 2021-03-26 by github-actions[bot]