dresden-elektronik/deconz-rest-plugin #4478: Can't get Touchlink scan to deliver nearby bulb results

pa1okz, 2021-02-25:
## Describe the question or issue you are having

## Screenshots

## Environment

## deCONZ Logs

## Additional context

Mimiix, 2021-02-25:
Touchlink is not yet supported in Conbee II https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3485

pa1okz, 2021-02-25:
@Mimiix: Ah, I simply missed that little detail. Interesting that the response in the REST API suggests that it -is- working, considering the principle responses. 

Then I believe that I am out of options in resetting these bloody expensive Philips bulbs, other than buying myself a remote. Let's take the positive out of that, luckily enough I can bind it to the ConbeeII.

Mimiix, 2021-03-01:
@pa1okz I wish i had better news and i do agree with you on that. I've created a feature request.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4498

It's a shame you need those for the Hue devices. However, the remote is nice and cheap and you'll find a use for it!

closed 2021-03-31 by github-actions[bot]