dresden-elektronik/deconz-rest-plugin #4481: Automatic restart if devices are not reachable

Claus1985, 2021-02-25:
Hi Guys,

since the last phoscon update my setup crashes almost once per day. Before the update it could run about 2 weeks without a problem.
I gave up on getting this fixed for quiet some time but I’m really tiered of manually restarting the software every day.
Is there a way to restart deconz automatically if more than one device is not reachable for longer than 15 minutes?
If not natively in deconz maybe via nodered

## Screenshots

## Environment
- Raspberry Pi 2b+
- Raspbian
 - Firmware version: 26680700
 - deCONZ version: 2.09.03
 - Device: ConBee II /
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? No

Mimiix, 2021-03-01:
It would be nice to see some logs on why it crashes. Are you able to provide those?

Claus1985, 2021-03-01:
Hi Dennis,

can you tell me a fast way to get the logs?
Thx,

Claus

Mimiix, 2021-03-01:
Hi Claus,

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

That would be the fastest way ! :)

Smanar, 2021-03-01:
Hello, and what is crashing ?
It s the OS, deconz, the API, or something in the zigbee network ?

Claus1985, 2021-03-02:
Thx for the hints, I will get a log and see whats the problem.
System still runs finde (incl. FHEM and NodeRed), its just deconz.
But even the webinterface is still responsible its just that the device connection
of all connected devices is lost.

Smanar, 2021-03-02:
Ok so can be a phoscon freeze ?
So yes, logs can be usefull, you can find in it a device spaming request for exemple.

Smanar, 2021-03-02:
If the webinterface is responsible I don't think at a deconz freeze.
No power supply or SD card issue ?

Just unplug/replug the conbee change something ?

closed 2021-04-01 by github-actions[bot]