dresden-elektronik/deconz-rest-plugin #4482: Tuya Zigbee LED Strip Light Controller 5 in 1 LED WZ5 (_TZE200_3thxjahu, TS0601)

skorpa, 2021-02-25:
## Device

- Product name: Tuya Zigbee LED Strip Light Controller 5 in 1 LED WZ5
- Manufacturer: _TZE200_3thxjahu
- Model identifier: TS0601 (Phoscon shows also Fairy lights)
- Device type: Light (controller of the led strip)

It is this product https://www.aliexpress.com/item/1005002021916931.html?spm=a2g0s.9042311.0.0.7dd94c4dLBn4tM

I'm able to add this device to Phoscon in the Light section, but it shows up with an icon of a socket and in Home Assistant it allows me only to turn it on/off and not to set colors, brightness, etc. I'm using the latest firmware in my Conbee II.

## Screenshots

### Basic cluster
[image]

### Node info
[image]

### Cluster OTAU
[image]

### Groups cluster
[image]

### Scenes cluster
[image]

### Cluster Tuya Specific
[image]

##Phoscon and Home Assistant app details
[image]
[image]
[image]
[image]

Mimiix, 2021-03-01:
@Smanar This seems like something you like :)

skorpa, 2021-03-07:
Is there any chance to add support for this device? Or can you recommend some other RGBW Led Strip controller which has Conbee II support and also supports RF remote, so I can control it over gateway or via classic remote? Thanks

Smanar, 2021-03-07:
This device use probably same command than for the Xmas strip.
Can you change the color for exemple (using the remote), and show the log you have in same time in deconz ? (deconz/help/log, with "info" and "info_l2"), The logs starting by "tuya debug"

skorpa, 2021-03-07:
This is what I see in deconz, I don't know if there is something valuable for you :)

This is what I see when I turn on and off led strip via remote I have:
[image]

I noticed that sometimes when I'm switching led strip on and off again and again this line will appear:
[image]

And this is log when I was changing colors via remote, I think that there is nothing:
[image]

Smanar, 2021-03-08:
Unfortunately the 0x01 0x01 for on/off is the only knowed command ^^.
I m using lines starting by "tuya debug", and yes you haven't one for color, same for brightness ? And I don't find information for tuya color command on other project.

Smanar, 2021-03-09:
You have it on your first capture, the 16th line.
All tuya request in and out are logged on deconz, I just need to mimic them, unfortunately the only one you had is the 0x01 0x01 for on / off

As your device don't have color or level cluster, it will use for sure the tuya cluster (if it s possible to change the color using zigbee, for exemple with tuya gateway, or zigbee switch)

skorpa, 2021-03-09:
Oh, sorry, I don't know why I didn't see it :)

Ok so I found out that I see Tuya debug line when I'm changing the brightness of the light:

[image]

But when I'm changing colors, running some pre-programmed scenes via RF remote (auto changing color, speed of change, etc.) I don't see anything in the deconz log.

I don't have a Tuya gateway but based on the product description there is a possibility of changing colors via the Tuya app so probably it will be using the Tuya cluster as you said. So it means that you cannot make support for this device in Conbee?

Smanar, 2021-03-10:
For the brightness (0x02 0x03) it s WIP, code is finished but not in PR yet https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4414

This one can be easily added.

But still no way for color ....
This device don't use same command than lidl strip for brightness, so I think it will be same for color.

And I don't find information on other project yet.

skorpa, 2021-03-18:
Ok, thank you very much for trying, I understand. Can you please recommend me some controller which I can use for controlling RGBW strip with support of RF remote and ofcourse with supportt Conbee II? Thank you!

AssetBurned, 2021-03-27:
@skorpa with which setting did you paired the controller with deCONZ ? The WZ5 behaves differently depending on the mode it is in. It looks like there is an LED on that thing that gives you an idea which mode it has. I guess it was just white when you paired it?
The RGB, RGBW and the other functions require that the LED has another Color.

Smanar, 2021-04-02:
@skorpa have you still the device ? And a linux machine to make some code test ?

skorpa, 2021-04-04:
Hi, @Smanar I have one Raspberry with HASS OS and I have another one with Raspberry Pi OS, would it be enough for tests? If not, I think will be able to get some Linux machine.

And about @AssetBurned - I will pair it again because first I made it work "offline" - just RGBW strip with RF remote control and after it I was pairing it with Conbee, but during this pairing I know that Conbee didn't see it and I was pressing the button for mode select on the device and after it it appeared on phoscon software. I will try it again and I will be watching exactly in what mode it is during pairing. I'll let you know.

Smanar, 2021-04-05:
Nice, thing have moved since the last time, it seem the device can use the Xmas code for color, will make a try today.
And yes a rapberry is perfect to test (the compilation can take time on small one)

From that I have read, you need to set the "mode" before including the device, because this action "change" the device.

Smanar, 2021-04-05:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch tuya_strip https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

So this code only work if your device have the manufacture name "_TZE200_3thxjahu", remember this can change.

So first try
- The device is able having a valid JSON (almost, color capabilities will missing)
- The device need to be controled by the API, not sure it will work on phoscon.
- The Level need to work
- The color can be modified but using hue,sat,level (XY not working)
- No return yet for color.

Smanar, 2021-04-10:
No results ?

closed 2021-05-09 by github-actions[bot]