dresden-elektronik/deconz-rest-plugin #4483: Heiman Sirens (HS2WD) looses connection after a restart of Deconz

fribse, 2021-02-26:
## Describe the bug

The siren goes offline after a restart of deconz. It looks like I can only get it back after unpluggin it, and waiting for it to loose battery power.

## Steps to reproduce the behavior

  1. Have it connected
  2. shut down deconz
  3. start deconz
  4. Siren is offload

## Expected behavior

It should reconnect on its own.

## Screenshots

[image]

## Environment

 - Host system: NUC with PVE
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.09.03)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes, Aeotec z-wave

## deCONZ Logs

## Additional context

fribse, 2021-02-27:
I've now had them out of the wall socket for 2 days, this has finally depleated the backupbatteries. And I've plugged them back in, and they are now back online. So doing a 'cold boot' on them, solves the problem. So deconz just needs to do this as well :-)

Mimiix, 2021-03-01:
@SwoopX  @Smanar Is this something we can fix?

SwoopX, 2021-03-01:
Not sure how to put this... Well, I got my own special experience on Satureday. I basically threw all the China stuff out of my network to replace it with some way more sophisticated stuff. I replaced 7 devices, where only one went through smoothly.

**TLDR**
Presumably, a lot of circumstances came together. After successful pairing, no or just some bindings were created. Consequently, so data was send from the devices or just occasionally. That, in 2 cases, resulted in 2 sensors being marked as unreachable. As I was sniffing the traffic, I could see that the "unreachable" sensors were sending something, but this was just not sensor related, so they were operational.

I believe the root cause is that although the devices have a poll control cluster, often an indication for light sleepers, the rather unusual non-changeable configuration makes them deep sleepers. When you pair them, you got like 15 secs to set them up to get responses. If you exceed that, you need to keep the device constantly awake with requests which must not take longer than 2-3 secs. In my case, I suppose deconz hasn't kept up the required pace, so the configuration didn't took place or was incomplete...

Now for the solution: the devices can be set into a mode, where they are responsive for like 3 mins. I just entered that and did all the bindings manually et voilà, all working as expected. The unreachable state occurred because (simplified) no valid sensor data ever reached deconz.
Additionally, the current way how deconz updates its available sensors in terms of last reception time is sub-optimal, as it is not ensured the correct sensors gets the updated timestamp, which can potentially results in a false unreachable state. I got a promising alternate approach for that running in my test network with a 100% success rate so far, but this must be tested on a larger scale.

Coming back to the original issue: I'd assume some of the described above is also applicable for the device in question. As it has a backup battery, a simple unplug/plug in is apparently not enough for the device to announce itself once again to the network. As I don't know the device's clusters, I can only fanasize about possible solutions. Eventually, setting up an attribute reporting for attribute 0x0002 on the IAS zone cluster could help to regularly "notify" deconz it is still there (given that's not already the case)...

fribse, 2021-03-02:
It's getting worse and worse here.
After going to 2.09.03 two weeks ago, (firmware 26670700) the entire network has gotten completely unusable, I run around 1-2 times a day and restarts bulbs, range extenders (all from IKEA).
This is driving me and the family crazy, and it's getting unusable.
This has never been this bad before, it's getting so bad that I'm considering scrapping the lot.
I guess I have to spend a day resetting all zigbee devices, and redoing the entire network first.

fribse, 2021-03-04:
Ok, so I took the siren that was acting weird outside in our mailbox (sort of faraday cage), and removed it from the network.
And now it's almost back to normal.
That's really weird. But I guess it's a problem with the heiman siren, and thus I close this case.

closed 2021-03-04 by fribse