dresden-elektronik/deconz-rest-plugin #4485: Deconz show light is on when off

Karstensson, 2021-02-26:
## Describe the bug
I have noticed that my deconz setup with conbee two is reporting that one of my 2 Lidl RGB lists and one Lidl E14 RGB light shows that the state is on even though it is off and should be off. The control of the light always works and the light shows off when I turn it off. But after a few hours it reports as on again even if it is not.
This happens daily. However my lidl GU10 always shows the correct state.

## Steps to reproduce the behavior
  1. Turn light on
  2. Turn light off and verify light is reported as off
  3. Wait a few hours
  4. Light is reported as on even though it is off

## Expected behavior
Light is expected to show ass off when it is off.

## Screenshots

## Environment
 - Host system: PC 
 - Running method:  Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.09.03
 - Device: ConBee II 
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context
Any suggestions or anything I can provide to make the debugging easier?

Smanar, 2021-02-26:
Hello, by curisity, your third app permit a websocket logging ?
To see if the problem is coming from a bad websocket notification, and wich one if we can see it, because will be harder to log all notifications from the device itself on deconz log.

Karstensson, 2021-02-26:
Thanks for the quick reply. I use the deconz integration for homeassistant. But The light also shows the wrong value in phoscon. However if I through VNC click network update in the deconz gui for the node in question it updates to the correct status again.

Smanar, 2021-02-26:
>But The light also shows the wrong value in phoscon

This is normal, phoscon use the same API (so the same websocket notification too) than HA, so you will have same result.

>However if I through VNC click network update in the deconz gui for the node in question it updates to the correct status again

Here, I don't understand your manipulation. the "network update" ? it s something specific to HA ?

Karstensson, 2021-02-26:
Sorry, by "network update" I mean pressing these buttons. 
[image]

Smanar, 2021-02-26:
I think the upload have failed ^^

Karstensson, 2021-02-26:
[image]

Smanar, 2021-02-26:
For information the "light state" is on cluster 0x0006, attribute 0x0000.
And the API is sending notification using the websocket everytime this value change.
You can check the value in same time in deconz, but take care, when you press the "read" button on deconz, it can update the value in the API too (so it can false the result)

Karstensson, 2021-02-26:
I was only pressing that to test if it made a difference. But are you suggesting me doing something with the cluster and attribute information or just general knowledge?

Smanar, 2021-02-26:
Yep, pls read the value to see if it s the good one when it bug for you.

Karstensson, 2021-02-27:
[image]
That attribute shows the correct state as in the picture. When the screenshot was taken the light was off, but was reported to be on in HomeAssistant and phoscon. When I pressed the read-button, phoscon and HomeAssistant showed the correct state again.

Smanar, 2021-02-27:
Ok, but will be hard to check.

If the device use reporting like other, you need to have

> 
>     Turn light on
>     Turn light off and verify light is reported as off
>     Wait a few hours
>     Light is reported as on even though it is off

Then after a time 

> Light is reported as off after a time ?

No ? Because when you have pressed the "read" button it have trigger a read attribute request, so deconz have read the correct value on the device. The state on the device is good, so the problem can be from a bad websocket notification, but as the device send periodicaly this value, you need to have this result naturaly (The api send notification eveytime this value change), you have not log about a "polling" for this MAC adress ?

Something that can be usefull, if your third app doesn't provide a websocket logging.
Use a plugin in your browser, for exemple I m using "browser websocket client" on firefox, use the url `ws://IP:WEBSOCKET_PORT` for exemple for me `ws://192.168.1.1:8088`
It will log all websockets notifications, like that you can find the bad one, and hoping for something strange in it to find the problem.

Karstensson, 2021-02-27:
I have no such log, and not really sure about how to get it. But I will try to monitor the web socket traffic. Thanks for the help.

closed 2021-03-29 by github-actions[bot]