dresden-elektronik/deconz-rest-plugin #4486: Mesh not working properly?

koolma, 2021-02-26:
## Describe the question or issue you are having

Hi,
I am using a Conbee 2 stick with an extension cord on a Raspi in my basement in combination with an openHAB installation.
I am mostly using Osram LED stripes and plugs and also some IKEA motion sensors. Some of my devices are not reachable from time to time, even though they should imo as they are only a few metres away from devices that are connected all the time. From my understanding the chord-powered LED stripes and the plugs should repeat the signal.
Since I use the Conbee stick in a headless environment, I cannot visualize my network, but the undocumented /connectivity2 endpoint I read about in another issue returns the neighbours I expect.
So why would some devices fail to connect to the network regularly and what can I do do improve the networks stability.

## Screenshots

## Environment

 - Host system: Pi Zero
 - Running method: Raspian
 - Firmware version: (26660700)
 - deCONZ version: (2.07.01 / 08.12.2020)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Nope

## deCONZ Logs

## Additional context

popy2k14, 2021-02-26:
You can install X server + VNC to show the deconz GUI on headless installations (like PI).
There you can check the Mesh.

I know from my devices that some chord-powered devices do also not act as a mesh-router.
HUE's works the best as mesh routers in my case.

Smanar, 2021-02-26:
> Some of my devices are not reachable from time to time

Thoses one are sensor or router ?

Mimiix, 2021-03-01:
As @popy2k14  mentioned, you can do that https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

Logs would help here :)

marcomania75, 2021-03-01:
i had the same issue (sort of) and I had multiple sources. 

Placing of the Conbee Stick (behind / nearby my Synology NAS)
non screened wire (LAN) nearby wich also had an influence on my Powerline Network and DSL-Connection
faulty USB extension

koolma, 2021-03-01:
Thank you for the feedback. After some digging I was able to get the GUI working using X-forwarding.
Here is a screenshot of the network:
[image]
The device in question is the "Steckdose Stehlampe" in the top right corner. It should be connected to the "LED Streifen Esszimmer" (also top right) all the time as the devices are only like 3 meters apart in the same room.

Smanar, 2021-03-01:
Effectively, your mesh is not terrific ...
Wich one device is the "Steckdose Stehlampe" ?
Can be the device type (e module inside a wall have lower range than a bulb), can be the brand too (I don't have fantastic return for sonoff device), or perhaps a problem with wifi (never see this kind of issue on my side, but try to chnage your wifi channel)

koolma, 2021-03-01:
That's an [Osram Smart+ Plug](https://www.amazon.de/Osram-schaltbare-fernbedienbar-Lichtsteuerung-kompatibel/dp/B074PZLX2P), not as exposed as a light bulb, but also not mounted within a wall.

Smanar, 2021-03-01:
And the "LED Streifen Esszimmer" ? you have 3 devices with this MAC adress range 0x7CB03EA, and 2 of 3 have more orange than green "lines"

koolma, 2021-03-01:
All three "LED Streifen" are [Osram Smart+ LED stropes](https://www.amazon.de/Osram-ZigBee-Streifen-Warmwei%C3%9F-tageslicht/dp/B07482ZJLY).

closed 2021-03-31 by github-actions[bot]