dresden-elektronik/deconz-rest-plugin #4489: SD image with RaspBee 2 not working

xymor, 2021-02-26:
## Describe the question or issue you are having
Using the SD image: https://phoscon.de/en/raspbee/sdcard
WIth PI4 8GB
deCONZ fails to recognize the device RaspBee 2

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: (Raspbian from DE with VNC
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device:  RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? no

## deCONZ Logs
```
22:04:37:801 wait reconnect 1 seconds
22:04:37:843 dev /dev/ttyS0 (/dev/ttyS0)
22:04:37:845 COM check bootloader
22:04:38:800 failed to reconnect to network try=2
22:04:38:848 dev /dev/ttyS0 (/dev/ttyS0)
22:04:38:854 GW update firmware not found: 
22:04:39:849 COM timout query bootloader, assume application
22:04:41:300 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:04:41:301 device state timeout ignored in state 2
22:04:43:800 try to reconnect to network try=3
22:04:44:301 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:04:44:301 device state timeout ignored in state 2
22:04:47:300 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:04:47:301 device state timeout ignored in state 2
22:04:48:801 try to reconnect to network try=4
22:04:48:853 dev /dev/ttyS0 (/dev/ttyS0)
22:04:48:859 GW update firmware not found: 
22:04:50:301 command queue EVENT_TIMEOUT, cmd: CMD_CHANGE_NET_STATE, seq: 59
22:04:53:301 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:04:53:301 device state timeout ignored in state 2
22:04:53:801 try to reconnect to network try=5
22:04:55:301 command queue EVENT_TIMEOUT, cmd: CMD_CHANGE_NET_STATE, seq: 59
22:04:58:301 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:04:58:302 device state timeout ignored in state 2
22:04:58:800 try to reconnect to network try=6
22:04:58:851 dev /dev/ttyS0 (/dev/ttyS0)
22:04:58:857 GW update firmware not found: 
22:05:00:301 command queue EVENT_TIMEOUT, cmd: CMD_CHANGE_NET_STATE, seq: 59
22:05:03:301 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:05:03:301 device state timeout ignored in state 2
22:05:03:801 try to reconnect to network try=7
22:05:05:301 command queue EVENT_TIMEOUT, cmd: CMD_CHANGE_NET_STATE, seq: 59
22:05:08:301 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:05:08:301 device state timeout ignored in state 2
22:05:08:801 try to reconnect to network try=8
22:05:08:854 dev /dev/ttyS0 (/dev/ttyS0)
22:05:08:861 GW update firmware not found: 
22:05:10:300 command queue EVENT_TIMEOUT, cmd: CMD_CHANGE_NET_STATE, seq: 61
22:05:13:300 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:05:13:301 device state timeout ignored in state 2
22:05:13:801 try to reconnect to network try=9
22:05:15:301 command queue EVENT_TIMEOUT, cmd: CMD_CHANGE_NET_STATE, seq: 61
22:05:18:301 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:05:18:301 device state timeout ignored in state 2
22:05:18:801 try to reconnect to network try=10
22:05:18:850 dev /dev/ttyS0 (/dev/ttyS0)
22:05:18:856 GW update firmware not found: 
22:05:20:300 command queue EVENT_TIMEOUT, cmd: CMD_CHANGE_NET_STATE, seq: 61
22:05:23:301 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:05:23:301 device state timeout ignored in state 2
22:05:23:800 reconnect network failed, try later
```

## Additional context
Trying to update deCONZ manually the device reboot loops when applying the firmware step of this guide:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

```
22:09:01:601 using firmware file: deCONZ_RaspBeeII_0x26690700.bin.GCF 
22:09:01:654 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
22:09:01:654 check symlink /dev/ttyAMA0 (/dev/ttyS0)
22:09:01:654 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
22:09:01:654 check symlink /dev/ttyS0 (/dev/ttyS0)
22:09:01:654 dev /dev/ttyS0 (/dev/ttyS0)
Reboot device RaspBee (RaspBee)
22:09:01:656 retry, elapsed 21 seconds
22:09:01:761 parse commands: /usr/bin/GCFFlasher_internal.bin -x 3 -t 60 -f deCONZ_RaspBeeII_0x26690700.bin.GCF

22:09:01:761 using firmware file: deCONZ_RaspBeeII_0x26690700.bin.GCF 
22:09:01:814 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
22:09:01:814 check symlink /dev/ttyAMA0 (/dev/ttyS0)
22:09:01:814 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
22:09:01:814 check symlink /dev/ttyS0 (/dev/ttyS0)
22:09:01:814 dev /dev/ttyS0 (/dev/ttyS0)
Reboot device RaspBee (RaspBee)
22:09:01:815 retry, elapsed 21 seconds
22:09:01:910 parse commands: /usr/bin/GCFFlasher_internal.bin -x 3 -t 60 -f deCONZ_RaspBeeII_0x26690700.bin.GCF
```

Mimiix, 2021-03-01:
http://phoscon.de/en/raspbee2/install#raspbian

Did you follow this guide?

xymor, 2021-03-01:
Yes but it fails too

´´´
pi@phoscon:/tmp/raspbee2-rtc-master $ sudo apt-get install raspberrypi-kernel-headers -y 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20210201-1).
0 upgraded, 0 newly installed, 0 to remove and 151 not upgraded.
pi@phoscon:/tmp/raspbee2-rtc-master $ make
checking deb package dependencies...
i2c-tools
build-essential
raspberrypi-kernel-headers
build rtc module...
make -C /lib/modules/5.4.72-v7l+/build M=/tmp/raspbee2-rtc-master modules
make[1]: *** /lib/modules/5.4.72-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:17: build] Error 2
´´´

xymor, 2021-03-01:
Yes but the HW clock step fails:

```pi@phoscon:~ $ sudo hwclock --verbose
hwclock from util-linux 2.33.1
System Time: 1550139173.526348
Trying to open: /dev/rtc0
Trying to open: /dev/rtc
Trying to open: /dev/misc/rtc
No usable clock interface found.
hwclock: Cannot access the Hardware Clock via any known method.```

Mimiix, 2021-03-01:
Did you put the battery in it?

xymor, 2021-03-01:
It came installed in the box but checking it, looks properly installed
according to the image https://phoscon.de/raspbee2/img/raspbee2-battery.png

closed 2021-03-01 by xymor

SwoopX, 2021-03-01:
The HW clock works when you update to 0x26690700