dresden-elektronik/deconz-rest-plugin #4490: Tradfri On/Off switch reporting wrong battery levels

brujoand, 2021-02-27:
## Describe the bug
The battery levels reported by Tradfri On/Off switches are all over the place

## Steps to reproduce the behavior
  1. Connect ikea Trådfri On/Off switch
  2. Hit the api on the device, and read battery level
  3. Repeat after a few hours and be amazed 
  4. Observe that the reported battery levels go from 80 to 20 and everywhere between over the course of a few days
  5. Eventually loose connectivity which might or might not be related to #4218
-->

## Expected behavior
That the correct battery level should be reported like they are with Tråtdfri motion sensors, which show a slow decrease over time.

## Screenshots
I can share some if there is a reasonable place to host images I can use

## Environment
 - Host system: PC
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.09.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system?: No

## deCONZ Logs
There is a lot of output, please advice on what I should be searching for here. 

## Additional context
I have three of these switches, all three report completely random battery levels. Two of them stopped working 2 days ago after being in use for less than 2 months and 1 of them already had a battery replaced. I'll manually measure their voltage and update this ticket.

brujoand, 2021-02-27:
Measured the voltage on both the batteries of the On/Off switches that were no longer being seen by deconz, they both shoved 3v each, but from the API they are being reported as 87 and 21. As soon as I placed the battery back in they both started working as expected though, but with random battery levels.

brujoand, 2021-02-27:
Result of calling the api on one of the switches before and after i restarted it by removing the battery. So the battery is the same, yet now it's reported as a much higher value.
```
{
  config: {
    alert: none,
    battery: 21,
    group: 5,
    on: true,
    reachable: false
  },
  ep: 1,
  etag: f0de280120d796fec35b058b1501bc04,
  lastseen: null,
  manufacturername: IKEA of Sweden,
  mode: 1,
  modelid: TRADFRI on/off switch,
  name: Sov Trym,
  state: {
    buttonevent: 2003,
    lastupdated: 2021-02-25T19:50:14.754
  },
  type: ZHASwitch,
  uniqueid: 5c:02:72:ff:fe:81:fb:26-01-1000
}
```

```
{
  "config": {
    "alert": "none",
    "battery": 74,
    "group": "5",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "d3585ffe570ef7940630ec235dc3243b",
  "lastseen": "2021-02-27T08:53Z",
  "manufacturername": "IKEA of Sweden",
  "mode": 1,
  "modelid": "TRADFRI on/off switch",
  "name": "Sov Trym",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2021-02-27T08:48:06.248"
  },
  "type": "ZHASwitch",
  "uniqueid": "5c:02:72:ff:fe:81:fb:26-01-1000"
}
```

Mimiix, 2021-03-01:
I think this is unrelated to deCONZ and more to the device. Devices report battery status very unreliable. Doubt that there can be done something about it. 

@SwoopX @Smanar @manup  Maybe you can elaborate.

SwoopX, 2021-03-01:
Well, that's basically it. Deconz is hopelessly dependent on the values the device reports. If the device's measurements are jumpy, this is what you get.

brujoand, 2021-03-01:
Ah, right that makes sense. I was hoping that these battery readings were an indicatior of why the remotes suddenly dissapeared, but that seems unrelated then. Thanks for the info 👍

closed 2021-03-01 by brujoand