dresden-elektronik/deconz-rest-plugin #4491: Xiaomi temperatur Sensor only a es

Claus1985, 2021-02-27:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2021-03-01:
You mentioned "but it only has 0 degrees Celsius as temperature value. It regularly send updates but always 0 degrees." Is this in the API?

Can you read the cluster in the Gui? See if that provides the correct temp?

SwoopX, 2021-03-01:
Try to re-pair the sensor and push the button once a second while it is pairing. I assume it is still the "older" verion WSDCGQ11LM?

Claus1985, 2021-03-07:
Tried many different things including pressing the button while pairing and repaired like 20 times..
Solution: I´m gonna buy a new sensor.

Thx anyways.

closed 2021-03-07 by Claus1985